coolsnowwolf/lede #5411: 阉割MT7621的veic/vint功能，workaround死机问题

lunzima, 2020-09-03:
OpenWrt上游bug FS#1102, FS#1618, FS#1539, FS#1514, FS#2554, FS#1603, FS#2245, FS#1170……等数不清的bug都与此相关。
凡是出现机器随机卡死/单核CPU占用100%的问题并且crashlog的trace内包含except_vec_vi_end的都与之相关。

据从胡伟武临时工@FlyGoat（18岁是学生，太强了！！！）那里了解到，早期步进的MT7621 eco1/2机器的veic/vint有硬件bug。由于Errata是口口相传，所以具体影响哪些步进暂时不可考。
最简单的复现方法是bash脚本fork bomb；
在/etc/sysctl.conf内加入net.core.busy_poll=100和net.core.busy_read=100的系统极易触发此Bug；
开启hardware offload后此Bug触发几率也会加大。

我测试下来的结果是，上游OpenWrt 18.06用GCC 7.3+4.14内核编译有此Bug，GCC 7.5+4.14内核编译此Bug难以复现，GCC 8此Bug又再次出现。总之此问题的触发条件和barrier/ordering相关。可惜据FlyGoat介绍目前上游的Hauke没有bisect出问题根源来。

按下图所示阉割veic/vint之后bash bomb只会吃光内存，而hwnat/busy_poll的问题都消失了。

[image]

MTK/Ralink的NDA源码里面没有cpu-feature-overrides.h的内容，是上游的内核开发者手动添加的；后来内核才支持自动检测识别并开启veic/vint，这才导致bug暴露。

我这里的项目不是直接clone的大雕的源码（本质上是lede-17.01/openwrt-18.06的混合userland+含部分openwrt-19.07的补丁的4.14 kernel)，所以就不提交pull request了。
最后再次感谢FlyGoat的指导以及农企临时工孜孜不倦地背黑锅。

FlyGoat, 2020-09-04:
I'm contacting upstream to get the issue fixed in upstream.
Please carry this patch downstream before that.

277383761, 2020-09-04:
难怪我这款newifi3-d2刷的openwrt开启ACC加速后,总是不定时不定期随机性的自动重启,后来我重新编译不再集成这个ACC确实稳定的运行了

jxph1123, 2020-09-04:
大神们能否更新下

EnnawYang, 2020-09-04:
> 大神们能否更新下

https://github.com/coolsnowwolf/lede/issues/5411#issuecomment-686824289

FlyGoat, 2020-09-04:
> 
> 
> > 大神们能否更新下
> 
> [#5411 (comment)](https://github.com/coolsnowwolf/lede/issues/5411#issuecomment-686824289)
> dalao不是说等上游修复吗

上游还没个ETA，下游先carry着吧。

Cathgao, 2020-09-04:
> 难怪我这款newifi3-d2刷的openwrt开启ACC加速后,总是不定时不定期随机性的自动重启,后来我重新编译不再集成这个ACC确实稳定的运行了

D2是newifi mini吗？还是newifi？我的newifi开着HWNAT，倒是很少见到有重启的情况，不过偶见断流和WiFi信号闪断

lunzima, 2020-09-04:
> newifi mini

> > 难怪我这款newifi3-d2刷的openwrt开启ACC加速后,总是不定时不定期随机性的自动重启,后来我重新编译不再集成这个ACC确实稳定的运行了
> 
> D2是newifi mini吗？还是newifi？我的newifi开着HWNAT，倒是很少见到有重启的情况，不过偶见断流和WiFi信号闪断

Newifi Y1和Newifi Y1S都是MT7620A的机器，和本issue无关

sxx1314, 2020-09-04:
> 难怪我这款newifi3-d2刷的openwrt开启ACC加速后,总是不定时不定期随机性的自动重启,后来我重新编译不再集成这个ACC确实稳定的运行了

近期各架构都有莫名重启问题。包括x86_64 ipq 等。日志里还没什么可用信息。所以异常重启还真不一定是这原因

lunzima, 2020-09-04:
> > 难怪我这款newifi3-d2刷的openwrt开启ACC加速后,总是不定时不定期随机性的自动重启,后来我重新编译不再集成这个ACC确实稳定的运行了
> 
> 近期各架构都有莫名重启问题。包括x86_64 ipq 等。日志里还没什么可用信息。所以异常重启还真不一定是这原因

上游openwrt的master以及coolsnowwolf的fork的master的情况我不是很了解，尤其是5.4内核的

Cathgao, 2020-09-04:
> > newifi mini
> 
> > > 难怪我这款newifi3-d2刷的openwrt开启ACC加速后,总是不定时不定期随机性的自动重启,后来我重新编译不再集成这个ACC确实稳定的运行了
> > 
> > 
> > D2是newifi mini吗？还是newifi？我的newifi开着HWNAT，倒是很少见到有重启的情况，不过偶见断流和WiFi信号闪断
> 
> Newifi Y1和Newifi Y1S都是MT7620A的机器，和本issue无关

newifi3是7621

Wilson-coder, 2020-09-05:
> > > 难怪我这款newifi3-d2刷的openwrt开启ACC加速后,总是不定时不定期随机性的自动重启,后来我重新编译不再集成这个ACC确实稳定的运行了
> > 
> > 
> > 近期各架构都有莫名重启问题。包括x86_64 ipq 等。日志里还没什么可用信息。所以异常重启还真不一定是这原因
> 
> 上游openwrt的master以及coolsnowwolf的fork的master的情况我不是很了解，尤其是5.4内核的
> 但是顶楼说的bug影响至今所有的mt7621机器。

顶楼只提到了MT7621 eco1/2的芯片。我前不久买的小米AC2100，已经是eco3的芯片了，两周前也遇到过一次自动重启，不过那时候刚好是WAN连接到了48小时，被运营商自动断掉重连，所以无法判定是这个原因、还是系统bug导致的重启。

lunzima, 2020-09-05:
> 顶楼只提到了MT7621 eco1/2的芯片。我前不久买的小米AC2100，已经是eco3的芯片了，两周前也遇到过一次自动重启，不过那时候刚好是WAN连接到了48小时，被运营商自动断掉重连，所以无法判定是这个原因、还是系统bug导致的重启。

我的机器实际上也是eco3的。。

AmadeusGhost, 2020-09-05:
> 两周前也遇到过一次自动重启，不过那时候刚好是WAN连接到了48小时，被运营商自动断掉重连，所以无法判定是这个原因、还是系统bug导致的重启。

这个应该跟 flow offload 没关系😂

hcym, 2020-09-05:
可能是luci兼容性问题，刚刚试用今天早上armv8的也是，不知道和那个有冲突，什么加速减速的都没开，ssh操作正常，网页会卡住然后重启

jxph1123, 2020-09-07:
好像大雕的这个MT7621里面默认就没有veic/vint，没找到。。

lunzima, 2020-09-30:
农企临时工替联发科背黑锅的第27天。。

summershrimp, 2020-12-08:
mark

closed 2021-06-20 by coolsnowwolf