coolsnowwolf/lede #5417: 自编译固件安装到N1盒子后除了material主题以外其它主题全是变成了系统主题

Margular, 2020-09-05:
##  1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：例如选择Argon_Light_Green主题，实际上变成了openwrt的默认主题，导致有的插件用不了，现在只有Material主题能正常切换，其它都是会变成系统默认的那个主题

### (2) 路由器型号和固件版本

主机名 | OpenWrt
-- | --
主机型号 | Phicomm N1
架构 | ARMv8 Processor rev 4 (v8l) x 4
固件版本 | OpenWrt R20.8.27 / LuCI Master (git-20.223.26773-d18ef13)
内核版本 | 5.4.60-flippy-42+o

CMJ781, 2020-09-05:
主题的对应源增加了吗？

Margular, 2020-09-05:
没有刻意添加源，但是make menuconfig里面有这些主题，应该是代表没问题吧？

xxp2277, 2020-09-05:
意思是编译了主题，但是切换后变成默认的主题了吧？那估计是主题没更新进去。
我[今天](https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4019532&page=1&extra=#pid9227007)编译的没问题。

yonggedebaqi, 2020-09-23:
修改openwrt/feeds/luci/collections/luci/Makefile文件就可以了

closed 2021-06-20 by coolsnowwolf