coolsnowwolf/lede #5418: 请大佬更新luci-app-ssr-plus对raw加密订阅的更新能力

Dimitri2020007, 2020-09-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
更新这类型的ssr订阅失败“https://raw.githubusercontent.com/ssrsub/ssr/master/ssrsub”无法解析
### (1) 具体问题

A：“https://raw.githubusercontent.com/ssrsub/ssr/master/ssrsub”加密链接更新ssr订阅失败

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

这是路由器的日志

#raw解析失败的

shadowsocksr_subscribe[24856]: https://raw.githubusercontent.com/ssrsub/ssr/master/ssrsub 订阅数据获取失败 错误代码: 4
:https://raw.githubusercontent.com/ssrsub/ssr/master/ssrsub 订阅数据获取失败 错误代码: 4
date: multiple output formats specified
wget-ssl: unable to resolve host address 'raw.githubusercontent.com'
Resolving raw.githubusercontent.com... failed: Name does not resolve.
--2020-09-05 22:26:38--  https://raw.githubusercontent.com/ssrsub/ssr/master/ssrsub
:开始解析节点信息...
date: multiple output formats specified
:下载订阅成功...
date: multiple output formats specified
wget-ssl: unable to resolve host address 'raw.githubusercontent.com'
Resolving raw.githubusercontent.com... failed: Name does not resolve.
--2020-09-05 22:26:38--  https://raw.githubusercontent.com/ssrsub/ssr/master/ssrsub
:开始下载订阅链接到本地临时文件，请稍等...
date: multiple output formats specified
:开始更新在线订阅列表...
date: multiple output formats specified
:从 https://raw.githubusercontent.com/ssrsub/ssr/master/ssrsub 获取订阅
date: multiple output formats specified
2020-09-05 22:26:36 Shadowsocks/ShadowsocksR 2 Threads Started!

#普通的

shadowsocksr_subscribe[29170]: muma16fx 服务器订阅更新成功 服务器数量: 20 新增服务器: 20 删除服务器: 0
:在线订阅列表更新完成!请等待网页自动刷新!
2020-09-05 22:27:44 (1.35 KB/s) - written to stdout [5964/5964]

     0K .....                                                 100% 1.35K=4.3s

2020-09-05 22:27:43 Shadowsocks/ShadowsocksR 2 Threads Started!
Saving to: 'STDOUT'
Length: 5964 (5.8K) [text/html]
HTTP request sent, awaiting response... 200 OK
Connecting to muma16fx.netlify.app|130.193.50.240|:443... connected.
Resolving muma16fx.netlify.app... 130.193.50.240, 84.201.139.79
--2020-09-05 22:27:38--  https://muma16fx.netlify.app/
Location: https://muma16fx.netlify.app/ [following]
HTTP request sent, awaiting response... 301 Redirect
Connecting to muma16fx.netlify.com|130.193.50.240|:443... connected.
Resolving muma16fx.netlify.com... 130.193.50.240, 84.201.139.79
--2020-09-05 22:27:37--  https://muma16fx.netlify.com/
:开始下载订阅链接到本地临时文件，请稍等...
date: multiple output formats specified
:开始更新在线订阅列表...
date: multiple output formats specified
:从 https://muma16fx.netlify.com/ 获取订阅

#这是咋回事啊，只能支持ssr://的链接导入吗？

9l, 2020-09-06:
刚测试了，你给的连接可以用，没问题

Dimitri2020007, 2020-09-06:
也许吧

Dimitri2020007, 2020-09-06:
我刚刚把github和raw.githubusercontent.com加入gfwlist解析成功了，确实是网络问题，谢谢大家。

closed 2020-09-06 by Dimitri2020007