coolsnowwolf/lede #5421: 可道云提示上传文件太大，修改了php.ini和nginx.conf依然如此

happyboy19, 2020-09-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：可道云上传文件提示文件太大

### (2) 路由器型号和固件版本

A：x86， R20.8.27 

### (3) 详细日志

A： 提示信息413: request entity too large 
修改了/etc/php.ini和/etc/nginx/nginx.conf的大小限制和时间限制，还是不行，

**_nginx.conf的配置如下_**

# Please consider creating files in /etc/nginx/conf.d/ instead of editing this.
# For details see https://openwrt.org/docs/guide-user/services/webserver/nginx

worker_processes auto;

user root;

events {}

http {
        access_log off;
        log_format openwrt
                '$request_method $scheme://$host$request_uri => $status'
                ' (${body_bytes_sent}B in ${request_time}s) <- $http_referer';

        include mime.types;
        default_type application/octet-stream;
        sendfile on;
        tcp_nopush on;
        tcp_nodelay on;
        keepalive_timeout 65;
        types_hash_max_size 2048;
        client_max_body_size 100M;
        client_body_buffer_size 256k;
        large_client_header_buffers 200 100k;
        client_header_timeout 3600s;
        client_body_timeout 3600s;
        fastcgi_connect_timeout 3600s;
        fastcgi_send_timeout 3600s;
        fastcgi_read_timeout 3600s;

#..............................................................................
        client_body_in_file_only clean;
        client_body_temp_path /dev/shm 1 2;
        fastcgi_param  REQUEST_BODY_FILE  $request_body_file;

        gzip on;
        gzip_vary on;
        gzip_proxied any;

        root /www;

        include conf.d/*.conf;
}

 **_php.ini配置如下_**

[PHP]
zend.ze1_compatibility_mode = Off
engine = On
precision    =  12
y2k_compliance = On
output_buffering = Off
zlib.output_compression = Off
implicit_flush = Off
unserialize_callback_func =
serialize_precision = 100

open_basedir = /mnt/sdb1/:/tmp/:/proc/
disable_functions =
disable_classes =
expose_php = On
max_execution_time = 30
max_input_time = 60
memory_limit = 800M
error_reporting  =  E_ALL & ~E_NOTICE & ~E_STRICT

display_errors = On
display_startup_errors = Off
log_errors = Off
log_errors_max_len = 1024
ignore_repeated_errors = Off
ignore_repeated_source = Off
report_memleaks = On
track_errors = Off

variables_order = "EGPCS"
request_order = "GP"
register_globals = Off
register_long_arrays = Off
register_argc_argv = On
auto_globals_jit = On
post_max_size = 100M
magic_quotes_runtime = Off
magic_quotes_sybase = Off
auto_prepend_file =
auto_append_file =
default_mimetype = "text/html"

;doc_root = "/www"
user_dir =
extension_dir = "/usr/lib/php"
enable_dl = On
cgi.fix_pathinfo=1
- /etc/php.ini 1/56 1%

happyboy19, 2020-09-06:
max_execution_time = 30
max_input_time = 60
这两项每次修改成3600，重启路由就会恢复默认

happyboy19, 2020-09-06:
修改 /var/etc/kodexplorer/nginx.conf,可是重启路由后全恢复了默认，

**root@OpenWrt:~# vi /var/etc/kodexplorer/nginx.conf**
user root root;
worker_processes  1;
pid /var/run/nginx_kodexplorer.pid;
events {
    worker_connections  1024;
}
http {
    include /etc/nginx/mime.types;
    sendfile on;
    keepalive_timeout 65;
    client_max_body_size 500M;
    server {
    listen 8080;
        listen [::]:8080;
        server_name  localhost;
        location / {
            root /mnt/sdb1/kodexplorer;
            index index.html index.htm index.php;
        }
        error_page   500 502 503 504  /50x.html;
        location = /50x.html {
            root   html;
        }
        location ~ \.php$ {
            root /mnt/sdb1/kodexplorer;
            try_files $uri = 404; # PHP .....................404
            fastcgi_pass unix:/var/run/php7-fpm.sock; # ...... Unix ...........
            fastcgi_index index.php;
            fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name; #
            include /etc/nginx/fastcgi_params;
        }
    }
}
~

lexin8, 2020-09-07:
目前上传最大10G都没问题。重启路由就会恢复默认，应该是你的固件没有写入权限!不能在线解压大一点的压缩包，目前没找到教程。

happyboy19, 2020-09-07:
> 目前上传最大10G都没问题。重启路由就会恢复默认，应该是你的固件没有写入权限!不能在线解压大一点的压缩包，目前没找到教程。

换了一块板子，刷了新的固件，仍然是同样的问题，我觉得这不是个例的问题了，是不是有bug

eallion, 2020-09-09:
修改 `/etc/nginx/nginx.conf`里面的`client_max_body_size`试试。

happyboy19, 2020-09-09:
> 修改 `/etc/nginx/nginx.conf`里面的`client_max_body_size`试试。

试了，也不行

xyskwjy, 2020-10-21:
同样的问题。解决不了，换成了filebrowser了。

closed 2021-06-20 by coolsnowwolf