coolsnowwolf/lede #5426: 编译x86出错

SoDebug, 2020-09-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译x86出错

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：/home/w/lede/scripts/ipkg-build -c -o 0 -g 0 /home/w/lede/build_dir/target-x86_64_musl/gmp-6.1.2/ipkg-x86_64/libgmp /home/w/lede/bin/packages/x86_64/base
Packaged contents of /home/w/lede/build_dir/target-x86_64_musl/gmp-6.1.2/ipkg-x86_64/libgmp into /home/w/lede/bin/packages/x86_64/base/libgmp10_6.1.2-2_x86_64.ipk
echo "libgmp" >> /home/w/lede/staging_dir/target-x86_64_musl/pkginfo/gmp.default.install
rm -rf /home/w/lede/tmp/stage-gmp
mkdir -p /home/w/lede/tmp/stage-gmp/host /home/w/lede/staging_dir/target-x86_64_musl/packages
install -d -m0755 /home/w/lede/tmp/stage-gmp/usr/include
cp -fpR /home/w/lede/build_dir/target-x86_64_musl/gmp-6.1.2/ipkg-install/usr/include/gmp* /home/w/lede/tmp/stage-gmp/usr/include/
install -d -m0755 /home/w/lede/tmp/stage-gmp/usr/lib
cp -fpR /home/w/lede/build_dir/target-x86_64_musl/gmp-6.1.2/ipkg-install/usr/lib/libgmp.{a,so*} /home/w/lede/tmp/stage-gmp/usr/lib/
find /home/w/lede/tmp/stage-gmp -name '*.la' | xargs -r rm -f;   	find /home/w/lede/tmp/stage-gmp -name '*.la' | xargs -r rm -f; 
if [ -f /home/w/lede/staging_dir/target-x86_64_musl/packages/gmp.list ]; then /home/w/lede/scripts/clean-package.sh "/home/w/lede/staging_dir/target-x86_64_musl/packages/gmp.list" "/home/w/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/w/lede/tmp/stage-gmp ]; then (cd /home/w/lede/tmp/stage-gmp; find ./ > /home/w/lede/tmp/stage-gmp.files); 	SHELL= flock /home/w/lede/tmp/.staging-dir.flock -c ' mv /home/w/lede/tmp/stage-gmp.files /home/w/lede/staging_dir/target-x86_64_musl/packages/gmp.list && cp -fpR /home/w/lede/tmp/stage-gmp/* /home/w/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/w/lede/tmp/stage-gmp
touch /home/w/lede/staging_dir/target-x86_64_musl/stamp/.gmp_installed
make[3]: Leaving directory '/home/w/lede/package/libs/gmp'
time: package/libs/gmp/compile#93.35#9.60#186.79
make[2]: Leaving directory '/home/w/lede'
make[1]: *** [package/Makefile:107: /home/w/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/w/lede'
make: *** [/home/w/lede/include/toplevel.mk:224：world] 错误 2

hcym, 2020-09-10:
编译没什么问题，就是最近的版本容易莫名其妙的重启，要命的事，用官方master编译lienol的反而没一点问题，大雕的只能用六月份版本度日

qq1316237299, 2020-09-18:
> 份版本度日

111

freeleeman, 2020-09-24:
我的是32位

closed 2021-06-20 by coolsnowwolf