coolsnowwolf/lede #5428: ipq40xx 交换机中wan口缺失，不能设置VLAN

876873549, 2020-09-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：  虚拟交换机中wan口缺失，不能设置VLAN

### (2) 路由器型号和固件版本
竞斗云ipq40xx  P&W R619AC 128M 
openWrt R20.8.27 内核版本4.19.138
A：
[image]

[image]

### (3) 详细日志

A：

wqgk, 2020-09-09:
我也是这个问题，直接修改源码编译出来，网络会出现问题

jindaxia, 2020-09-09:
我的也是。这是哪里有什么问题吗》

467815891a, 2020-09-10:
我也是这个问题，ifconfig里面也没有wan

467815891a, 2020-09-10:
> 我也是这个问题，直接修改源码编译出来，网络会出现问题

什么问题，我遇到了有线网络不通的问题，我们一样吗

AmadeusGhost, 2020-09-13:
没 wan 其实应该也是可以划 vlan 的，vlan 要有 wan 的话可以用这里的源码：https://github.com/project-openwrt/openwrt

876873549, 2020-09-13:
> 没wan其实应该应该也是可以划vlan的，vlan要有wan的话可以用这里的源码：[https](https://github.com/project-openwrt/openwrt) : [//github.com/project-openwrt/openwrt](https://github.com/project-openwrt/openwrt)

设置了，连不上网，所有接口变成LAN。

请教一下如何设置？

876873549, 2020-09-13:
设置了VLAN，接口就用不了

AmadeusGhost, 2020-09-13:
你还是换源码吧...

easehu, 2020-09-13:
我也改过了，结果和上面出现的现象一样，vlan 无法添加，加上去 cpu 那个就有问题。不知道具体的是哪几个文件是交换机这块的，知道的话就可以通过对比去找出不同。我见到几个版本的固件，有的有wan口的，而且正常用，好像也是lean的固件版本是git-19.299.52050-0b735ea，OpenWrt R9.11.9。

876873549, 2020-09-13:
> 我也改过了，结果和上面出现的现象一样，vlan 无法添加，加上去 cpu 那个就有问题。不知道具体的是哪几个文件是交换机这块的，知道的话就可以通过对比去找出不同。我见到几个版本的固件，有的有wan口的，而且正常用，好像也是lean的固件版本是git-19.299.52050-0b735ea，OpenWrt R9.11.9。
无WAN的原因可能是换了openwrt官方的源码驱动。  但是即使换了，这VLAN的功能总该可以用吧。现在就是这功能用不了

llfj, 2020-09-18:
L大说了，和官方一致，就没有VLAN了，想要VLAN的需要改内核，以前修改过后来经过大家的投票取消了。这是你们投票的结果。哎~~~福气你们了。搞得想要用VLAN的都没法用。

hzjnet, 2020-09-19:
> L大说了，和官方一致，就没有VLAN了，想要VLAN的需要改内核，以前修改过后来经过大家的投票取消了。这是你们投票的结果。哎~~~福气你们了。搞得想要用VLAN的都没法用。

热  真是扯淡啊

easehu, 2020-09-19:
我感觉还是公布修改方法，不然需要的人就很麻烦。老的源码不知道哪里有。

llfj, 2020-09-19:
> > L大说了，和官方一致，就没有VLAN了，想要VLAN的需要改内核，以前修改过后来经过大家的投票取消了。这是你们投票的结果。哎~~~福气你们了。搞得想要用VLAN的都没法用。
> 
> 热 真是扯淡啊

给你看看群里的对话，你就明白了。都是都是民主选择的。L大自己说的，类似于这种优化他做了很多，可惜有些人不懂，就认为官方的好。官方的好的话，他有必要费时费力去优化吗。。。。
[https://github.com/llfj/openwrt/blob/master/%E5%AF%B9%E8%AF%9D.jpg](url)

llfj, 2020-09-19:
> 我感觉还是公布修改方法，不然需要的人就很麻烦。老的源码不知道哪里有。

看看对话，貌似不简单。。。

closed 2020-09-19 by 876873549

paranoiagu, 2020-10-04:
VLAN 还能用吗？我尝试用软vlan，但是没成功。

leewp14, 2021-01-05:
我就分享在这里了吧！我相信应该对有些人有帮助
感谢 @llfj 在上楼提供的群聊记录 我自行在项目抽一个2月的版本来编译 测试vlan操作完美 因为我的isp是需要vlan500才能上网的 所以真的没办法就自己来弄了

https://github.com/leewp14/lede/releases/tag/leewp14-20210105

我的版本是非常，非常干净的 什么都没有 只有ddns 就整个几乎跟op官方编译的没差 因为我海外人所以不需要什么特别的需求。如果不介意的话 欢迎尝试使用！