coolsnowwolf/lede #5435: 麻烦帮我看下这个编译报错

jtzgc, 2020-09-09:
之前编译都没问题，昨晚更新源码再编译就一直报这个错

[image]

lxwq, 2020-09-10:
我也卡在这,有文件下载不下来,naiveproxy 编译出错,

AmadeusGhost, 2020-09-10:
缺了 curl 命令， `apt install curl`

EnnawYang, 2020-09-10:
> 缺了 curl 命令， `apt install curl`

发现群里面不少人都少curl (

pangfeng161, 2020-09-10:
这个月以来很多人编译出错

lxwq, 2020-09-10:
curl: (28) Connection timed out after 20000 milliseconds
unzip:  cannot find or open /home/zz/lean/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip, /home/zz/lean/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip.zip or /home/zz/lean/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip.ZIP.
bash: ./gn/out/gn: No such file or directory
ninja: Entering directory `out/Release'
ninja: fatal: chdir to 'out/Release' - No such file or directory
make[3]: *** [Makefile:89: /home/zz/lean/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built] Error 1
make[3]: Leaving directory '/home/zz/lean/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#1.38#0.55#153.12
make[2]: *** [package/Makefile:113: package/feeds/helloworld/naiveproxy/compile] Error 2
没有出国条件,如何下载gn???

xtwz, 2020-09-10:
install curl  重样出来这个报错

stztao, 2020-09-10:
> 缺了 curl 命令， `apt install curl`
装过还是依然这个错误，原来一直正常，更新9号的源码后就一直报这个错

closed 2021-06-20 by coolsnowwolf