coolsnowwolf/lede #5438: sfe和flowoffload 什么区别？

a1392429292, 2020-09-11:
最近编译了一次，发现 有一个 luci-app-sfe 和 luci-app-flowoffload 这2个之间什么区别啊？好像都是 Turbo ACC 网络加速设置 ？

另外这次发现  luci-app-nlbwmon 成了必装，这是干嘛用的？可以不要咩？

Fddh2012, 2020-09-13:
二选一，推荐用sfe

Dimitri2020007, 2020-09-16:
sfe的
[image]

flowoffload的
[image]

sfe有5个功能（桥接加速，ipv6加速，sfe加速）
flowoffload有4个功能（hwnat，flowoffload加速）
bbr,dns加速是两个都有的。

两个一起开没啥区别，就是多几个加速功能。
可以两个加速都编译进固件的。（x86可以这样子，别的就不知道了。）

NagaseKouichi, 2020-11-23:
是不是flowoffload对MTK的7621A有硬件加速，sfe只是软件加速？

closed 2021-06-20 by coolsnowwolf