coolsnowwolf/lede #5439: 高级重启 警告：设备（linksys,wrt1200ac）未知或不是双分区设备！

dunkeng, 2020-09-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：高级重启的UI有问题
警告：设备（linksys,wrt1200ac）未知或不是双分区设备！

root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00200000 00020000 "u-boot"
mtd1: 00040000 00020000 "u_env"
mtd2: 00040000 00020000 "s_env"
mtd3: 00100000 00020000 "devinfo"
mtd4: 02800000 00020000 "kernel1"
mtd5: 02200000 00020000 "rootfs1"
mtd6: 02800000 00020000 "kernel2"
mtd7: 02200000 00020000 "ubi"
mtd8: 02600000 00020000 "syscfg"
mtd9: 00680000 00020000 "unused_area"

root@OpenWrt:~# fw_printenv boot_part
boot_part=1
root@OpenWrt:~# fw_setenv boot_part 2
root@OpenWrt:~# fw_printenv boot_part
boot_part=2
root@OpenWrt:~# fw_setenv boot_part 1
root@OpenWrt:~# fw_printenv boot_part
boot_part=1

### (2) 路由器型号和固件版本

A：wrt1200ac v2 

OpenWrt R20.8.27 / LuCI Master (git-20.223.26773-d18ef13)
--

### (3) 详细日志

A：
不知道日志在哪

dunkeng, 2020-09-11:
补充说明一下，没办法切换分区，因为kernel1和2的地址一样了，怎么切都是同一个地址

dunkeng, 2020-09-11:
补充一下，我命令行切分区后，重启路由，还是这个版本，没办法切到前一个版本

dunkeng, 2020-09-11:
强刷官方固件
root@OpenWrt:/tmp# sysupgrade -n -v -F ./FW_WRT1200ACV2_2.0.6.191786_release.img
能正常刷进去，也能正常切换分区，现在官方和OPENWRT能正常切换，只有OPENWRT高级重启的UI上无法切换

edmondfr, 2020-11-28:
也遇到同样的问题，暂时使用下面的命令登陆进去切换分区，希望老大抽空解决一下。

查看当前分区：
fw_printenv boot_part

如果返回结果是 1，那么切换另一分区命令：
fw_setenv boot_part 2 && reboot

如果是 2，则：
fw_setenv boot_part 1 && reboot

closed 2021-06-20 by coolsnowwolf

dunkeng, 2021-06-20:
@coolsnowwolf 我觉得如果不解决这个问题，可以把"advanced_reboot"这个功能删除了，用户量应该不大，没几个人提出来，而且我们也都会手工切换两个分区