coolsnowwolf/lede #5441: ipq40XX无法自动挂载exfat文件，缺少libblkid，可道云无法安装升级，默认缺少unzip

yinzhefuyun, 2020-09-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

ipq40XX无法自动挂载exfat文件，缺少libblkid，可道云无法安装升级，默认缺少unzip

### (2) 路由器型号和固件版本

r619ac-128mb

### (3) 详细日志

A：

FusionPlmH, 2020-09-11:
自己解決就好，不用提交issues

yinzhefuyun, 2020-09-11:
> 自己解決就好，不用提交issues

主要是我不会写pr，所以在这里写下方法，有能力的可以写个pr让lean合并

AmadeusGhost, 2020-09-11:
发 pr 这个其实百度下就知道怎么操作了

yinzhefuyun, 2020-09-11:
> 发 pr 这个其实百度下就知道怎么操作了

问题是怎么写改代码才可以加上去这两个依赖

EnnawYang, 2020-09-11:
> > 发 pr 这个其实百度下就知道怎么操作了
> 
> 问题是怎么写改代码才可以加上去这两个依赖

找个Makefile照抄即可

AmadeusGhost, 2020-09-11:
> 问题是怎么写改代码才可以加上去这两个依赖

你 clone 你自己 fork 的库到本地，配置好 push 环境，然后我直接远程

closed 2020-09-15 by coolsnowwolf

AmadeusGhost, 2020-09-15:
> 要不你也pr下libblkid吧，要不然即使安装了kmod-fs-exfat，也是无法自动挂载的

安装了就挂载正常了嘛？我是打算等有时间的时候一起看下挂载乱码是什么问题。

yinzhefuyun, 2020-09-15:
> > 要不你也pr下libblkid吧，要不然即使安装了kmod-fs-exfat，也是无法自动挂载的
> 
> 安装了就挂载正常了嘛？我是打算等有时间的时候一起看下挂载乱码是什么问题。

是的，那天我在群里面问的，因为没安装这个所以exfat的U盘的uuid无法获取，然后安装了libblkid重启后exfatU盘直接挂载，这个还是LGA1150说的，最开始不是在群里面讨论很久