coolsnowwolf/lede #5450: 9月12日源码编译错误

panther706, 2020-09-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

9月12日源码编译错误

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
WNDR4300/OpenWrt R20.8.27 / LuCI Master (git-20.223.26773-d18ef13)
A：

### (3) 详细日志

A：+ cut -d= -f2 src/chrome/VERSION
+ + tr \n .
cut -d. -f1-4
+ have_version=85.0.4183.83
+ cat CHROMIUM_VERSION
+ want_version=85.0.4183.83
+ [ 85.0.4183.83 = 85.0.4183.83 ]
+ exit 0
Archive:  /home/pwh/lede/dl/gn-git_revision:7d7e8deea36d126397bda2cf924682504271f0e1.zip
  inflating: gn/out/gn               
  inflating: gn/out/.cipdpkg/manifest.json  
ERROR at //skia/BUILD.gn:817:5: Assertion failed.
    assert(false, "Need to port cpu specific stuff from skia_library_opts.gyp")
    ^-----
Need to port cpu specific stuff from skia_library_opts.gyp
See //BUILD.gn:80:5: which caused the file to be included.
    "//skia:skia_unittests",
    ^----------------------
ninja: Entering directory `out/Release'
ninja: error: loading 'build.ninja': No such file or directory
Makefile:85: recipe for target '/home/pwh/lede/build_dir/target-mips_24kc_musl/naiveproxy-85.0.4183.83-3/.built' failed
make[3]: *** [/home/pwh/lede/build_dir/target-mips_24kc_musl/naiveproxy-85.0.4183.83-3/.built] Error 1
make[3]: Leaving directory '/home/pwh/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#8.33#12.54#134.36
package/Makefile:111: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[2]: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/pwh/lede'
package/Makefile:107: recipe for target '/home/pwh/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/pwh/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pwh/lede'
/home/pwh/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

sgllszr, 2020-09-12:
sudo apt-get update
sudo apt install curl

joketans, 2020-09-13:
同编译错误,x86 软路由。
target/Makefile:19: recipe for target '/home/test/lede/staging_dir/target-x86_64_musl/stamp/.target_install' failed
make[1]: *** [/home/test/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/test/lede'
/home/test/lede/include/toplevel.mk:222: recipe for target 'world' failed

joketans, 2020-09-13:
> sudo apt-get update
> sudo apt install curl

试过了，还是这个错误

panther706, 2020-09-14:
gn/out/.cipdpkg/manifest.json
ERROR at //skia/BUILD.gn:817:5: Assertion failed.
assert(false, "Need to port cpu specific stuff from skia_library_opts.gyp")
^-----
Need to port cpu specific stuff from skia_library_opts.gyp
See //BUILD.gn:80:5: which caused the file to be included.
"//skia:skia_unittests",

应该是这里出问题，
不是sudo apt-get update
sudo apt install curl这个问题

x-magic, 2020-09-14:
A library required by naiveproxy cannot be compiled for mips cpu. Apparently they haven't implemented it yet. 

Remove naiveproxy and its luci app should solve the problem.

closed 2020-09-15 by coolsnowwolf