coolsnowwolf/lede #5452: openwrt仓库编译N1固件报错

bin20088, 2020-09-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：openwrt仓库使用bcm27xx_bcm2710_DEVICE_rpi-3架构编译N1固件报错，日志在下面。编译X64没问题

### (2) 路由器型号和固件版本

A：N1，9月12日开始报错的，之前的版本没问题

### (3) 详细日志

A：
[image]

sgllszr, 2020-09-13:
缺少curl依赖安装就好了。我也这个问题昨天搞了很久
sudo apt-get update
sudo apt install curl

bin20088, 2020-09-13:
> 缺少curl依赖安装就好了。我也这个问题昨天搞了很久
> sudo apt-get update
> sudo apt install curl

不应该，看这报错跟curl并没关系，而且我的curl应该是有的

EnnawYang, 2020-09-13:
> 缺少curl依赖安装就好了。我也这个问题昨天搞了很久
> sudo apt-get update
> sudo apt install curl

您真的看过他的错误日志了么？

joketans, 2020-09-13:
同编译错误。
target/Makefile:19: recipe for target '/home/test/lede/staging_dir/target-x86_64_musl/stamp/.target_install' failed
make[1]: *** [/home/test/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/test/lede'
/home/test/lede/include/toplevel.mk:222: recipe for target 'world' failed

afeiy, 2020-09-14:
实测通过。
[image]
[image]

bin20088, 2020-09-14:
> 实测通过。
> [image]
> [image]

也是醉了，都不看上面报错的日志吗？你不选报错的插件当然没问题，在这说个毛线

afeiy, 2020-09-14:
看你口气本来不想回了，耐不住就只多说一句：
什么插件哦？取消ntfs文件系统不就可以了吗？

bin20088, 2020-09-14:
> 看你口气本来不想回了，耐不住就只多说一句：
> 什么插件哦？取消ntfs文件系统不就可以了吗？

我更醉了，我都说了你没选报错的插件当然没问题，言外之意不就是不选就ok了，这还用你说？

closed 2021-06-20 by coolsnowwolf