coolsnowwolf/lede #5453: ipsec vpn 仅访问内网，不转发外网流量

xiaozhuai, 2020-09-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

在公司的电脑连接家里软路由的ipsec vpn之后，公司的电脑外网流量也会走家里的路由器，实际上我只需要能够访问家里的内网就行了，外网还是正常走公司的网络，该怎么配置呢？

qq1316237299, 2020-09-13:
这个内核就这样

xiaozhuai, 2020-09-14:
@qq1316237299 有什么方法可以解决这个问题吗？

StereoApp, 2020-09-15:
> @qq1316237299 有什么方法可以解决这个问题吗？

这个和路由器应该无关，在macOS中（暂时手头没有windows，我相信都是类似的），VPN的高级选项中有一个"通过VPN连接转发所有流量"，关闭这个选项，然后通过配置静态路由把家中的网段路由到VPN的，我的配置是这样的
```
/sbin/route add 192.168.50.0/24 -iface ppp0
```
windows我没有配置过，但是大概思路是一样的

xiaozhuai, 2020-09-15:
1. ”通过VPN连接转发所有流量“ 这个选项只对 L2TP/IPSEC 有效，Cisco IPSEC 没这个选项
2. 手动加路由表的方法比较脏，污染了客户端电脑的环境。
3. 根据苹果提供的文档来看，Cisco IPSEC不支持”通过VPN连接转发所有流量“选项的原因是，Cisco IPSEC 是否转发所有流量由客户端和服务端协商，见图，所以这个应该是服务端可以配置的，google了好久，没找到资料，所以来问下有没有遇到同样问题的小伙伴

[image]

@coolsnowwolf 另外感谢lean大亲自关注了这个问题！

pupboss, 2020-09-15:
你这个问题基本上无解，只能在本地改转发路由，家里内网地址的转发到 VPN 网关，无非是手动加还是写个脚本自动加的区别，企业里也是这么干的，四楼说的对

下面是 VPN 连接日志，我在家连上之后，学校的内网走 VPN，其他还是直连，基本上只能这么做

```
add net 10.88.0.0: gateway 10.22.35.174
add net 10.84.0.0: gateway 10.22.35.174
add net 10.80.0.0: gateway 10.22.35.174
add net 172.31.0.0: gateway 10.22.35.174
add net 172.22.0.0: gateway 10.22.35.174
add net 172.21.0.0: gateway 10.22.35.174
add net 172.20.0.0: gateway 10.22.35.174
add net 172.19.0.0: gateway 10.22.35.174
add net 155.69.0.0: gateway 10.22.35.174
```

skill7899, 2020-09-23:
同等解决方案~一样是macos

goer99, 2020-09-25:
决定本机访问各个ip段走哪条路由，是本机路由表负责的，和远端ipsec vpn服务没关系。最简单的解决办法是在本机配置静态路由，简单直接。

若一定要通过远端vpn去配置本机路由表，这个需要本机运行有相应功能的客户端，这个有点累。openvpn貌似有这个功能，也需要本机做配置。非要ipsec去实现的话，太累了吧。

所谓“手动加路由表的方法比较脏，污染了客户端电脑的环境”，这个就有点意思了。不管是手动还是自动，本机必须要有路由表，才能实现想要的访问模式。路由表你是绕不开的。

goer99, 2020-09-25:
openvpn可以做到远端配置本机路由，要ipsec实现这个功能有点舍近求远，而且太累了。

本机指定路由不香吗？本机路由表你绕不过去的。

xiaozhuai, 2020-09-25:
@StereoApp @skill7899 尝试修改路由表没有成功
连接iPsec之前路由器是这样的

```
Routing tables

Internet:
Destination        Gateway            Flags        Refs      Use   Netif Expire
default            192.168.3.1        UGSc          121        2     en7
default            link#22            UCSI            1        0 bridge1      !
10.211.55/24       link#23            UC              2        0   vnic0      !
10.211.55.255      ff:ff:ff:ff:ff:ff  UHLWbI          0       17   vnic0      !
127                127.0.0.1          UCS             0        0     lo0
127.0.0.1          127.0.0.1          UH             47 28061370     lo0
169.254            link#8             UCS            11        0     en7      !
169.254            link#24            UCSI            0        0   vnic1      !
169.254.38.2       98:22:ef:c8:5b:79  UHLSW           0        0     en7      !
169.254.60.244     d2:15:2c:8e:a3:8f  UHLSW           0        1     en7      !
169.254.143.147    e0:b5:5f:f2:de:87  UHLSW           0        0     en7      !
169.254.151.193    link#8             UHLSW           0        1     en7      !
169.254.203.195    7c:7a:91:9:b9:fb   UHLSW           0        0     en7      !
169.254.222.82     f0:18:98:50:30:da  UHLSW           0        0     en7      !
```
连接之后路由表是这样的

```
Routing tables

Internet:
Destination        Gateway            Flags        Refs      Use   Netif Expire
default            link#26            UCS           118        0   utun2
default            192.168.3.1        UGScI           5        0     en7
default            link#22            UCSI            1        0 bridge1      !
10.0.0.1           link#26            UHWIi          21       44   utun2
10.0.0.129         10.0.0.129         UH              1       11   utun2
10.211.55/24       link#23            UC              2        0   vnic0      !
10.211.55.255      ff:ff:ff:ff:ff:ff  UHLWbI          0       37   vnic0      !
14.22.3.35         link#26            UHW3I           0        4   utun2     10
```

en7这个网卡是物理网卡，utun0是iPsec，vpn的子网是 `10.0.0.1/24`

我的想法是
1. 先删掉 `default            link#26            UCS           118        0   utun2` 这个
2. 加一条 `10.0.0.1/24 10.0.0.1 utun2`

第一步，执行`sudo route delete default -iface utun2` 然后重新查看路由表确实删掉了，但是彻底没网了...
感觉不用走第二步了...不知道哪里出了问题...

xiaozhuai, 2020-09-25:
ok了，已解决，分四步，再连接vpn后执行

```
sudo route delete default                    # 删除iPsec添加的默认路由
sudo route delete default -ifscope en7       # 删除已经没用了的en7  (添加了RTF_IFSCOPE，不太清楚是干嘛的)
sudo route add default 192.168.3.1           # 重新添加default路由    (公司的路由器地址是192.168.3.1)
sudo route add 10.0.0.1/24 -iface utun2      # 添加家里子网的路由     (家里的子网是10.0.0.1/24)
```

还有一件事情就是修改DNS服务器，连上vpn后，dns会被覆盖成10.0.0.1，这个在网络偏好设置里修改。不知道有没有办法能不覆盖本地的dns server

每次重连vpn都需要重新执行上面的4行命令，而且这几行命令也不具备普适性，如果换成在WiFi环境下，或者换个地方连接VPN，这些命令都可能需要变动，总之感觉并不优雅。

抛转引玉，还有更好的方法吗？

goer99, 2020-09-26:
写个脚本/批处理文件呗，每次连接的时候运行一下。很简单也挺优雅。

这个话题就不是issue，而是定制化功能需求。如果只是少数人需求的话，开发的性价比太低，而且还有其他替代软件已有类似功能。自己能写个脚本、批处理文件等动动手指就能解决的问题，若非要变成一个功能去开发，给开源开发者增加工作量不是很值得。

定制化需求多了，都这么提需求，且没人为这些个性化需求付费，开源开发者要累死了。优雅不是让别人工作而自己享受其便利。

xiaozhuai, 2020-09-27:
@goer99     : )

StereoApp, 2020-09-27:
L2TP/IPSEC 可以编辑 /etc/ppp/ip-up 实现启动自动运行脚本，不知道Cisco IPSEC是否可以。

xiaozhuai, 2020-09-27:
@StereoApp 简单看了下 `/etc/ppp/ip-up`，确实很优雅，我试试Cicso IPSEC是否可用。

StereoApp, 2020-09-27:
> @StereoApp 简单看了下 `/etc/ppp/ip-up`，确实很优雅，我试试Cicso IPSEC是否可用。

查了一下应该是不行，但是这个方案看起来似乎可行，https://blog.sycx.me/blog/2012/12/21/cisco-ipsec-vpn-automatic-routing-table/

xiaozhuai, 2020-09-27:
@StereoApp racoon的配置文件是动态创建的，钩子需要加到动态创建的配置文件里，`/var/run/racoon/YOUR_VPN_SERVER_IP.conf`，最好的方法是在全局配置里加钩子，我尝试一下看看。
```
script "/etc/ppp/ip-up" phase1_up;	
script "/etc/ppp/ip-down" phase1_down;
```

xiaozhuai, 2020-09-27:
@StereoApp 
```
             script script phase1_up
             script script phase1_down
             script script phase1_dead
                     Shell scripts that get executed when a phase 1 SA goes up or down, or when
                     it is detected as dead by DPD.  All scripts get either phase1_up ,
                     phase1_down or phase1_dead as first argument, and the following variables
                     are set in their environment:
                     LOCAL_ADDR
                             The local address of the phase 1 SA.
                     LOCAL_PORT
                             The local port used for IKE for the phase 1 SA.
                     REMOTE_ADDR
                             The remote address of the phase 1 SA.
                     REMOTE_PORT
                             The remote port used for IKE for the phase 1 SA.
                     REMOTE_ID
                             The remote identity received in IKE for the phase 1 SA.
                     The following variables are only set if mode_cfg was enabled:
                     INTERNAL_ADDR4
                             An IPv4 internal address obtained by ISAKMP mode config.
                     INTERNAL_NETMASK4
                             An IPv4 internal netmask obtained by ISAKMP mode config.
                     INTERNAL_CIDR4
                             An IPv4 internal netmask obtained by ISAKMP mode config, in CIDR
                             notation.
                     INTERNAL_DNS4
                             The first internal DNS server IPv4 address obtained by ISAKMP mode
                             config.
                     INTERNAL_DNS4_LIST
                             A list of internal DNS servers IPv4 address obtained by ISAKMP mode
                             config, separated by spaces.
                     INTERNAL_WINS4
                             The first internal WINS server IPv4 address obtained by ISAKMP mode
                             config.
                     INTERNAL_WINS4_LIST
                             A list of internal WINS servers IPv4 address obtained by ISAKMP mode
                             config, separated by spaces.
                     SPLIT_INCLUDE
                             The space separated list of IPv4 addresses and masks (address slash
                             mask) that define the networks to be encrypted (as opposed to the
                             default where all the traffic should be encrypted) ; obtained by
                             ISAKMP mode config ; SPLIT_INCLUDE and SPLIT_LOCAL are mutually
                             exclusive.
                     SPLIT_LOCAL
                             The space separated list of IPv4 addresses and masks (address slash
                             mask) that define the networks to be considered local, and thus
                             excluded from the tunnels ; obtained by ISAKMP mode config.
                     SPLIT_INCLUDE_CIDR
                             Same as SPLIT_INCLUDE, with netmasks in CIDR notation.
                     SPLIT_LOCAL_CIDR
                             Same as SPLIT_LOCAL, with netmasks in CIDR notation.
                     DEFAULT_DOMAIN
                             The DNS default domain name obtained by ISAKMP mode config.
```
关于racoon的配置，钩子要写到Remote Nodes内，Remote Nodes是动态生成的

xiaozhuai, 2020-09-27:
```
#!/usr/bin/env bash

VPN_INTERFACE=utun2
VPN_SUBNET=10.0.0.1/24
NET_INTERFACE=$(netstat -nrf inet | grep default | egrep "en\d+\s*$" | awk '{print $6}')
NET_GATEWAY=$(netstat -nrf inet | grep default | egrep "en\d+\s*$" | awk '{print $2}')

sudo route delete default
sudo route delete default -ifscope $NET_INTERFACE
sudo route add default $NET_GATEWAY
sudo route add $VPN_SUBNET -iface $VPN_INTERFACE
```

稍微修改了下，使用上面的脚本来修改路由表，可以自动识别出当前的网卡和网关

goer99, 2020-09-27:
使用了issue反馈模板，但你提的这个是issue？

请看清楚第一句：反馈bug/问题模板，提建议请删除

goer99, 2020-09-27:
> 实际上如果iPsec server下发路由规则的时候不要修改default，只加一条 `10.0.0.1/24` 的规则，这样是最舒适的，不知道能不能实现。

上面这行是啥意思呢？要实现这句话，是否需要开源开发工作量？

呵呵

xiaozhuai, 2020-09-27:
@goer99 别逗了 不知道的以为你为开源社区做了多大贡献 前面还有一句”抛砖引玉“ 看到了吗？我要求过任何人做什么吗？还是那句话，我问你了吗？找啥存在感

goer99, 2020-09-27:
呵呵。不要转移话题。你说那话是啥意思呢？还不让人评论了？这是你家自留地？你的脾气有点大呀。

我有没有贡献，和这件事没关系。你想要自己舒服，却会给别人带来一大堆工作量。我看不惯当然要说话。

况且我也给你提了解决办法。

xiaozhuai, 2020-09-27:
@goer99 你提啥办法了？换OpenVPN还是写脚本？这个需要你来抖机灵？上来就优雅给别人带来工作量，我向谁提什么要求了？这就是个技术方案讨论的贴，你这么喜欢站在高点批判？开源社区都这么交流？

goer99, 2020-09-27:
得，玻璃心炸了。

懒得回复了。不值当。

xiaozhuai, 2020-09-27:
@goer99 得了，谢谢您！

skill7899, 2020-09-29:
> ```
> #!/usr/bin/env bash
>
> VPN_INTERFACE=utun2
> VPN_SUBNET=10.0.0.1/24
> NET_INTERFACE=$(netstat -nrf inet | grep default | egrep "en\d+\s*$" | awk '{print $6}')
> NET_GATEWAY=$(netstat -nrf inet | grep [...]

上面的优雅方案，根据事件是不是实现不了？这个脚本每次都要连接执行一次有点麻烦

xiaozhuai, 2020-09-29:
@skill7899 目前没找到很好的办法可以勾住Cicso IPSEC连接事件，由于Remote Nodes配置是动态生成的，但钩子又必须加在这里面，这就有冲突了。
如果服务端IP是固定的，可以将生成的配置拷贝一份 加上钩子，见 @StereoApp 提供的 [https://blog.sycx.me/blog/2012/12/21/cisco-ipsec-vpn-automatic-routing-table/](https://blog.sycx.me/blog/2012/12/21/cisco-ipsec-vpn-automatic-routing-table/) 这里的方法。
我的服务端ip是动态的，配合ddns使用的，所以不适用这个方法

skill7899, 2020-09-29:
> @skill7899 目前没找到很好的办法可以勾住Cicso IPSEC连接事件，由于Remote Nodes配置是动态生成的，但钩子又必须加在这里面，这就有冲突了。
> 如果服务端IP是固定的，可以将生成的配置拷贝一份 加上钩子，见 @StereoApp 提供的 https://blog.sycx.me/blog/2012/12/21/cisco-ipsec-vpn-automatic-routing-table/ 这里的方法。
> 我的服务端ip是动态的，配合ddns使用的，所以不适用这个方法

明白了，我也是ip动态的。我刚看了一下里面生成的文件，还记录了本地的ip地址，要是换了网络还是不行的。

Zxilly, 2020-10-20:
> 得，玻璃心炸了。
> 
> 懒得回复了。不值当。

58个源全是fork的，你装啥呢？
一年的commit还没我一天多，不知道以为多hardcore呢

Zack0617, 2020-10-25:
我用了下面这个ipsec源码，可以正常使用。我自己用官方的19.07+ssr-plus+ipsec使用一直很稳定，大家可以自己去尝试一下。
https://github.com/xiaorouji/openwrt-package/tree/master/lienol/luci-app-ipsec-vpnserver-manyusers

qq1316237299, 2020-10-25:
是用官方的源码加这个链接的ipsec吗。还是lean的源码和这个连接的ipsec

 

 https://github.com/xiaorouji/openwrt-package/tree/master/lienol/luci-app-ipsec-vpnserver-manyusers

Zack0617, 2020-10-25:
> 是用官方的源码加这个链接的ipsec吗。还是lean的源码和这个连接的ipsec
> […](#)

我用的官方源，加了链接中这个ipsec和lean的ssr-plus

qq1316237299, 2020-10-25:
牛逼

qwgaan, 2020-10-29:
其实为啥不用openvpn呢？ipsec用起来确实需要各种路由的配合，我现在就是公司电脑自动拨openvpn回家里的openwrt，然后wrt通过openvpn下发路由表过来，这样也不影响dns，互通起来也方便，目前通过这个openvpn互联来同步家里的nas，感觉还是很方便的

goer99, 2020-10-31:
> > 得，玻璃心炸了。
> > 懒得回复了。不值当。
> 
> 58个源全是fork的，你装啥呢？
> 一年的commit还没我一天多，不知道以为多hardcore呢

装？用的着装么？我根本不是程序员。

但是我给用过的每一个有价值的开源项目付费支持，我理解优秀开源开发者付出的价值，并且不希望他们被无聊的功能需求拖跨。重复造轮子有必要么？lean对这个项目长久独力支撑时的疲惫你知不知道？

为了尽量少麻烦程序员，我还强迫自己自学了一点shell什么的，这样的话至少自己还能写个脚本而不用找人帮忙。

有些程序员水平不一定咋地，脾气大的惊人。要是能有linus那么牛，或许也可以任性一下。问题是，人家linus怼人虽然猛，可还是讲道理的。别好的没学到，坏毛病学了个全。

goer99, 2020-10-31:
> 其实为啥不用openvpn呢？ipsec用起来确实需要各种路由的配合，我现在就是公司电脑自动拨openvpn回家里的openwrt，然后wrt通过openvpn下发路由表过来，这样也不影响dns，互通起来也方便，目前通过这个openvpn互联来同步家里的nas，感觉还是很方便的

同样使用openvpn在办公室和家里各自在一个路由器上双向打通。ipsecvpn另有其他妙用。

同样不理解，有现成更好用的解决方案而不用，非要再实现一遍，ipsec有啥openvpn比不上的好处？我是不太明白。

goer99, 2020-10-31:
说说为什么我会在这个话题下回复。

不是程序员，不太经常上github，只是有需要的时候才上来一下。某天上来看lean项目，发现这个置顶issue，有关ipsec vpn的，以为有什么bug，因为一直也在使用ipsec vpn，所以赶紧来看看。

来了才发现并不是bug，而是一个定制化需求。由于有过相关使用经验，所以才回复了。没成想有些程序员的反应挺激烈。

从优秀的开源项目中也得益不少，对他们的奉献很感谢，同时也发现不少优秀的开源开发者维护优秀项目时所付出的时间和精力巨大。我的初衷是要尊重这些优秀开发者，他们的时间很宝贵，希望能用到对更多人有意义的地方。

当然，如果您愿意花时间开发小众需求，那是您乐意，但是，别开issue，开需求和建议好不好。这个issue的标题会让正常使用ipsec vpn功能的人受一次虚惊。

StereoApp, 2020-10-31:
> 说说为什么我会在这个话题下回复。
>
> 不是程序员，不太经常上github，只是有需要的时候才上来一下。某天上来看lean项目，发现这个置顶issue，有关ipsec vpn的，以为有什么bug，因为一直也在使用ipsec vpn，所以赶紧来看看。
>
> 来了才发现并不是bug，而是一个定制化需求。由于有过相关使用经验，所以才回复了。没成想有些程序员的反应挺激烈。 [...]

求求你别来秀下限了好吗，“还受了一次虚惊”，之前实在是不想和你探讨，可是你没完没了的在这里纠缠，首先就问题本身来说，IPSec协议是应该由服务端下发默认路由的，但是看起来是存在一些问题，确实脚本可以解决，但是不是谁都会写脚本的，有人能看到这个issue有了解决思路不好吗，其次开源社区也不是你理解的那样的，乱开issue的问题确实存在，但是这个issue是有意义的，浏览整个issue的回复，有人提出问题，有人提出方案，有人表示同样有问题，只有你在这不停的杠，你说发现这个置顶issue，谁pin的这个issue？不是作者吗，如果是无意义的为什么不直接关掉，作者pin了这个issue，让你受了“虚惊”，你这心理承受力要不就别来看issue了，不然，1、容易受惊吓。2、不是程序员，对于真正bug不能贡献。3、对于非bug的issue嗤之以鼻。莫非你是特派的开源项目issue巡查员，那真的是冒犯了啊。

goer99, 2020-10-31:
> > 说说为什么我会在这个话题下回复。
> > 不是程序员，不太经常上github，只是有需要的时候才上来一下。某天上来看lean项目，发现这个置顶issue，有关ipsec vpn的，以为有什么bug，因为一直也在使用ipsec vpn，所以赶紧来看看。
> > 来了才发现并不是bug，而是一个定制化需求。由于有过相关使用经验，所以才回复了。没成想有些程序员的反应挺激烈。
> > 从优秀的开源项目中也得益不少，对他们的奉献很感谢，同时也发现不少优秀的开源开发者维护优秀项目时所付出的时间和精力巨大。我的初衷是要尊重这些优秀开发者，他们的时间很宝贵，希望能用到对更多人有意义的地方。
> > 当然，如果您愿意花时间开发小众需求，那是您乐意，但是，别开issue，开需求和建议好不好。这个issue的标题会让正常使用ipsec vpn功能的人受一次虚惊。
> 
> 求求你别来秀下限了好吗，“还受了一次虚惊”，之前实在是不想和你探讨，可是你没完没了的在这里纠缠，首先就问题本身来说，IPSec协议是应该由服务端下发默认路由的，但是看起来是存在一些问题，确实脚本可以解决，但是不是谁都会写脚本的，有人能看到这个issue有了解决思路不好吗，其次开源社区也不是你理解的那样的，乱开issue的问题确实存在，但是这个issue是有意义的，浏览整个issue的回复，有人提出问题，有人提出方案，有人表示同样有问题，只有你在这不停的杠，你说发现这个置顶issue，谁pin的这个issue？不是作者吗，如果是无意义的为什么不直接关掉，作者pin了这个issue，让你受了“虚惊”，你这心理承受力要不就别来看issue了，不然，1、容易受惊吓。2、不是程序员，对于真正bug不能贡献。3、对于你认为非bug的issue嗤之以鼻。莫非你是特派的开源项目issue巡查员，那真的是冒犯了啊。

好呀。期待你们能把lede里带的ipsec vpn实现出企业级功能，能下发路由、做各种策略等等，造福用户。加油。

xiaozhuai, 2020-11-02:
`issue巡查员` 这个词很棒，求你 @goer99 去别处巡查吧，这里没人愿意看你的LJ回复。

BraveSail, 2020-11-20:
zerotier解君愁，家庭内部设备全部用内网访问，不会走外网。
[image]

goer99, 2020-12-05:
> `issue巡查员` 这个词很棒，求你 @goer99 去别处巡查吧，这里没人愿意看你的LJ回复。

本来都不想理你们这些低能儿了，你们还一起上脸了，动不动就满嘴喷粪搞人身攻击。

@xiaozhuai 你们先搞出来“优雅的”解决方案，再来喷粪吧。如果搞不出来，就该叫你们自己LJ了。

按照LINUS的说法，talk is cheap, show me the code.

一堆脑残估计连ipsec文档估计都没读过，上来就知道喷，抢着来显示自己的素质低吗？

我看你们怎么搞出来解决方案，搞不出来，还随便喷人，你们连当垃圾的资格都不配。

xiaozhuai, 2020-12-05:
@goer99 别尼玛装了，求你了，除了脏词，你会啥？自圆其说都不能，还talk is cheap？上面除了你，其他大部分人都给了具体方案，只有你在充当巡查员的角色，你自己的话才是最廉价的，还不自知？还issue和requirement都分不清，那也比你一个只会star和fork，把这里当朋友圈的强，你分得清个啥？这么多人踩你，你还有脸回来喷？脸皮这么厚，建议去研制防弹衣。最后一次回复，没人这么闲在这和个傻子对线。

goer99, 2020-12-06:
> @goer99 别尼玛装了，求你了，除了脏词，你会啥？自圆其说都不能，还talk is cheap？上面除了你，其他大部分人都给了具体方案，只有你在充当巡查员的角色，你自己的话才是最廉价的，还不自知？还issue和requirement都分不清，那也比你一个只会star和fork，把这里当朋友圈的强，你分得清个啥？这么多人踩你，你还有脸回来喷？脸皮这么厚，建议去研制防弹衣。最后一次回复，没人这么闲在这和个傻子对线。

@xiaozhuai 睁开你的狗眼看看，是哪条狗先豹粗口的？十足的垃圾。。。

再睁开你的狗眼好好看看，使用脚本、使用其他vpn软件的解决方案，是谁第一个提出的。

我倒是要看看，你们不用脚本、不用其他vpn软件，不大幅修改ipsec服务器代码和客户端代码，怎么“优雅”地实现ipsec的路由功能。

有本事你做出来啊。做不出来还敢喷，不是垃圾就是狗屎。

closed 2021-06-20 by coolsnowwolf