coolsnowwolf/lede #5465: V2升级4.28.1需要服务器和客户端同时升级

hzaxi, 2020-09-15:
如果不是自己控制的服务器建议谨慎升级。

coolsnowwolf, 2020-09-15:
那还是先回退吧，不然会出问题

niphor, 2020-09-15:
请问 不升级会有什么问题？

现在 客户端是 4.26的 服务器是 4.28.1的
连是能正常连的...

XiaoliChan, 2020-09-15:
Well, my "Mystery Network“ did not work anymore after commit #5447 .

Itxiaoniao, 2020-09-15:
alterID改一下就可以，0改成1，或者1改成0，就能用了

niphor, 2020-09-15:
推上的，我引用了一下

> 有一些人产生了困惑，简单答疑：
>
>如果你两边都能升级到最新版，你不需要关注这个事情；
>如果你只能升级客户端，那就不要用 alterid 0

hzaxi, 2020-09-16:
> alterID改一下就可以，0改成1，或者1改成0，就能用了

这个看服务器端配置的，有些不能随便改。

Rangtian, 2020-09-17:
@coolsnowwolf 大佬，别退回呀。没法保持同步的人将alterID设为非0即可，不存在无法使用的问题。可咱们这些等着用AEAD，等着测试新版VLESS的人，眼巴巴盼更新呢。

jyy5206, 2020-09-17:
难怪前天起新编译的V2ray用不了了

我现在客户端是4.28了
可是服务器端是机场啊，我改不了啊。。。

我想重新编译
可是如何在重新编译时
设定V2ray回退到老版本4.27呢？
求各位大神指点

谢谢！

MrSeven590, 2020-09-17:
> 难怪前天起新编译的V2ray用不了了
> 
> 我现在客户端是4.28了
> 可是服务器端是机场啊，我改不了啊。。。
> 
> 我想重新编译
> 可是如何在重新编译时
> 设定V2ray回退到老版本4.27呢？
> 求各位大神指点
> 
> 谢谢！

参照readme里的方法，先git pull，然后重新编译，因为最新的源码已经退回4.27了

jyy5206, 2020-09-17:
好的
谢谢！

tufu9441, 2020-09-21:
那么如何自行升级到4.28呢？

31415926pai, 2020-09-22:
不用退回的，我刚才试了一下。
alterID改一下就可以，0改成1就好了。
机场是老版本的，SS是新版本的。改一下就正常了。如果节点多，经常更新是有些麻烦。

nftbty, 2020-09-23:
还是应该升级到新版本的，V2Ray最近的更新一直都是针对安全性方面的，还是跟进比较好。至于说机场服务器端是老版本，麻烦去仔细看下从4.27.x以来的每个版本的更新日志。赞同楼上的，alterid不设0的话，跟以前老版本是完全兼容的，只有alterid=0才会启用AEAD新特性。至于机场多个节点，既没升级AEAD，还把alterid设置0，如此不安全的做法，这样的机场是不太敢用的，不如去用AEAD的ss节点。

XiaoliChan, 2020-10-04:
Change alterid from 0 to 1 did not work !

Env: Client with v2ray in version 4.29

dunkeng, 2020-10-20:
https://github.com/fw876/helloworld/issues/216

4.28.2以上版本客户端可以通过增加export V2RAY_VMESS_AEAD_DISABLED=true来兼容旧版本的服务端

closed 2021-06-20 by coolsnowwolf