coolsnowwolf/lede #5466: 选上naiveproxy插件时就编译出错

lqhorochi, 2020-09-15:
echo "dnsmasq-full" >> /home/kenneth/lede/staging_dir/target-x86_64_musl/pkginfo/dnsmasq.full.install
make[3]: Leaving directory '/home/kenneth/lede/package/network/services/dnsmasq'
time: package/network/services/dnsmasq/full/compile#0.11#0.03#0.32
make[3]: Entering directory '/home/kenneth/lede/package/utils/lua'
make[3]: Leaving directory '/home/kenneth/lede/package/utils/lua'
time: package/utils/lua/host-compile#0.12#0.01#0.17
make[3]: Entering directory '/home/kenneth/lede/feeds/packages/devel/ninja'
make[3]: Leaving directory '/home/kenneth/lede/feeds/packages/devel/ninja'
time: package/feeds/packages/ninja/host-compile#0.11#0.01#0.20
make[3]: Entering directory '/home/kenneth/lede/feeds/packages/libs/nspr'
mkdir -p /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/kenneth/lede/tmp/.root-copy.flock -c 'cp -fpR /home/kenneth/lede/build_dir/target-x86_64_musl/nspr-4.25/.pkgdir/nspr/. /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.nspr_installed
echo "nspr" >> /home/kenneth/lede/staging_dir/target-x86_64_musl/pkginfo/nspr.default.install
make[3]: Leaving directory '/home/kenneth/lede/feeds/packages/libs/nspr'
time: package/feeds/packages/nspr/compile#0.14#0.04#0.57
make[3]: Entering directory '/home/kenneth/lede/feeds/packages/libs/sqlite3'
mkdir -p /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '0' | cmp -s - /home/kenneth/lede/staging_dir/target-x86_64_musl/pkginfo/libsqlite3.version || echo '0' > /home/kenneth/lede/staging_dir/target-x86_64_musl/pkginfo/libsqlite3.version
SHELL= flock /home/kenneth/lede/tmp/.root-copy.flock -c 'cp -fpR /home/kenneth/lede/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3/. /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libsqlite3_installed
echo "libsqlite3" >> /home/kenneth/lede/staging_dir/target-x86_64_musl/pkginfo/sqlite3.default.install
make[3]: Leaving directory '/home/kenneth/lede/feeds/packages/libs/sqlite3'
time: package/feeds/packages/sqlite3/compile#0.13#0.03#0.34
make[3]: Entering directory '/home/kenneth/lede/feeds/packages/libs/nss'
mkdir -p /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/kenneth/lede/tmp/.root-copy.flock -c 'cp -fpR /home/kenneth/lede/build_dir/target-x86_64_musl/nss-3.53/.pkgdir/libnss/. /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/kenneth/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libnss_installed
echo "libnss" >> /home/kenneth/lede/staging_dir/target-x86_64_musl/pkginfo/nss.default.install
make[3]: Leaving directory '/home/kenneth/lede/feeds/packages/libs/nss'
time: package/feeds/packages/nss/compile#0.20#0.11#0.97
make[3]: Entering directory '/home/kenneth/lede/feeds/packages/libs/expat'
make[3]: Leaving directory '/home/kenneth/lede/feeds/packages/libs/expat'
time: package/feeds/packages/expat/host-compile#0.13#0.01#0.25
make[3]: Entering directory '/home/kenneth/lede/feeds/packages/libs/libffi'
make[3]: Leaving directory '/home/kenneth/lede/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/host-compile#0.12#0.02#0.23
make[3]: Entering directory '/home/kenneth/lede/package/utils/bzip2'
make[3]: Leaving directory '/home/kenneth/lede/package/utils/bzip2'
time: package/utils/bzip2/host-compile#0.11#0.01#0.15
make[3]: Entering directory '/home/kenneth/lede/feeds/packages/lang/python/python3'
make[3]: Leaving directory '/home/kenneth/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#0.21#0.07#0.69
make[3]: Entering directory '/home/kenneth/lede/feeds/helloworld/naiveproxy'
rm -f /home/kenneth/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.built
touch /home/kenneth/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.built_check
( cd /home/kenneth/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3 ; ./tools/import-upstream.sh ; . ./init_env.sh "x86_64" "x86" "" "" "/home/kenneth/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl" "/home/kenneth/lede/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/home/kenneth/lede/staging_dir/target-x86_64_musl" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/home/kenneth/lede/staging_dir/hostpkg/bin/python3" ; /home/kenneth/lede/staging_dir/hostpkg/bin/ninja -C "out/Release" naive ; )
+ cut -d. -f1-4
+ tr \n .
+ cut -d= -f2 src/chrome/VERSION
+ have_version=85.0.4183.83
+ cat CHROMIUM_VERSION
+ want_version=85.0.4183.83
+ [ 85.0.4183.83 = 85.0.4183.83 ]
+ exit 0
Done. Made 12712 targets from 2280 files in 42018ms
ninja: Entering directory `out/Release'
[1/1922] CXX obj/base/base_static/base_switches.o
FAILED: obj/base/base_static/base_switches.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/base/base_static/base_switches.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/blacklist.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fno-ident -fdata-sections -ffunction-sections -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/kenneth/lede/staging_dir/target-x86_64_musl/usr/include -I/home/kenneth/lede/staging_dir/target-x86_64_musl/usr/include/nss -I/home/kenneth/lede/staging_dir/target-x86_64_musl/usr/include/nspr -fvisibility-inlines-hidden --target=x86_64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../base/base_switches.cc -o obj/base/base_static/base_switches.o
/bin/sh: 1: ../../third_party/llvm-build/Release+Asserts/bin/clang++: not found
[2/1922] CXX obj/buildtools/third_party/libc++/libc++/memory.o
FAILED: obj/buildtools/third_party/libc++/libc++/memory.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/memory.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=__attribute__\(\(__visibility__\(\"default\"\)\)\) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080 [... cut ...]
make[3]: Leaving directory '/home/kenneth/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#17.54#2.56#45.80
make[2]: *** [package/Makefile:113: package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/kenneth/lede'
make[1]: *** [package/Makefile:107: /home/kenneth/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kenneth/lede'
make: *** [/home/kenneth/lede/include/toplevel.mk:224：world] 错误 2

看了众多反馈,也安装了crul...选上naiveproxy编译就出现错误
是插件问题还是???

pangfeng161, 2020-09-15:
下载naiveproxy放到helloworld目录编译,成功编译

hcym, 2020-09-15:
这个一直没有问题，lienol的还要早，就是不会用

pangfeng161, 2020-09-15:
> 这个一直没有问题，lienol的还要早，就是不会

> 这个一直没有问题，lienol的还要早，就是不会用

这个新得代理软件,很少介绍看来要一段时间才成熟

lqhorochi, 2020-09-15:
> 下载naiveproxy放到helloworld目录编译,成功编译

哪里下载?求告知

pangfeng161, 2020-09-15:
> > 下载naiveproxy放到helloworld目录编译,成功编译
> 
> 哪里下载?求告知
  搜索不会吗?

hcym, 2020-09-16:
代理问题吧，从来没下载什么包，以前不懂还保留dl目录做个软链接，现在梯子好用的话根本不需要，所有源码包五分钟的事，就是go程序麻烦一点，最好是全局

ihuaer, 2020-09-16:
github找了个插件naiveproxy 替换进去，正在重新编译

Dimitri2020007, 2020-09-16:
naiveproxy是要放进lede/feeds/helloworld目录里吗？还是没能编译成功。错误和第一个报错的一样。

以下是make -j1 V=99的报错信息

make[3]: Entering directory '/home/dimitri/lede/feeds/helloworld/naiveproxy'
touch /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.prepared_9503f292df667d6a4caabc6e3b962eb9_6664517399ebbbc92a37c5bb081b5c53_check
. /home/dimitri/lede/include/shell.sh; gzip -dc /home/dimitri/lede/dl/naiveproxy-85.0.4183.83-3.tar.gz | tar -C /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3

Applying ./patches/001-build-add-OpenWrt-staging-libraries.patch using plaintext:
patching file src/build/config/posix/BUILD.gn

Applying ./patches/002-build-add-mips-x86-OpenWrt-toolchain-definitions.patch using plaintext:
patching file src/build/toolchain/linux/BUILD.gn

Applying ./patches/003-build-drop-useless-deps-simd_asm.patch using plaintext:
patching file src/BUILD.gn
Hunk #1 succeeded at 756 (offset 10 lines).
touch /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.prepared_9503f292df667d6a4caabc6e3b962eb9_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.configured_*
rm -f /home/dimitri/lede/staging_dir/target-x86_64_musl/stamp/.naiveproxy_installed
(cd /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/./; if [ -x ./configure ]; then find /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/ -name config.guess | xargs -r chmod u+w; find /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/ -name config.guess | xargs -r -n1 cp --remove-destination /home/dimitri/lede/scripts/config.guess; find /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/ -name config.sub | xargs -r chmod u+w; find /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/ -name config.sub | xargs -r -n1 cp --remove-destination /home/dimitri/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3=naiveproxy-85.0.4183.83-3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3=naiveproxy-85.0.4183.83-3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3=naiveproxy-85.0.4183.83-3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/dimitri/lede/staging_dir/target-x86_64_musl/usr/include -I/home/dimitri/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/dimitri/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/dimitri/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/dimitri/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/dimitri/lede/staging_dir/target-x86_64_musl/lib -L/home/dimitri/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/dimitri/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.built
touch /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.built_check
( cd /home/dimitri/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3 ; ./tools/import-upstream.sh ; . ./init_env.sh "x86_64" "x86" "" "" "/home/dimitri/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl" "/home/dimitri/lede/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/home/dimitri/lede/staging_dir/target-x86_64_musl" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/home/dimitri/lede/staging_dir/hostpkg/bin/python3" ; /home/dimitri/lede/staging_dir/hostpkg/bin/ninja -C "out/Release" naive ; )
+ cut -d= -f2 src/chrome/VERSION
+ + cuttr \n .
 -d. -f1-4
+ have_version=85.0.4183.83
+ cat CHROMIUM_VERSION
[... cut ...]
make[3]: Leaving directory '/home/dimitri/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#14.14#5.63#74.90
make[2]: *** [package/Makefile:113: package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/dimitri/lede'
make[1]: *** [package/Makefile:107: /home/dimitri/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dimitri/lede'
make: *** [/home/dimitri/lede/include/toplevel.mk:224: world] Error 2

lxwq, 2020-09-17:
删除/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/src/chrome/android/profilesafdo.prof这个文件试试,这个文件的版本控制好像有点有问题,只要确认文件存在,就以为是最新版本了,

hcym, 2020-09-17:
不清楚了，我编译是没有什么问题，不光用lean的，把插件包扔到1907.4和master也可以，因为最近用lean的源码会莫名其妙的重启

pangfeng161, 2020-09-18:
> 这个一直没有问题，lienol的还要早，就是不会用

这两天试用感觉比trojan还好用,速度稳定多看来抗干扰目前比其他都强

xingpo, 2020-09-19:
我也是存在这个问题，x86也是一直没编译过去，不选naiveproxy就可以顺利编译！也安装了clang,libnss3等必要软件！前两处错误我根据错误提示手动创建了软链接和目录文件，最后还是没编译过去，最后出错信息如下，我在想是不是我少安装了哪个软件包
Done. Made 12712 targets from 2280 files in 9437ms
ninja: Entering directory `out/Release'
[1/1922] CXX obj/base/base_static/base_switches.o
FAILED: obj/base/base_static/base_switches.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/base/base_static/base_switches.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/blacklist.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fno-ident -fdata-sections -ffunction-sections -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/zyy/compile/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/zyy/compile/openwrt/staging_dir/target-x86_64_musl/usr/include/nss -I/home/zyy/compile/openwrt/staging_dir/target-x86_64_musl/usr/include/nspr -fvisibility-inlines-hidden --target=x86_64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../base/base_switches.cc -o obj/base/base_static/base_switches.o
warning: unknown warning option '-Wno-non-c-typedef-for-linkage'; did you mean '-Wno-return-type-c-linkage'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-max-tokens'; did you mean '-Wno-extra-tokens'? [-Wunknown-warning-option]
error: unable to find plugin 'find-bad-constructs'
2 warnings and 1 error generated.
[2/1922] CXX obj/buildtools/third_party/libc++/libc++/memory.o
FAILED: obj/buildtools/third_party/libc++/libc++/memory.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/memory.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=__attribute__\(\(__visibility__\(\"default\"\)\)\) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/blacklist.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -fstrict-aliasing -fPIC -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversi [... cut ...]
make[2]: *** [Makefile:88: /home/zyy/compile/openwrt/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built] Error 1
make[2]: Leaving directory '/home/zyy/compile/openwrt/feeds/diy1/package/naiveproxy'
time: package/feeds/diy1/naiveproxy/compile#16.20#2.21#14.33
make[1]: *** [package/Makefile:113: package/feeds/diy1/naiveproxy/compile] Error 2
make[1]: Leaving directory '/home/zyy/compile/openwrt'
make: *** [/home/zyy/compile/openwrt/include/toplevel.mk:236: package/feeds/diy1/naiveproxy/compile] Error 2

1715173329, 2020-09-19:
一个不给梯子加钱，一个不给软件升级，hhh

closed 2021-06-20 by coolsnowwolf