coolsnowwolf/lede #5468: SoftEther 在旁路网关下的配置（解决功能不完整的问题）供大家参考

pupboss, 2020-09-15:
众所周知 IPSec 方案最近抽风，有时候 crash 有时候无法连接公网只能访问内网，包括 Lean 也是建议用 SoftEther https://github.com/coolsnowwolf/lede/issues/5262。

在旁路网关方案下，如果你没有改变主路由的 DHCP 策略，那么设备连上 SoftEther VPN 之后也会继承旧的设置，换句话说就是不经过旁路网关。

解决方案就是在 SoftEther VPN 下的虚拟 NAT 功能中，关闭 NAT 但是打开 DHCP，网段和主路由一致，尽可能别冲突，比如虚拟主机 IP 设置 `192.168.1.200`，网关和 DNS 设置为`旁路网关的地址`，DHCP 网段 `192.168.1.210 - 192.168.1.230`，有那么几个可用 IP 就够了。具体 IP 段的选择，可以根据自己家路由器的特性，看看已分配的集中在什么地方，尽可能避开即可。

pupboss, 2020-09-16:
发现这个虚拟 DHCP 功能，会污染到主路由的 DHCP，可能是因为桥接的原因，不建议这么用了。后面会研究一下不同子网的配置方法

closed 2020-09-16 by pupboss

jacywy, 2023-01-11:
可以将旁路由唯一的网口改成wan，新建一个lan口，和softether的tap网口桥接，这个lan口开启dhcp，softether里关闭虚拟nat和dhcp就可以了，实测内外网全部正常

bskdsbz, 2024-01-05:
楼上的方法可行，关于修改网络接口那段，我补充一下，方便新手
网络-接口，编辑原来的lan口，从桥接设备（比如lan），改成物理端口（比如eth0）
修改网络-接口-设备，把桥接设备（lan）所含端口，去掉eth0，改成vpn的端口
网络-接口，添加新接口，设备选择桥接设备（lan）
此时一共有2个端口，均为静态地址，防火墙设置均为lan，eht0与主路由同网段（192.168.1.1/24），网关为主路由，lan口设置新网段(192.168.2.1/24)，网关为eth0（192.168.1.1）