coolsnowwolf/lede #5475: NEWIFI3在编译PHP时出错

hackyes, 2020-09-16:
Makefile:584: recipe for target '/home/jingsong/lede/build_dir/target-mipsel_24kc_musl/php-7.4.7/.configured_fa77754e9d01069270447dcc8f3da842' failed
make[3]: *** [/home/jingsong/lede/build_dir/target-mipsel_24kc_musl/php-7.4.7/.configured_fa77754e9d01069270447dcc8f3da842] Error 1
make[3]: Leaving directory '/home/jingsong/lede/feeds/packages/lang/php7'
time: package/feeds/packages/php7/compile#9.66#5.98#16.97
package/Makefile:111: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: *** [package/feeds/packages/php7/compile] Error 2
make[2]: Leaving directory '/home/jingsong/lede'
package/Makefile:107: recipe for target '/home/jingsong/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/jingsong/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jingsong/lede'

hackyes, 2020-09-16:
把PHP改成 7.4.10 又可以编译了，默认的7.4.7不能编译

coolsnowwolf, 2020-09-16:
发个 PR

closed 2021-06-20 by coolsnowwolf