coolsnowwolf/lede #5480: 编译出错

Guxing0321, 2020-09-17:
Hi Team,

拉的最近的代码，编译失败

```ruby
FAILED: obj/net/net/url_request_context_builder.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/net/net/url_request_context_builder.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=728755d2fb5455d1d3293b6b10704fca859d97dd -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DHAVE_PTHREAD -DOPENSSL_NO_ASM -DNET_IMPLEMENTATION -DENABLE_BUILT_IN_DNS -I../.. -Igen -I../../third_party/protobuf/src -Igen/protoc_out -I../../third_party/protobuf/src -I../../third_party/boringssl/src/include -I../../third_party/zlib -I../../third_party/brotli/include -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -fcomplete-member-pointers -march=mipsel -mcpu=mips32r2 -msoft-float -mtune=24kc -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fno-ident -fdata-sections -ffunction-sections -Wno-header-guard -I/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nss -I/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nspr -Wexit-time-destructors -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl -I/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nss -I/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nspr -fvisibility-inlines-hidden --target=mipsel-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../net/url_request/url_request_context_builder.cc -o obj/net/net/url_request_context_builder.o
Killed
[62/716] CXX obj/net/net/url_request_job.o
FAILED: obj/net/net/url_request_job.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/net/net/url_request_job.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=728755d2fb5455d1d3293b6b10704fca859d97dd -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DHAVE_PTHREAD -DOPENSSL_NO_ASM -DNET_IMPLEMENTATION -DENABLE_BUILT_IN_DNS -I../.. -Igen -I../../third_party/protobuf/src -Igen/protoc_out -I../../third_party/protobuf/src -I../../third_party/boringssl/src/include -I../../third_party/zlib -I../../third_party/brotli/include -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -fcomplete-member-pointers -march=mipsel -mcpu=mips32r2 -msoft-float -mtune=24kc -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fno-ident -fdata-sections -ffunction-sections -Wno-header-guard -I/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nss -I/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nspr -Wexit-time-destructors -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl -I/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nss -I/data/lede/staging_dir/target-mipsel_24kc_musl/usr/include/nspr -fvisibility-inlines-hidden --target=mipsel-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../net/url_request/url_request_job.cc -o obj/net/net/url_request_job.o
Killed
[66/716] CXX obj/net/net/url_request_job_manager.o
ninja: build stopped: subcommand failed.
Makefile:85: recipe for target '/data/lede/build_dir/target-mipsel_24kc_musl/naiveproxy-85.0.4183.83-3/.built' failed
make[3]: *** [/data/lede/build_dir/target-mipsel_24kc_musl/naiveproxy-85.0.4183.83-3/.built] Error 1
make[3]: Leaving directory '/data/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#227.20#133.29#590.15
package/Makefile:111: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[2]: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/data/lede'
package/Makefile:107: recipe for target '/data/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/data/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/data/lede'
/data/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
```

BR
Robin

xzjue5, 2020-09-17:
把SSRplus的naiveproxy取消掉

coolsnowwolf, 2020-09-17:
naiveproxy 在 mips 平台存在问题

hcym, 2020-09-17:
难怪没遇过，有半年没编译mtk

Dimitri2020007, 2020-09-17:
取消naiveproxy的编译，x86没编译成功

afeiy, 2020-09-18:
ubuntu20.4 server编译ac2100和x86都通过，带naiveproxy编译

Guxing0321, 2020-09-18:
Dear all, 

昨晚尝试了将 naiveproxy 去掉，编译成功了
感谢大佬！

BR
Robin

closed 2020-09-18 by Guxing0321

ctr54188, 2020-10-11:
/bin/sh: 1: ../../third_party/llvm-build/Release+Asserts/bin/clang++: not found
ninja: build stopped: subcommand failed.
Makefile:83: recipe for target '/home/long/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-4/.built' failed
make[2]: *** [/home/long/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-4/.built] Error 1
make[2]: Leaving directory '/home/long/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#15.98#1.34#8.81
package/Makefile:111: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[1]: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
make[1]: Leaving directory '/home/long/lede'
/home/long/lede/include/toplevel.mk:222: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
long@ubuntu-pve:~/lede$ ninja: build stopped: subcommand failed./naiveproxy-85.0.4183.83-4/.built' failed

x86同样的错误