coolsnowwolf/lede #5482: 请教下各位大神，关于在状态-概览里添加磁盘信息的问题

trombonist852, 2020-09-17:
请教下各位大神，我想在概览里面添加磁盘信息，现在已知在路由器里面把disc_status.htm、nvme_status.htm和disc.htm这三个文件分别添加到/usr/lib/lua/luci/view/admin_status/和/usr/lib/lua/luci/view/admin_status/index/这两个位置下就可以成功显示，那如果我要添加到源码里编译的话是应该修改源码里面的哪个文件夹？路由器里添加好之后效果如下
[image]

BOBINIUNIU, 2020-09-17:
编译前在根目录新建files文件夹，然后在这个文件夹下新建/usr/lib/lua/luci/view/admin_status和/usr/lib/lua/luci/view/admin_status/index，然后把对应的文件拷贝进去
话说你的disc_status.htm是在哪里找到的，我用的4.19，没看到这几个文件
希望大佬可以加入查看磁盘信息的功能，这样就可以查看硬盘状态和温度了。如果能查看SMART信息就更好了

BOBINIUNIU, 2020-09-17:
我用的是openwrt分支，不知道是不是要编译disk man才有disc_status.htm、nvme_status.htm

trombonist852, 2020-09-17:
> 编译前在根目录新建files文件夹，然后在这个文件夹下新建/usr/lib/lua/luci/view/admin_status和/usr/lib/lua/luci/view/admin_status/index，然后把对应的文件拷贝进去
> 话说你的disc_status.htm是在哪里找到的，我用的4.19，没看到这几个文件
> 希望大佬可以加入查看磁盘信息的功能，这样就可以查看硬盘状态和温度了。如果能查看SMART信息就更好了
好的，感谢，我试试，建好文件夹拖进去文件直接编译就行了？不用其他操作？不光4.19没有，5.4也没有，我是从别的固件里面拖出来的，哈哈，其实在编译的时候选中diskman就可以了，只不过磁盘信息只能是在diskman下才能查看，不能在概览里显示，我是想着在概览里显示磁盘能方便查看

BOBINIUNIU, 2020-09-17:
> 编译前在根目录新建files文件夹，然后在这个文件夹下新建/usr/lib/lua/luci/view/admin_status和/usr/lib/lua/luci/view/admin_status/index，然后把对应的文件拷贝进去
> 话说你的disc_status.htm是在哪里找到的，我用的4.19，没看到这几个文件 [...]

是的，新建好文件夹，编译后就会把你自定义的这几个文件编译进固件里。一般用来自定义etc\config下的配置文件

trombonist852, 2020-09-17:
> 是的，新建好文件夹，编译后就会把你自定义的这几个文件编译进固件里。一般用来自定义etc\config下的配置文件

好  我去试试！太感谢啦！你需要这三个文件的话我传给你啊

BOBINIUNIU, 2020-09-17:
> 好 我去试试！太感谢啦！你需要这三个文件的话我传给你啊

嗯嗯，把那三个文件共享一下吧，我也编译一下看看，谢谢。

trombonist852, 2020-09-17:
> 嗯嗯，把那三个文件共享一下吧，我也编译一下看看，谢谢。

[files.zip](https://github.com/coolsnowwolf/lede/files/5240235/files.zip)

closed 2021-06-20 by coolsnowwolf