coolsnowwolf/lede #5484: 多个设备编译报错！patch失败！

IvanSolis1989, 2020-09-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：patch报错

### (2) 路由器型号和固件版本

A：树莓派4 K2P K2T B70 MI R3P K3 R2S

### (3) 详细日志

Applying ./patches/010-gcc10.patch using plaintext: 
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/src/http.h
|+++ b/src/http.h
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
can't find file to patch at input line 13
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/src/tls.h
|+++ b/src/tls.h
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/010-gcc10.patch!
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_cortex-a72_musl/shadowsocks-libev-3.3.5/.prepared_813e41c8eec3047cba30468fa06d1691_6664517399ebbbc92a37c5bb081b5c53] Error 1
Makefile:132: recipe for target '/workdir/openwrt/build_dir/target-aarch64_cortex-a72_musl/shadowsocks-libev-3.3.5/.prepared_813e41c8eec3047cba30468fa06d1691_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/shadowsocks-libev'
time: package/feeds/packages/shadowsocks-libev/compile#0.17#0.07#0.22
make[2]: *** [package/feeds/packages/shadowsocks-libev/compile] Error 2
package/Makefile:111: recipe for target 'package/feeds/packages/shadowsocks-libev/compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

xwcoco, 2020-09-18:
同样的问题....

qianxu2001, 2020-09-18:
似乎是我发的那个更新 ss-libev 的 pr 导致的，等大雕看看吧，我也不确定我那个 pr 没问题。。。

qianxu2001, 2020-09-18:
应该是 packages 仓库那边 ss-libev 更新导致的问题，大雕好像已经修复了，再编译应该就行了

IvanSolis1989, 2020-09-18:
是吗。我看看额

IvanSolis1989, 2020-09-18:
已修复 编译正常 谢谢

closed 2020-09-18 by IvanSolis1989