coolsnowwolf/lede #5488: opkg_install_pkg: Package size mismatch

Openwrtfunboy, 2020-09-18:
root@OpenWrt:~# opkg install htop
Installing htop (3.0.2-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/mipsel_24kc/packages/htop_3.0.2-1_mipsel_24kc.ipk
Installing terminfo (6.2-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/mipsel_24kc/base/terminfo_6.2-1_mipsel_24kc.ipk
Collected errors:
 * opkg_install_pkg: Package size mismatch: terminfo is 6955 bytes, expecting 6960 bytes
 * opkg_install_cmd: Cannot install package htop.

Maked yesterday firmware for newifi3 d2

When I making it in August, there was no such problem and i can install any packages. 

Now i cant install any packages.

Yes, of course, before opkg install i execute opkg update. 

Sorry if there is already an issue about this issue. I don't understand Chinese, and the translator doesn't translate well.

Openwrtfunboy, 2020-09-18:
if i installing directly all ok 

root@OpenWrt:~# opkg install /tmp/upload/terminfo_6.2-1_mipsel_24kc.ipk
Installing terminfo (6.2-1) to root...
Configuring terminfo.

Openwrtfunboy, 2020-09-18:
Also why i cant install qos/sqm?

Installing luci-app-qos (git-20.259.58095-dc51a2f-1) to root...
Downloading https://openwrt.proxy.ustclug.org/releases/18.06.8/packages/mipsel_24kc/luci/luci-app-qos_git-20.259.58095-dc51a2f-1_all.ipk

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-qos:
 * 	kernel (= 5.4.65-1-70f97ad5368ab3f3a02f0694a397f029)
 * opkg_install_cmd: Cannot install package luci-app-qos.

Openwrtfunboy, 2020-09-20:
so i set official openwrt repo (for free country)

and problem gone

Openwrtfunboy, 2020-09-20:
fuck china :)

AmadeusGhost, 2020-09-25:
First of all, it is extremely ridiculous to raise the problem to a national level. Second, open source repo have no obligation to answer any questions. Third, because of hybridization, you need to compile it yourself instead of using opkg install. Finally, your abuse can only prove your stupidity.
Wdnmd，有种别用这里的代码，滚

1715173329, 2020-09-25:
@Openwrtfunboy I've checked all your issues.
Everything you've ever posted sounds like a joke.

Since you've registered, only lots of garbage issues I found, but never a contribution.
Any complaints? Just roll your own source code, why you use ours?

And, if you ever did any research or had any knowledge about OpenWrt, you would never ask such a silly question about the installation of kernel modules.

Then, about "fuck China", stop using your device (Newifi3) right now, the openwrt source code etc which are made or contributed by Chinese, never touch them, as you are fucked.

If you do like rewrite, just do everything by yourself, otherwise, shut the fuck up.
Such a foolish! Don't shame your country, get out of here!

1715173329, 2020-09-25:
顺带一提，你个傻逼装你🐴的外国人呢搁这，当洋人等着被舔🐴¿你🐴是你🐴还是你🐎¿
发了一堆狗屁不通的英语搞笑呢这是？

滚吧，把你的新三扔了，用 OpenWrt，
、PandoraBox 干嘛啊？还 fuck China，你是天天被 penis 操到神智不清了？
搁这扔 nm 一大堆傻逼问题，你这么喜欢重写，滚去写啊，还 use firmware drivers，这是你写的驱动？你定的协议？你开的源？

1715173329, 2020-09-25:
趁早给老子滚，有多远滚多远，丢 nm 的人。

buildiffuse, 2020-09-25:
这哥们脑门被夹了吧，在这里煽动仇恨，这态度什么人都不想教会你，不喜欢就别用，没人求着你！fuck off！

Openwrtfunboy, 2020-09-26:
You can insult me ​​as much as you like. In my country I can talk with a friend about the events in Tanmanyan Square, and also make funny pictures with my president. If you do same you go to the gulag. Chinese gulag. :)

closed 2020-09-26 by Openwrtfunboy

1715173329, 2020-09-26:
笑死我了，用个固件还用出国家优越感了
你就是壬上壬？
openwrt 众 mt76 用户跪求宁重写驱动，如果你还是个人，，qqqxx

AmadeusGhost, 2020-09-27:
@Openwrtfunboy Since your discussion rise to the level of attacking the country, don’t use this code here any more, this repo does not welcome such malicious attackers.