coolsnowwolf/lede #5489: 编译naiveproxy,报错

xinyitextile, 2020-09-19:
ninja: build stopped: subcommand failed.
Makefile:85: recipe for target '/home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-85.0.4183.83-3/.built' failed
make[3]: *** [/home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-85.0.4183.83-3/.built] Error 1
make[3]: Leaving directory '/home/parallels/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#527.36#275.50#723.08
package/Makefile:111: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[2]: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/parallels/lede'
package/Makefile:107: recipe for target '/home/parallels/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/parallels/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/parallels/lede'
/home/parallels/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Dimitri2020007, 2020-09-19:
要下载ninja和naiveproxy软件包。编译中下载失败了才会出这个错误。解决的办法就是跳过naiveproxy软件的编译。因为你即使2个都下载进dl了一样还是无法编译。

1715173329, 2020-09-19:
钱没加够，加钱买内存即可

coolsnowwolf, 2020-09-20:
内存不足，确认编译机有 8G 或者以上内存

closed 2020-10-01 by xinyitextile