coolsnowwolf/lede #5490: 编译报错

Sunnice, 2020-09-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：gcc错误

### (2) 路由器型号和固件版本

A：竞斗云

### (3) 详细日志

A：2020-09-18T11:35:30.6975587Z [ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocks-libev-3.3.5
2020-09-18T11:35:30.7039469Z 
2020-09-18T11:35:30.7041193Z Applying ./patches/010-gcc10.patch using plaintext: 
2020-09-18T11:35:30.7053222Z can't find file to patch at input line 3
2020-09-18T11:35:30.7054276Z Perhaps you used the wrong -p or --strip option?
2020-09-18T11:35:30.7055088Z The text leading up to this was:
2020-09-18T11:35:30.7055720Z --------------------------
2020-09-18T11:35:30.7058117Z |--- a/src/http.h
2020-09-18T11:35:30.7058718Z |+++ b/src/http.h
2020-09-18T11:35:30.7059233Z --------------------------
2020-09-18T11:35:30.7059731Z No file to patch.  Skipping patch.
2020-09-18T11:35:30.7060250Z 1 out of 1 hunk ignored
2020-09-18T11:35:30.7063999Z make[3]: *** [/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocks-libev-3.3.5/.prepared_813e41c8eec3047cba30468fa06d1691_6664517399ebbbc92a37c5bb081b5c53] Error 1
2020-09-18T11:35:30.7068291Z can't find file to patch at input line 13
2020-09-18T11:35:30.7069102Z Perhaps you used the wrong -p or --strip option?
2020-09-18T11:35:30.7069656Z The text leading up to this was:
2020-09-18T11:35:30.7070223Z --------------------------
2020-09-18T11:35:30.7070786Z |--- a/src/tls.h
2020-09-18T11:35:30.7071215Z |+++ b/src/tls.h
2020-09-18T11:35:30.7071742Z --------------------------
2020-09-18T11:35:30.7072230Z No file to patch.  Skipping patch.
2020-09-18T11:35:30.7072659Z 1 out of 1 hunk ignored
2020-09-18T11:35:30.7073331Z Patch failed!  Please fix ./patches/010-gcc10.patch!
2020-09-18T11:35:30.7074784Z Makefile:132: recipe for target '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocks-libev-3.3.5/.prepared_813e41c8eec3047cba30468fa06d1691_6664517399ebbbc92a37c5bb081b5c53' failed
2020-09-18T11:35:30.7076260Z make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/shadowsocks-libev'
2020-09-18T11:35:30.7077324Z time: package/feeds/packages/shadowsocks-libev/compile#0.16#0.06#0.21
2020-09-18T11:35:30.7078362Z package/Makefile:111: recipe for target 'package/feeds/packages/shadowsocks-libev/compile' failed
2020-09-18T11:35:30.7079290Z make[2]: Leaving directory '/workdir/openwrt'
2020-09-18T11:35:30.7115707Z make[2]: *** [package/feeds/packages/shadowsocks-libev/compile] Error 2
2020-09-18T11:35:30.7116796Z package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
2020-09-18T11:35:30.7117997Z make[1]: *** [/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
2020-09-18T11:35:30.7118499Z make: *** [world] Error 2
2020-09-18T11:35:30.7119081Z make[1]: Leaving directory '/workdir/openwrt'
2020-09-18T11:35:30.7119873Z /workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
2020-09-18T11:35:30.7436353Z ##[error]Process completed with exit code 2.
2020-09-18T11:35:30.7794889Z ##[group]Run df -hT
2020-09-18T11:35:30.7795358Z [36;1mdf -hT[0m
2020-09-18T11:35:30.9790940Z shell: /bin/bash -e {0}
2020-09-18T11:35:30.9791156Z env:

AmadeusGhost, 2020-09-19:
更新 feeds 即可

Sunnice, 2020-09-20:
> 更新 feeds 即可

好的，我重新试试看。对了大佬，请问一下为什么使用lede源码进行云编译生成的固件经常没有无线，时有时无，稳定版源码就没有这种问题。另外还有一个问题就是最近编译的固件，无论是lede源码或者稳定版源码都会莫名其妙地出现一个flowoffloading的配置未保存的情况，我编译的时候也没有选flowoffloading，不知道是什么原因造成的。
[image]

closed 2021-06-20 by coolsnowwolf