coolsnowwolf/lede #5491: 703N编译naiveproxy错误

afeiy, 2020-09-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：勾选naiveproxy已编译成功的固件有x86、斐讯N1、红米ac2100，单编译703时则出现错误，必须取消勾选naiveproxy才成功。

### (2) 路由器型号和固件版本

A：tp-link wr703N

### (3) 详细日志

A：make[3]: Entering directory '/home/afeiy/lede/feeds/helloworld/naiveproxy'
rm -f /home/afeiy/lede/build_dir/target-mips_24kc_musl/naiveproxy-85.0.4183.83-3/.built
touch /home/afeiy/lede/build_dir/target-mips_24kc_musl/naiveproxy-85.0.4183.83-3/.built_check
( cd /home/afeiy/lede/build_dir/target-mips_24kc_musl/naiveproxy-85.0.4183.83-3 ; ./tools/import-upstream.sh ; . ./init_env.sh "mips" "ar71xx" "24kc" "" "/home/afeiy/lede/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl" "/home/afeiy/lede/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/home/afeiy/lede/staging_dir/target-mips_24kc_musl" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/home/afeiy/lede/staging_dir/hostpkg/bin/python3" ; /home/afeiy/lede/staging_dir/hostpkg/bin/ninja -C "out/Release" naive ; )
+ cut -d= -f2 src/chrome/VERSION
+ + tr \n .
cut -d. -f1-4
+ have_version=85.0.4183.83
+ cat CHROMIUM_VERSION
+ want_version=85.0.4183.83
+ [ 85.0.4183.83 = 85.0.4183.83 ]
+ exit 0
ERROR at //skia/BUILD.gn:817:5: Assertion failed.
    assert(false, "Need to port cpu specific stuff from skia_library_opts.gyp")
    ^-----
Need to port cpu specific stuff from skia_library_opts.gyp
See //BUILD.gn:80:5: which caused the file to be included.
    "//skia:skia_unittests",
    ^----------------------
ninja: Entering directory `out/Release'
ninja: error: loading 'build.ninja': No such file or directory
make[3]: *** [Makefile:88: /home/afeiy/lede/build_dir/target-mips_24kc_musl/naiveproxy-85.0.4183.83-3/.built] Error 1
make[3]: Leaving directory '/home/afeiy/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#2.61#0.42#12.16
make[2]: *** [package/Makefile:113: package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/afeiy/lede'
make[1]: *** [package/Makefile:107: /home/afeiy/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/afeiy/lede'
make: *** [/home/afeiy/lede/include/toplevel.mk:224: world] Error 2

xingpo, 2020-09-19:
你这个是不是没有安装ninja-build ???或是样子没下载到这个源码包引起的！
我也是存在这个问题，x86也是一直没编译过去，不选naiveproxy就可以顺利编译！也安装了clang,libnss3等必要软件！前两处错误我根据错误提示手动创建了软链接和目录文件，最后还是没编译过去，最后出错信息如下，我在想是不是我少安装了哪个软件包，希望大神帮忙，我水平有限，小问题能解决，像这情况以我的水平已经不知问题出在哪里了！
Done. Made 12712 targets from 2280 files in 9437ms
ninja: Entering directory `out/Release'
[1/1922] CXX obj/base/base_static/base_switches.o
FAILED: obj/base/base_static/base_switches.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/base/base_static/base_switches.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION="n356902-4e813bbd-1" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/blacklist.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fno-ident -fdata-sections -ffunction-sections -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/zyy/compile/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/zyy/compile/openwrt/staging_dir/target-x86_64_musl/usr/include/nss -I/home/zyy/compile/openwrt/staging_dir/target-x86_64_musl/usr/include/nspr -fvisibility-inlines-hidden --target=x86_64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../base/base_switches.cc -o obj/base/base_static/base_switches.o
warning: unknown warning option '-Wno-non-c-typedef-for-linkage'; did you mean '-Wno-return-type-c-linkage'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-max-tokens'; did you mean '-Wno-extra-tokens'? [-Wunknown-warning-option]
error: unable to find plugin 'find-bad-constructs'
2 warnings and 1 error generated.
[2/1922] CXX obj/buildtools/third_party/libc++/libc++/memory.o
FAILED: obj/buildtools/third_party/libc++/libc++/memory.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/memory.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=attribute((visibility("default"))) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION="n356902-4e813bbd-1" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/blacklist.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -fstrict-aliasing -fPIC -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-templat [... cut ...]
make[2]: *** [Makefile:88: /home/zyy/compile/openwrt/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-1/.built] Error 1
make[2]: Leaving directory '/home/zyy/compile/openwrt/feeds/diy1/package/naiveproxy'
time: package/feeds/diy1/naiveproxy/compile#16.20#2.21#14.33
make[1]: *** [package/Makefile:113: package/feeds/diy1/naiveproxy/compile] Error 2
make[1]: Leaving directory '/home/zyy/compile/openwrt'
make: *** [/home/zyy/compile/openwrt/include/toplevel.mk:236: package/feeds/diy1/naiveproxy/compile] Error 2

1715173329, 2020-09-19:
@afeiy 是的，我并没有仔细查看过有关于 mips 架构上关于 naiveproxy 的构建，
因为在此架构上运行 naiveproxy 实在吃力，因此取消勾选即可，如果有空我也会看看构建方案

@xingpo 不跟着源码走那就自己负责吧

afeiy, 2020-09-19:
> @afeiy 是的，我并没有仔细查看过有关于 mips 架构上关于 naiveproxy 的构建，
> 因为在此架构上运行 naiveproxy 实在吃力，因此取消勾选即可，如果有空我也会看看构建方案
> 
> @xingpo 不跟着源码走那就自己负责吧

已了解，谢谢。还是十分您期待有空构建该机型，目前ss和trojian能跑1080，出差伴侣，我的第一个op路由，有感情。

closed 2020-09-19 by afeiy