coolsnowwolf/lede #5502: 竞斗云固件没有编译flowoffloading，但是刷完固件进去后为何会出现相关设置未保存的提示？另外，编译的竞斗云固件居然没有无线，不知道是什么原因？

Sunnice, 2020-09-20:
竞斗云固件没有编译flowoffloading，但是刷完固件进去后为何会出现相关设置未保存的提示？另外，编译的竞斗云固件居然没有无线，不知道是什么原因？
[image]

AmadeusGhost, 2020-09-21:
实测无线正常，你固件做小一点就好（别到将近 100M 这种程度）

Sunnice, 2020-09-21:
> 实测无线正常，你固件做小一点就好（别到将近 100M 这种程度）

那我可能就是超体积了，谢谢暗云大佬（永远嘀神！

Sunnice, 2020-09-21:
> 实测无线正常，你固件做小一点就好（别到将近 100M 这种程度）

请问一下大佬我图片上显示flowoffload配置问题是怎么回事呀？我没有编译flowoff插件呀，这个配置未保存的提示每次都有，不论是lede还是稳定版源码.

AmadeusGhost, 2020-09-22:
> 请问一下大佬我图片上显示flowoffload配置问题是怎么回事呀？我没有编译flowoff插件呀，这个配置未保存的提示每次都有，不论是lede还是稳定版源码.

不保存配置升级试试

closed 2020-09-23 by coolsnowwolf

Sunnice, 2020-09-24:
> > 请问一下大佬我图片上显示flowoffload配置问题是怎么回事呀？我没有编译flowoff插件呀，这个配置未保存的提示每次都有，不论是lede还是稳定版源码.
> 
> 不保存配置升级试试

我固件都是从opboot刷的ubi文件，都是不保存配置的。每次进去都有这个，tg群里也有人跟我有相同情况

doumeki, 2020-11-01:
一样,可以这样改，不过大神可以来看看有什么副作用没

vi /etc/init.d/autocore
sysctl -w net.xxxx 那行前面加#号注释掉

[image]