coolsnowwolf/lede #5503: trojan内存泄漏，猜测可能与ssrplus客户端数据规则造成迂回有关

hawie, 2020-09-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
trojan内存泄漏，初以为是trojan本身问题，但经过观测，猜测可能与ssrplus的某iptables规则造成数据迂回有关。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
观测trojan服务端日志，发现 ssrplus的trojan客户端会不断产生迂回到客户端自身ip的23, 22, ...等各种端口（有时这个有时那个）的无效访问，此时trojan的VSZ内存不断急骤增大。
在Mac上使用另一客户端trojan-qt5连接同一个服务端，不存在这个问题，也不会引起服务端内存泄漏。
此时，ssrplus的trojan客户端同样内存不断增大。
尝试了几个网上下载的版本，有这个问题，然后从本源下载20200915 tag源码编译，现象一样。特此反馈请教。
启用的主要配置：
flowoffload, hwnat  fullcone nat.
mwan3, 策略备用成员改为“默认（使用主路由表）”（为了ipv6），电信光纤(PPPoE)、移动光纤(DHCP)，负载均衡。
ddns
无线禁用。
防火墙，lan->wan,wanb; wan->reject; wanb->reject.

### (1) 具体问题

A：ssrplus启用trojan客户端，可能存在某些访问策略导致了数据迂回，不断循环不止。具体不知是哪条引起，希望能帮助排查与解决。
也试了多个别人编译的以前的其他版本，测试现象一样。

### (2) 路由器型号和固件版本

A：
主机型号 | Newifi-D2 (CpuMark : 4411.926909 Scores)
架构 | MediaTek MT7621 ver:1 eco:3 x 4
固件版本 | OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)
内核版本 | 5.4.61
使用20200915 tag源码编译。

### (3) 详细日志

A：服务端，其中116.7.22.254为ssrplus所在路由的ip
root@libra:~# trojan --config /etc/trojan.json 
Welcome to trojan 1.16.0
[2020-09-20 14:33:00] [WARN] trojan service (server) started at 0.0.0.0:443
[2020-09-20 14:33:01] [INFO] 116.7.22.254:47640 authenticated as password1
[2020-09-20 14:33:01] [INFO] 116.7.22.254:47640 requested connection to 116.7.22.254:23
[2020-09-20 14:33:02] [INFO] 116.7.22.254:47638 authenticated as password1
[2020-09-20 14:33:02] [INFO] 116.7.22.254:47638 requested connection to 116.7.22.254:23
[2020-09-20 14:33:02] [INFO] 116.7.22.254:47634 authenticated as password1
[2020-09-20 14:33:02] [INFO] 116.7.22.254:47634 requested connection to 116.7.22.254:25105
[2020-09-20 14:33:03] [INFO] 116.7.22.254:47642 authenticated as password1
[2020-09-20 14:33:03] [INFO] 116.7.22.254:47642 requested connection to 116.7.22.254:23
[2020-09-20 14:33:03] [INFO] 116.7.22.254:47644 authenticated as password1
[2020-09-20 14:33:03] [INFO] 116.7.22.254:47644 requested connection to 116.7.22.254:23
[2020-09-20 14:33:05] [INFO] 116.7.22.254:47654 authenticated as password1
[2020-09-20 14:33:05] [INFO] 116.7.22.254:47654 requested connection to 116.7.22.254:23
[2020-09-20 14:33:05] [INFO] 116.7.22.254:47650 authenticated as password1
[2020-09-20 14:33:05] [INFO] 116.7.22.254:47650 requested connection to 116.7.22.254:25105
[2020-09-20 14:33:06] [INFO] 116.7.22.254:47656 authenticated as password1
[2020-09-20 14:33:06] [INFO] 116.7.22.254:47656 requested connection to 116.7.22.254:23
[2020-09-20 14:33:06] [INFO] 116.7.22.254:47652 authenticated as password1
[2020-09-20 14:33:06] [INFO] 116.7.22.254:47652 requested connection to 116.7.22.254:23
[2020-09-20 14:33:06] [INFO] 116.7.22.254:47662 authenticated as password1
[2020-09-20 14:33:06] [INFO] 116.7.22.254:47662 requested connection to 116.7.22.254:23
[2020-09-20 14:33:07] [INFO] 116.7.22.254:47634 disconnected, 0 bytes received, 16 bytes sent, lasted for 5 seconds
[2020-09-20 14:33:07] [INFO] 116.7.22.254:47650 disconnected, 0 bytes received, 16 bytes sent, lasted for 4 seconds
[2020-09-20 14:33:07] [INFO] 116.7.22.254:47664 authenticated as password1
[2020-09-20 14:33:07] [INFO] 116.7.22.254:47664 requested connection to 116.7.22.254:23
[2020-09-20 14:33:08] [INFO] 116.7.22.254:47660 authenticated as password1
[2020-09-20 14:33:08] [INFO] 116.7.22.254:47660 requested connection to 116.7.22.254:25105
[2020-09-20 14:33:08] [INFO] 116.7.22.254:47660 disconnected, 0 bytes received, 16 bytes sent, lasted for 1 seconds
[2020-09-20 14:33:08] [INFO] 116.7.22.254:47666 authenticated as password1
[2020-09-20 14:33:08] [INFO] 116.7.22.254:47666 requested connection to 116.7.22.254:23
[2020-09-20 14:33:09] [INFO] 116.7.22.254:47672 authenticated as password1
[2020-09-20 14:33:09] [INFO] 116.7.22.254:47672 requested connection to 116.7.22.254:23
[2020-09-20 14:33:10] [INFO] 116.7.22.254:47676 authenticated as password1
[2020-09-20 14:33:10] [INFO] 116.7.22.254:47676 requested connection to 116.7.22.254:23
[2020-09-20 14:33:10] [INFO] 116.7.22.254:47668 authenticated as password1
[2020-09-20 14:33:10] [INFO] 116.7.22.254:47668 requested connection to 116.7.22.254:25105

ps几次: 能明显看到服务端内存增长：
PID USER       VSZ STAT COMMAND
...
14228 root      4488 S    /usr/sbin/trojan --config /etc/trojan.json
...
14228 root      5136 S    /usr/sbin/trojan --config /etc/trojan.json
...
14228 root      6760 S    /usr/sbin/trojan --config /etc/trojan.json
...
14228 root     50320 R    /usr/sbin/trojan --config /etc/trojan.json
...
很快到内存耗尽：
root@libra:~# free
             total         used         free       shared      buffers
Mem:         61068        60004         1064          672          292
-/+ buffers:              59712         1356
Swap:            0            0            0

ps: ssrplus客户端：
21452 root     50444 S    /usr/sbin/trojan --config /var/etc/trojan-ssr-retcp.json
21453 root     48600 S    /usr/sbin/trojan --config /var/etc/trojan-ssr-retcp.json
21454 root     46120 S    /usr/sbin/trojan --config /var/etc/trojan-ssr-retcp.json
21455 root     48600 S    /usr/sbin/trojan --config /var/etc/trojan-ssr-retcp.json
四核，泄漏时每个占用近50M；正常时，每个在10M内或附近。

使用其他客户端（macosx, trojan-qt5)访问服务端，无多余的迂回访问，此时内存占用也正常：
root@libra:~# /usr/sbin/trojan --config /etc/trojan.json
Welcome to trojan 1.16.0
[2020-09-20 19:52:52] [WARN] trojan service (server) started at 0.0.0.0:443
[2020-09-20 19:53:18] [INFO] 116.7.22.254:51655 authenticated as password1
[2020-09-20 19:53:18] [INFO] 116.7.22.254:51655 requested connection to www.google.com:443
[2020-09-20 19:53:26] [INFO] 116.7.22.254:51668 authenticated as password1
[2020-09-20 19:53:26] [INFO] 116.7.22.254:51668 requested connection to www.google.com:443

hawie, 2020-09-20:
root@libra:~# /usr/sbin/trojan  --config /etc/trojan.json 
Welcome to trojan 1.16.0
[2020-09-20 20:38:33] [WARN] trojan service (server) started at 0.0.0.0:443
[2020-09-20 20:38:46] [INFO] 116.25.106.163:48578 authenticated as password1
[2020-09-20 20:38:46] [INFO] 116.25.106.163:48578 requested connection to 216.58.199.3:443
[2020-09-20 20:38:56] [INFO] 116.25.106.163:48580 authenticated as password1
[2020-09-20 20:38:56] [INFO] 116.25.106.163:48580 requested connection to 216.58.199.3:443

hawie, 2020-09-20:
续上面的日志，刚写贴这几秒钟之后就出现了。root@libra:~# /usr/sbin/trojan  --config /etc/trojan.json 
Welcome to trojan 1.16.0
[2020-09-20 20:38:33] [WARN] trojan service (server) started at 0.0.0.0:443
[2020-09-20 20:38:46] [INFO] 116.25.106.163:48578 authenticated as password1
[2020-09-20 20:38:46] [INFO] 116.25.106.163:48578 requested connection to 216.58.199.3:443
[2020-09-20 20:38:56] [INFO] 116.25.106.163:48580 authenticated as password1
[2020-09-20 20:38:56] [INFO] 116.25.106.163:48580 requested connection to 216.58.199.3:443
[2020-09-20 20:39:03] [INFO] 116.25.106.163:48582 authenticated as password1
[2020-09-20 20:39:03] [INFO] 116.25.106.163:48582 requested connection to 216.58.220.206:443
[2020-09-20 20:39:05] [INFO] 116.25.106.163:48584 authenticated as password1
[2020-09-20 20:39:05] [INFO] 116.25.106.163:48584 requested connection to 172.217.161.163:443
[2020-09-20 20:39:05] [INFO] 116.25.106.163:48586 authenticated as password1
[2020-09-20 20:39:05] [INFO] 116.25.106.163:48586 requested connection to 172.217.25.14:443
[2020-09-20 20:39:05] [INFO] 116.25.106.163:48590 authenticated as password1
[2020-09-20 20:39:05] [INFO] 116.25.106.163:48590 requested connection to 114.114.114.114:53
[2020-09-20 20:39:06] [INFO] 116.25.106.163:48590 disconnected, 105 bytes received, 40 bytes sent, lasted for 1 seconds
[2020-09-20 20:39:08] [INFO] 116.25.106.163:48588 authenticated as password1
[2020-09-20 20:39:08] [INFO] 116.25.106.163:48588 requested connection to 114.114.114.114:53
[2020-09-20 20:39:09] [INFO] 116.25.106.163:48592 authenticated as password1
[2020-09-20 20:39:09] [INFO] 116.25.106.163:48592 requested connection to 114.114.114.114:53
[2020-09-20 20:39:09] [INFO] 116.25.106.163:48588 disconnected, 56 bytes received, 40 bytes sent, lasted for 2 seconds
[2020-09-20 20:39:10] [INFO] 116.25.106.163:48592 disconnected, 52 bytes received, 36 bytes sent, lasted for 3 seconds
[2020-09-20 20:39:11] [INFO] 116.25.106.163:48594 authenticated as password1
[2020-09-20 20:39:11] [INFO] 116.25.106.163:48594 requested connection to 114.114.114.114:53
[2020-09-20 20:39:11] [INFO] 116.25.106.163:48594 disconnected, 105 bytes received, 40 bytes sent, lasted for 1 seconds
[2020-09-20 20:40:15] [INFO] 116.25.106.163:48596 authenticated as password1
[2020-09-20 20:40:15] [INFO] 116.25.106.163:48596 requested connection to 116.25.106.163:48578
[2020-09-20 20:40:15] [INFO] 116.25.106.163:48598 authenticated as password1
[2020-09-20 20:40:15] [INFO] 116.25.106.163:48598 requested connection to 116.25.106.163:48580
[2020-09-20 20:40:16] [INFO] 116.25.106.163:48600 authenticated as password1
[2020-09-20 20:40:16] [INFO] 116.25.106.163:48600 requested connection to 116.25.106.163:48580
[2020-09-20 20:40:17] [INFO] 116.25.106.163:48602 authenticated as password1
[2020-09-20 20:40:17] [INFO] 116.25.106.163:48602 requested connection to 116.25.106.163:48578
[2020-09-20 20:40:20] [INFO] 116.25.106.163:48606 authenticated as password1
[2020-09-20 20:40:20] [INFO] 116.25.106.163:48606 requested connection to 116.25.106.163:48580
[2020-09-20 20:40:21] [INFO] 116.25.106.163:48608 authenticated as password1
[2020-09-20 20:40:21] [INFO] 116.25.106.163:48608 requested connection to 116.25.106.163:48580
[2020-09-20 20:40:22] [INFO] 116.25.106.163:48604 authenticated as password1
[2020-09-20 20:40:22] [INFO] 116.25.106.163:48604 requested connection to 116.25.106.163:48578
[2020-09-20 20:40:23] [INFO] 116.25.106.163:48610 authenticated as password1
[2020-09-20 20:40:23] [INFO] 116.25.106.163:48610 requested connection to 116.25.106.163:48580
[2020-09-20 20:40:24] [INFO] 116.25.106.163:48614 authenticated as password1
[2020-09-20 20:40:24] [INFO] 116.25.106.163:48614 requested connection to 116.25.106.163:48580
[2020-09-20 20:40:24] [INFO] 116.25.106.163:48612 authenticated as password1
[2020-09-20 20:40:24] [INFO] 116.25.106.163:48612 requested connection to 116.25.106.163:48578
[2020-09-20 20:40:25] [INFO] 116.25.106.163:48616 authenticated as password1
[2020-09-20 20:40:25] [INFO] 116.25.106.163:48616 requested connection to 116.25.106.163:48580
[2020-09-20 20:40:26] [INFO] 116.25.106.163:48620 authenticated as password1
[2020-09-20 20:40:26] [INFO] 116.25.106.163:48620 requested connection to 116.25.106.163:20758
[2020-09-20 20:40:26] [INFO] 116.25.106.163:48618 authenticated as password1
[2020-09-20 20:40:26] [INFO] 116.25.106.163:48618 requested connection to 116.25.106.163:48578
[2020-09-20 20:40:26] [INFO] 116.25.106.163:48622 authenticated as password1
[2020-09-20 20:40:26] [INFO] 116.25.106.163:48622 requested connection to 116.25.106.163:48580

eallion, 2020-09-21:
先把光纤拔掉一根试试，不同运营商之间本来就不适合做负载均衡。

hawie, 2020-09-21:
> 先把光纤拔掉一根试试，不同运营商之间本来就不适合做负载均衡。

上面的日志已经是拔掉一根光纤之后做的测试，如果没拔掉，访问出去的IP是移动的，迂回访问的IP是电信的，就这个区别，其他一样现象。

昨天想着换个客户端试试，就把passwall编译进去。然后保留配置升级，关闭ssrplus改用passwall，同样环境目前一切正常。再继续观察两天，如果正常就暂时这么用着，不过还是很希望ssrplus能修复这个。

coolsnowwolf, 2020-09-24:
没有复现

hawie, 2020-09-24:
> 没有复现

老大，我这能100%复现。新刷一台新3，然后把配置一条条启用，然后就会出现这个现象。暂时改用passwall，跑了3天了，没这个现象，但它也有个问题是路由上再启用一个trojan服务端给远程不能google。

closed 2021-06-20 by coolsnowwolf

msdos03, 2024-07-08:
我也遇到了同样的问题，你当初解决了吗？