coolsnowwolf/lede #5511: 可道云 php库缺失GD库 无法使用缩略图

keeno1982, 2020-09-21:
php.ini里面没看到gd相关的配置，这个不开就会导致云盘里的图都会按原图全部下载下来
[image]

keeno1982, 2020-09-22:
Tue Sep 22 15:16:49 2020 daemon.err uhttpd[1689]: [22-Sep-2020 15:16:49] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'gd.so' (tried: /usr/lib/php/gd.so (Error loading shared library libjpeg.so.9: No such file or directory (needed by /usr/lib/libgd.so.3)), /usr/lib/php/gd.so.so (Error loading shared library /usr/lib/php/gd.so.so: No such file or directory)) in Unknown on line 0

Tue Sep 22 15:16:49 2020 daemon.err uhttpd[1689]: [22-Sep-2020 15:16:49] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'hash.so' (tried: /usr/lib/php/hash.so (Error loading shared library /usr/lib/php/hash.so: No such file or directory), /usr/lib/php/hash.so.so (Error loading shared library /usr/lib/php/hash.so.so: No such file or directory)) in Unknown on line 0

hackyes, 2020-09-22:
自己装个 php7-mod-gd , php的配置文件是 /etc/php7里面

keeno1982, 2020-09-22:
> 
> 
> 自己装个 php7-mod-gd , php的配置文件是 /etc/php7里面

php7-mod-gd已经有装了，还是提示找不到

hackyes, 2020-09-22:
> > 自己装个 php7-mod-gd , php的配置文件是 /etc/php7里面
> 
> php7-mod-gd已经有装了，还是提示找不到

文件是在 /usr/lib/php 里面，看看有没有gd.so文件， php.ini 添加 extension=gd.so

keeno1982, 2020-09-22:
> 
> 
> > > 自己装个 php7-mod-gd , php的配置文件是 /etc/php7里面
> > 
> > 
> > php7-mod-gd已经有装了，还是提示找不到
> 
> 文件是在 /usr/lib/php 里面，看看有没有gd.so文件， php.ini 添加 extension=gd.so

/usr/lib/php 是有gd.so，extension=gd.so这句话也加进去了，启动时还是照样报错，不知道是哪根筋错了

closed 2021-06-20 by coolsnowwolf