coolsnowwolf/lede #5513: WRT3200ACM无法编译

Val-istar-Guo, 2020-09-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

1. 我已经搜索过问题并且尝试升级过工具链，依旧无效
2. 我在ubuntu的docker中编译
3. 因编译日志较多，截取靠近错误的一段，如有需要我可以把编译日志存成文件上传
3. 取消了`feeds.conf.default`的helloworld行注释

### (1) 具体问题

A：无法编译...好似跟NodeJS有关系

### (2) 路由器型号和固件版本

A：WRT3200ACM，OpenWRT 19.07

### (3) 详细日志

[image]

```
  touch /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.target/tools/v8_gypfiles/v8_base.stamp
  g++ -m32 -o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot -pthread -rdynamic -m32 -L../../../../staging_dir/hostpkg/share/icu/current/lib  -Wl,--start-group /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/embedded-file-writer.o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-aix.o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-base.o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-generic.o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-mac.o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/embedded/platform-embedded-file-writer-win.o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/mksnapshot/deps/v8/src/snapshot/mksnapshot.o /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_base_without_compiler.a /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_init.a /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_libbase.a /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_libplatform.a /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_nosnapshot.a /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_libsampler.a /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_compiler.a /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.host/tools/v8_gypfiles/libv8_initializers.a -L/root/lede/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib -latomic -ldl -lrt -Wl,--end-group
  touch ebfbbedf33662feff1f589765579573fa20c1719.intermediate
  LD_LIBRARY_PATH=/root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/lib.host:/root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.target/v8_snapshot/geni; "/root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot" --turbo_instruction_scheduling "--target_os=linux" "--target_arch=arm" --startup_src "/root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.target/v8_snapshot/geni/snapshot.cc" --embedded_variant Default --embedded_src "/root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/obj.target/v8_snapshot/geni/embedded.S" --no-native-code-counters

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xf79bffd4

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xff875754
==== C stack trace ===============================

==== C stack trace ===============================

    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(v8::base::debug::StackTrace::StackTrace()+0x22) [0x57c5e872]
    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(+0xb2fe11) [0x57133e11]
    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(V8_Fatal(char const*, ...)+0xd8) [0x5712d028]
    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(v8::base::TimeTicks::HighResolutionNow()+0x98) [0x5712e9a8]
    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(+0xb2fd8a) [0x57133d8a]
    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(v8::platform::DelayedTaskQueue::GetNext()+0x56) [0x57136b76]
    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(v8::platform::DefaultWorkerThreadsTaskRunner::WorkerThread::Run()+0x50) [0x57135990]
    /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/out/Release/mksnapshot(+0xb2ddc4) [0x57131dc4]
    /lib32/libpthread.so.0(+0x860a) [0xf7bb660a]
    /lib32/libc.so.6(clone+0x6a) [0xf7ac9bca]
Illegal instruction
make[5]: *** [tools/v8_gypfiles/v8_snapshot.target.mk:17: ebfbbedf33662feff1f589765579573fa20c1719.intermediate] Error 132
rm e140f301a84d79f1041c55ba60d5d2b245216750.intermediate aa94f9dda57021c80085a4d282db3f87ddd81fcf.intermediate 6f534e9842dd009be3d3938c39de1bd36955addc.intermediate ebfbbedf33662feff1f589765579573fa20c1719.intermediate 68643ca3944bf7cd8ef903c698e560f824d94485.intermediate
make[4]: *** [Makefile:101: node] Error 2
make[4]: Leaving directory '/root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1'
make[3]: *** [Makefile:157: /root/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.16.1/.built] Error 2
make[3]: Leaving directory '/root/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#9585.46#544.48#2795.71
make[2]: *** [package/Makefile:113: package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:107: /root/lede/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:224: world] Error 2
root@c44e475be594:~/lede# make menuconfig
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
```

coolsnowwolf, 2020-09-23:
不支持 32bit 系统的编译

Val-istar-Guo, 2020-10-07:
谢谢，我考虑使用其他方案

closed 2020-10-07 by Val-istar-Guo