coolsnowwolf/lede #5516: 编译过程错误请教（一）

recinfo, 2020-09-22:
刚接触openwrt编译，按教程编译了两次，都是在漫长的等待后失败，没办法。留个系列问题贴
1、执行sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync后显示这个：
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成   
    
注意，选中 'zlib1g-dev' 而非 'libz-dev'
注意，选中 'node-uglify' 而非 'uglifyjs'
注意，选中 'git' 而非 'git-core'
注意，选中 'upx-ucl' 而非 'upx'

unzip 已经是最新版 (6.0-21ubuntu1)。
unzip 已设置为手动安装。
binutils 已经是最新版 (2.30-21ubuntu1~18.04.4)。
binutils 已设置为手动安装。
bzip2 已经是最新版 (1.0.6-8.1ubuntu0.2)。
bzip2 已设置为手动安装。
等
等
等

请问上面四个注意问题会不会影响下一步的操作？？

AmadeusGhost, 2020-09-22:
这个没影响

eallion, 2020-09-23:
直接贴摄氏日志比较好。

Dimitri2020007, 2020-09-27:
没有影响。

closed 2021-06-20 by coolsnowwolf