coolsnowwolf/lede #5520: 无法上国内网站，但google可以

rangerwill, 2020-09-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
斐讯n1和x64软路由都出现无法上国内网站，google等境外网站能正常访问的情况。
如果手动重启防火墙，则可以正常上网。
但过几小时之后，又不行了。

### (1) 具体问题

A：斐讯n1和x64软路由都出现无法上国内网站，google等境外网站能正常访问的情况。
如果手动重启防火墙，则可以正常上网。
但过几小时之后，又不行了。

### (2) 路由器型号和固件版本

A：斐讯N1
OpenWrt R20.9.15 / LuCI Master (git-20.223.26773-d18ef13) 
B：Intel J3710 + 8G DDR3
OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222) 
### (3) 详细日志

A：暂时无法提供

lexin8, 2020-09-22:
PC上运行SS客户端测试，先排除节点问题！有可能是DDNS修改过。

rangerwill, 2020-09-22:
> 
> 
> PC上运行SS客户端测试，先排除节点问题！有可能是DDNS修改过。

节点无问题，我尝试多次手动更新订阅，翻墙插件中测速到百度和谷歌的延迟都正常，
但是就是无法打开百度、搜狐等国内网站
在网络-防火墙-自定义中点击重启防火墙按钮，
网络恢复正常，
但过一段时间后，
网络又出现同样问题

rangerwill, 2020-09-22:
> 
> 
> PC上运行SS客户端测试，先排除节点问题！有可能是DDNS修改过。

而且我的n1和软路由同时都有这个问题，
我换了小米路由2直接拨号，
上网正常
说明运营商方面也没问题

rangerwill, 2020-09-22:
最近又装了一台Inter i5 6400+ 8G DDR4 的机器
还是用之前的OpenWrt系统盘，
能正常使用。
同样一块系统盘，
在J3710上就没法正常使用。
我怀疑是不是防火墙方面在代码层面有什么问题，
导致低配置机器出这样的问题。

afeiy, 2020-09-23:
N1要添加防火墙规则：iptables -t nat -I POSTROUTING -j MASQUERADE

garypang13, 2020-09-23:
我也偶尔遇到这种情况, service firewall restart 后才能恢复. 我用的是I3的机子固件是官方源码,和机子性能无关和固件无关,那大概率是ssr-plus导致的.

rangerwill, 2020-09-25:
> 
> 
> 我也偶尔遇到这种情况, service firewall restart 后才能恢复. 我用的是I3的机子固件是官方源码,和机子性能无关和固件无关,那大概率是ssr-plus导致的.

算上新装的i5 6400,
我家里三台机子都遇到了这样的情况，
现在我只能给crontab里加一个每三小时重启防火墙的命令来解决这个问题

rangerwill, 2020-09-25:
service firewall restart 之后，
网络能暂时恢复正常，
但是过一段时间，就又无法访问墙内网站了，
目前的解决方案就是crontab 每三小时重启一次firewall 进程

afeiy, 2020-09-25:
目前在用的N1，红米ac2100，x86均无此问题，应该不是ssrp的问题。

rangerwill, 2020-09-29:
> 
> 
> 目前在用的N1，红米ac2100，x86均无此问题，应该不是ssrp的问题。除了N1要添加防火墙规则才可访问国内（因我把eth0拿来做wan才需要），2100和x86就用默认的防火墙规则就ok。

老哥 你用的是最新版本的吗？还是半个月以上的老版本？

afeiy, 2020-09-30:
> > 目前在用的N1，红米ac2100，x86均无此问题，应该不是ssrp的问题。除了N1要添加防火墙规则才可访问国内（因我把eth0拿来做wan才需要），2100和x86就用默认的防火墙规则就ok。
> 
> 老哥 你用的是最新版本的吗？还是半个月以上的老版本？
lean的09.15源码，到今天刚好半个月。

rangerwill, 2020-09-30:
> 
> 
> > > 目前在用的N1，红米ac2100，x86均无此问题，应该不是ssrp的问题。除了N1要添加防火墙规则才可访问国内（因我把eth0拿来做wan才需要），2100和x86就用默认的防火墙规则就ok。
> > 
> > 
> > 老哥 你用的是最新版本的吗？还是半个月以上的老版本？
> > lean的09.15源码，到今天刚好半个月。

那就不奇怪了、 这个问题是近半个月的新版本出现的   我觉得我有必要继续用旧版本了

afeiy, 2020-09-30:
> > > > 目前在用的N1，红米ac2100，x86均无此问题，应该不是ssrp的问题。除了N1要添加防火墙规则才可访问国内（因我把eth0拿来做wan才需要），2100和x86就用默认的防火墙规则就ok。
> > > 
> > > 
> > > 老哥 你用的是最新版本的吗？还是半个月以上的老版本？
> > > lean的09.15源码，到今天刚好半个月。
> 
> 那就不奇怪了、 这个问题是近半个月的新版本出现的 我觉得我有必要继续用旧版本了

今天拉取lean的源码编译的x86物理机安装依然是ok的。

rangerwill, 2020-09-30:
> 
> 
> N1要添加防火墙规则：iptables -t nat -I POSTROUTING -j MASQUERADE

[image]

作为旁路由才需要加这个规则吧......?

rangerwill, 2020-10-02:
[image]

rangerwill, 2020-10-02:
[image]

rangerwill, 2020-10-08:
[image]
在系统-启动项中把firewall服务启用之后再重启貌似没有再次出现这个问题

rangerwill, 2020-10-21:
> 
> 
> [image]
> 在系统-启动项中把firewall服务启用之后再重启貌似没有再次出现这个问题

还是存在这个问题   .....

rangerwill, 2020-11-02:
service mwan3 stop
service mwan3 disable

done.

closed 2020-11-02 by rangerwill