coolsnowwolf/lede #5521: 请问大神怎么编译smba4呀，smba和smba4同时选上会编译失败，但是smba默认是勾选上的。。

759071793, 2020-09-22:
请问大神怎么编译smba4呀，smba和smba4同时选上会编译失败，但是smba默认是勾选上的。。
[image]

759071793, 2020-09-22:
补上错误日志：
<pre>
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
 * 	samba4-libs
 * opkg_install_cmd: Cannot install package luci-app-samba4.
 * check_conflicts_for: The following packages conflict with samba4-server:
 * check_conflicts_for: 	samba36-server * 
 * opkg_install_cmd: Cannot install package samba4-server.
package/Makefile:65: recipe for target &apos;package/install&apos; failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory &apos;/home/song/lede&apos;
package/Makefile:108: recipe for target &apos;/home/song/lede/staging_dir/target-x86_64_musl/stamp/.package_install&apos; failed
make[1]: *** [/home/song/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory &apos;/home/song/lede&apos;
/home/song/lede/include/toplevel.mk:222: recipe for target &apos;world&apos; failed
make: *** [world] Error 2
</pre>

Dimitri2020007, 2020-09-22:
samba和samba4只能2选1，两个都选编译时会有冲突

759071793, 2020-09-22:
> samba和samba4只能2选1，两个都选编译时会有冲突

谢谢，已经直接修改.config配置文件，重新编译中，。

Dimitri2020007, 2020-09-22:
那个软件包名是samba36开头的。已经选择了server好像取不掉。就在menuconfig的network项里。

759071793, 2020-09-22:
> 那个软件包名是samba36开头的。已经选择了server好像取不掉。就在menuconfig的network项里。

哇，我看到了，我
<pre>
CONFIG_PACKAGE_samba36-server=y
CONFIG_PACKAGE_SAMBA_MAX_DEBUG_LEVEL=-1
</pre>
这两个没有取消掉。。郁闷。。

closed 2020-09-22 by 759071793