coolsnowwolf/lede #5524: 编译过程错误请教（二）

recinfo, 2020-09-22:
继续，到了./scripts/feeds install -a这一步，这步以上正常，开始就提示如下：

WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'antfs-mount', which does not exist

下面还有好多，就不贴了，均是类似警告。

这该如何处理阿，不敢往下进行了。还是无视它继续进行下去。

759071793, 2020-09-22:
这个不用管它，不会影响你编译

Dimitri2020007, 2020-09-22:
这些警告是正常的不用管

recinfo, 2020-09-22:
谢谢回复

closed 2020-09-23 by coolsnowwolf