coolsnowwolf/lede #5531: FaceTime等UDP异常

SamuelZon258, 2020-09-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用facetime拨打电话的时候先是接通,然后1秒左右断开,20秒左右facetime才重连上
      使用tcpdump抓包lan口和pppoe拨号口发现都是 UDP只有本地发送给对方的数据,没收到任何UDP返回数据
      断线重连时会出发请求苹果,然后触发重连后就正常有往返UDP数据了

PS：本地申请了公网IP，对方应该还是大内网
        酸酸乳的UDP加速不管是关了还是开启貌似都没用(JC也不支持UDP加速)
        已将苹果相关服务加入酸酸乳白名单,规则是使用的lhie1的

### (2) 路由器型号和固件版本

A：Intel(R) Celeron(R) CPU N3160 @ 1.60GHz : 4 Core 4 Thread (CpuMark : 30717.140513 Scores)
      OpenWrt R20.8.27 / LuCI Master (git-20.223.26773-d18ef13)
      网卡是8111

### (3) 详细日志

A：不太清楚日志如何提取,如有需要时再提取

SamuelZon258, 2020-09-28:
出现这个问题的时候对方是可以收到我的数据看到我动不过我收不到对方数据
而且还有奇怪的现象就电脑每次完全断电后再开机会出现无法上网
禁用一下以太网再启动又好了,每次都这样.

SamuelZon258, 2020-09-29:
装了openwrt 19.07.4 打facetime没问题了...

closed 2020-10-09 by SamuelZon258

404user, 2020-12-06:
终于有人报出来这个BUG了，跟我情况一样！！！我是每次FaceTime的接通的时候卡1-3秒，然后断wi-fi 用4G 通话了。

hnewcity, 2021-02-01:
Me too. Can not find any solution.

5237216, 2023-03-23:
Me too,need help!!!