coolsnowwolf/lede #5532: v2ray 4.28.2编译报错

Dimitri2020007, 2020-09-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

在线编译v2ray4.28.2报错

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：下载v2ray4.28.2出错，使用脚本编译

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：2020-09-23T12:33:41.7306319Z make[3]: Entering directory '/workdir/openwrt/package/lean/v2ray'
2020-09-23T12:33:41.8638220Z mkdir -p /workdir/openwrt/dl
2020-09-23T12:33:41.8667543Z SHELL= flock /workdir/openwrt/tmp/.v2ray-core-4.28.2.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "v2ray-core-4.28.2.tar.gz" "e5d6603454622e1f466c4c20061992046124fc7c" "" "https://codeload.github.com/v2fly/v2ray-core/tar.gz/v4.28.2?"    '
2020-09-23T12:33:41.8852591Z Cannot find appropriate hash command, ensure the provided hash is either a MD5 or SHA256 checksum.
2020-09-23T12:33:41.8863373Z make[3]: *** [/workdir/openwrt/dl/v2ray-core-4.28.2.tar.gz] Error 255
2020-09-23T12:33:41.8864613Z Makefile:308: recipe for target '/workdir/openwrt/dl/v2ray-core-4.28.2.tar.gz' failed
2020-09-23T12:33:41.8867821Z make[3]: Leaving directory '/workdir/openwrt/package/lean/v2ray'
2020-09-23T12:33:41.8868499Z time: package/lean/v2ray/compile#0.12#0.03#0.16
2020-09-23T12:33:41.8869220Z make[2]: *** [package/lean/v2ray/compile] Error 2
2020-09-23T12:33:41.8937590Z make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2020-09-23T12:33:41.8938469Z package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
2020-09-23T12:33:41.8939763Z make[2]: Leaving directory '/workdir/openwrt'
2020-09-23T12:33:41.8940837Z package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
2020-09-23T12:33:41.8941780Z make[1]: Leaving directory '/workdir/openwrt'
2020-09-23T12:33:41.8949758Z /workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed

alisonY, 2020-09-23:
手动下一个 丢到dl目录试试

niceran, 2020-09-23:
把正确的MD5替换上去1D121D328B25108D67228B95B73C128E

759071793, 2020-09-23:
> 把正确的MD5替换上去SHA256: E78446333FD28EF8A3EC2C25612458BFC024A226008888ED5277201136F4CA97

请问在哪里文件替换，可否发一下目录。。。

Myoko, 2020-09-24:
hash 是错的！你手动替换一下正确的就可以编译了

niceran, 2020-09-24:
> > 把正确的MD5替换上去SHA256：E78446333FD28EF8A3EC2C25612458BFC024A226008888ED5277201136F4CA97
> 
> 请问在哪里文件替换，可否发一下目录。。。
目录源码路径：lede/package/lean/v2ray/Makefile
改代码
PKG_VERSION:=4.27.5
PKG_HASH:=f289d8d85ab0851851a6e3c101226e77bed0052fd60f9185df8852b601e657f8
就可以了。
【编译脚本不兼容v2ray-core-4.28.2】

closed 2020-09-24 by coolsnowwolf