coolsnowwolf/lede #5534: 谁能帮我看看这个是什么错误，如何修正

xiangfeigao, 2020-09-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
编译错误哦

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：[/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/fullconenat-2019-10-21-0cf3b48f/.prepared_c8e9e94196d8408caa1126759b90871f_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/lede/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.05#0.02#0.31
package/Makefile:111: recipe for target 'package/lean/openwrt-fullconenat/compile' failed
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory '/lede/lede'
package/Makefile:107: recipe for target '/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/lede/lede'
/lede/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
ub@ubuntu:/lede/lede$

Dimitri2020007, 2020-09-24:
fullconenat软件包下载完整了吗？

xiangfeigao, 2020-09-24:
我想应该找到原因了，/p910nd-0.97.tar.bz2下载地址失效了，希望大雕更新

xiangfeigao, 2020-09-24:
> fullconenat软件包下载完整了吗？

请问一下，就是这些download的软件下载链接地址存放在哪个文件中。告诉我路径，我可以修改下

xiangfeigao, 2020-09-24:
> fullconenat软件包下载完整了吗？

下完整了

GMAB007, 2020-09-24:
同样的错误信息，楼主解决了吗？

Dimitri2020007, 2020-09-24:
下载目录是lede\dl，可能是源码的问题

closed 2021-06-20 by coolsnowwolf

netlovehf, 2022-01-12:
一样的问题
make[3]: Entering directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/netlove/lede/dl
SHELL= flock /home/netlove/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/home/netlove/lede/scripts/download.pl "/home/netlove/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || (  /home/netlove/lede/scripts/dl_github_archive.py --dl-dir="/home/netlove/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/netlove/lede/tmp/dl && cd /home/netlove/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/netlove/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53_check
. /home/netlove/lede/include/shell.sh; xzcat /home/netlove/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
cp -fpR ./files/Makefile /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
rm -f /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.guess; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/li [... cut ...]
make[3]: Leaving directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.74#0.34#1.14
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/netlove/lede'
make[1]: *** [package/Makefile:110: /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/netlove/lede'
make: *** [/home/netlove/lede/include/toplevel.mk:230: world] Error 2