coolsnowwolf/lede #5537: 关于无线设置，配置界面少了接口配置的问题，这是驱动的问题，还是我漏选了软件包的问题？

cn-404-not-found, 2020-09-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
关于无线设置，配置界面少了接口配置的问题，
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
不好描述，看图吧，刷了几个固件，发现无线的设置不一样，这是驱动的问题吗，我是需要用第二张图那种版本，需要做接口配置，重点是我要将无线指定网络，但是部分固件又没有这个功能，这个应该在编译的时候如何选择？是驱动的问题吗？

[image]

[image]

AmadeusGhost, 2020-09-25:
图1是闭源驱动的界面，图2是开源驱动的界面，不一样的。

cn-404-not-found, 2020-09-25:
> 图1是闭源驱动的界面，图2是开源驱动的界面，不一样的。

那么，应该如何选择开源驱动编译呢，我是K2P，取消勾选MTWIFI吗？还是需要在内核里面选择其他驱动？还是需要加载什么包？谢谢

AmadeusGhost, 2020-09-25:
> 那么，应该如何选择开源驱动编译呢，我是K2P，取消勾选MTWIFI吗？还是需要在内核里面选择其他驱动？还是需要加载什么包？谢谢

k2p 的话开源驱动还只是初步支持啊，你确定？

cn-404-not-found, 2020-09-27:
> > 那么，应该如何选择开源驱动编译呢，我是K2P，取消勾选MTWIFI吗？还是需要在内核里面选择其他驱动？还是需要加载什么包？谢谢
> 
> k2p 的话开源驱动还只是初步支持啊，你确定？

我的截图是红米AC2100，我K2P和红米都有，都希望有这些功能，如果K2P不支持，也希望红米能用上这个功能，那么，如果AC2100要编译这个功能，应该如何加入开源驱动呢，因为网上找到的包，功能并不合适，我希望自己编译

AmadeusGhost, 2020-09-29:
去掉 kmod-mt7603e / kmod-mt7612e / mt_wifi / luci-app-mtwifi，然后选择 kmod-mt7603 / kmod-mt7615e / kmod-mt7615-firmware / wpad-openssl

nanchuci, 2020-10-16:
开源驱动设置：
CONFIG_PACKAGE_kmod-mt7603=y
CONFIG_PACKAGE_kmod-mt7603e=n
CONFIG_PACKAGE_kmod-mt76x2=y
CONFIG_PACKAGE_kmod-mt76x2-common=y
CONFIG_PACKAGE_kmod-mt76x2e=n
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wpa-supplicant=y

closed 2021-06-20 by coolsnowwolf