coolsnowwolf/lede #5539: fullconenat编译出错，大家碰到了吗，hc5962

xiangfeigao, 2020-09-24:
ERROR: module '/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/fullconenat-2018-12-15-d4daedd0/xt_FULLCONENAT.ko' is missing.
Makefile:65: recipe for target '/lede/lede/bin/targets/ramips/mt7621/packages/kmod-ipt-fullconenat_5.4.61+2018-12-15-d4daedd0-1_mipsel_24kc.ipk' failed
make[2]: *** [/lede/lede/bin/targets/ramips/mt7621/packages/kmod-ipt-fullconenat_5.4.61+2018-12-15-d4daedd0-1_mipsel_24kc.ipk] Error 1
make[2]: Leaving directory '/lede/lede/package/fullconenat'
time: package/fullconenat/compile#0.08#0.00#0.35
package/Makefile:111: recipe for target 'package/fullconenat/compile' failed
make[1]: *** [package/fullconenat/compile] Error 2
make[1]: Leaving directory '/lede/lede'

swxk521, 2020-09-24:
没有碰到,正常

Dimitri2020007, 2020-09-24:
请不要为了图快使用gitee去复制lean大的lede源码加速编译，最后会出现很多莫名奇妙的错误。我曾经也是图快用gitee克隆了L大源码加速编译，结果不是文件丢失就是其他错误。

建议搭好梯子以后从github的L大源码重新克隆一遍再编译。
在这之前请备份好lede/dl目录。
你不会想重新下载dl的
下载dl的过程很慢很痛苦的。

pangfeng161, 2020-09-24:
> 请不要为了图快使用gitee去复制lean大的lede源码加速编译，最后会出现很多莫名奇妙的错误。我曾经也是图快用gitee克隆了L大源码加速编译，结果不是文件丢失就是其他错误。
> 
> 建议搭好梯子以后从github的L大源码重新克隆一遍再编译。
> 在这之前请备份好lede/dl目录。
> 你不会想重新下载dl的
> 下载dl的过程很慢很痛苦的。

梯子线路好基本不成问题,但备份dl库很有必要

Dimitri2020007, 2020-09-25:
重点是不要图快用gitee，编译过程会坑哭你。

xiangfeigao, 2020-09-25:
谢谢各位建议，今日重来

xiangfeigao, 2020-09-25:
> 请不要为了图快使用gitee去复制lean大的lede源码加速编译，最后会出现很多莫名奇妙的错误。我曾经也是图快用gitee克隆了L大源码加速编译，结果不是文件丢失就是其他错误。
> 
> 建议搭好梯子以后从github的L大源码重新克隆一遍再编译。
> 在这之前请备份好lede/dl目录。
> 你不会想重新下载dl的
> 下载dl的过程很慢很痛苦的。

能否共享一下dl文件，其实建议大家建立一个共享文件，这样国内就能提速了

Dimitri2020007, 2020-09-25:
国内是云都不可信，原因你懂的，这里不能多说。

yinzhefuyun, 2020-09-26:
> > 请不要为了图快使用gitee去复制lean大的lede源码加速编译，最后会出现很多莫名奇妙的错误。我曾经也是图快用gitee克隆了L大源码加速编译，结果不是文件丢失就是其他错误。
> > 建议搭好梯子以后从github的L大源码重新克隆一遍再编译。
> > 在这之前请备份好lede/dl目录。
> > 你不会想重新下载dl的
> > 下载dl的过程很慢很痛苦的。
> 
> 能否共享一下dl文件，其实建议大家建立一个共享文件，这样国内就能提速了

dl是用多少下载多少 x86 arm misp 用的都不一样

Dimitri2020007, 2020-09-26:
取决于你要编译的机型和软件应用数量。
我的dl已经4个G了1千多个软件。新旧版本的都有

xiangfeigao, 2020-09-26:
我认为4个g如果放网盘里也比较快，主要是持续更新与分类，一个电视连续剧还占多呢，

mactavishmeng, 2020-11-26:
我觉得这个不是dl库的问题：
1. 我反复下载编译了多次（有梯子），每次都遇到这个同样的错误，dl库中相应的包存在且能打开；
2. 看报错的内容，是少了xt_FULLCONENAT.ko文件。进入对应的文件夹查看确实没有该文件，可能是压根就没有编译。
由于不太懂固件编译的过程，各种Makefile文件互相嵌套没太看明白，直观感受就是这个模块的编译配置有问题，而不是缺依赖。

希望能提供点思路，有没有大佬能看下真实原因

closed 2021-06-20 by coolsnowwolf

netlovehf, 2022-01-12:
make[3]: Entering directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/netlove/lede/dl
SHELL= flock /home/netlove/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/home/netlove/lede/scripts/download.pl "/home/netlove/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || (  /home/netlove/lede/scripts/dl_github_archive.py --dl-dir="/home/netlove/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/netlove/lede/tmp/dl && cd /home/netlove/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/netlove/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53_check
. /home/netlove/lede/include/shell.sh; xzcat /home/netlove/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
cp -fpR ./files/Makefile /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
rm -f /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.guess; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -zno [... cut ...]
make[3]: Leaving directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.74#0.34#1.14
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/netlove/lede'
make[1]: *** [package/Makefile:110: /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/netlove/lede'
make: *** [/home/netlove/lede/include/toplevel.mk:230: world] Error 2

zoverdoser, 2022-07-30:
> 我觉得这个不是dl库的问题：
> 
> 1. 我反复下载编译了多次（有梯子），每次都遇到这个同样的错误，dl库中相应的包存在且能打开；
> 2. 看报错的内容，是少了xt_FULLCONENAT.ko文件。进入对应的文件夹查看确实没有该文件，可能是压根就没有编译。
>    由于不太懂固件编译的过程，各种Makefile文件互相嵌套没太看明白，直观感受就是这个模块的编译配置有问题，而不是缺依赖。
> 
> 希望能提供点思路，有没有大佬能看下真实原因

请问后边解决这个问题了么？我遇到了相同的问题