coolsnowwolf/lede #5544: openwrt稳定版更新

pangfeng161, 2020-09-24:
openwrt另一个分支稳定版好久不更新,新内核感觉不怎样,不知道什么时候能更新

Dimitri2020007, 2020-09-25:
你指的是Lienol的openwrt吗？
他的openwrt内核版本比较稳定。
更新时间不太确定。

PassWall-OpenWrt, 2020-09-25:
**稳定版不会加新的功能，
只会有安全性更新，
内核也不会变，
例如 v2ray。
驱动和内核都不会变。**
所以你指的**更新**是更新什么东西？

新内核感觉不怎样
你到底要怎样？
或者你可以自己更新自己想要的东西的。

hxlls, 2020-09-26:
没想明白，为啥要把miui 的没事更新一把的坏习惯带到路由上来？又不是手机，还能更新下广告。在不增加功能的前提下，修复BUG就行了，刷版本号有啥意思？稳定才是路由最重要的，有你现在要用到的功能就行了，一堆路由的内核还是2.6的内核，不也用得欢？比如梅林的内核版本就不高

Dimitri2020007, 2020-09-26:
lean大openwrt内核版本5.4更高
openwrt官版内核还在4.14呢
稳定实用是最重要的
其他的都不重要

pangfeng161, 2020-09-26:
> lean大openwrt内核版本5.4更高
> openwrt官版内核还在4.14呢
> 稳定实用是最重要的
> 其他的都不重要

确实稳定最重要,想编译4.14版本但不知如何着手编译

Dimitri2020007, 2020-09-26:
vmware上装ubuntu的server版本。
搭好梯子，线路要足够稳定，要全局。客户端开启局域网共享。
vmware装好ubuntu设置一下主机代理，go语言的代理。
进lienol的openwrt网页，按照他的提示操作。
就有4.14版本的固件了

pangfeng161, 2020-09-26:
> vmware上装ubuntu的server版本。
> 搭好梯子，线路要足够稳定，要全局。客户端开启局域网共享。
> vmware装好ubuntu设置一下主机代理，go语言的代理。
> 进lienol的openwrt网页，按照他的提示操作。
> 就有4.14版本的固件了

谢谢,知道Lienol可以编译4.14版本,想知道leda怎样编译4.14版本,修改过文件但编译失败

qq1316237299, 2020-09-26:
> > vmware上装ubuntu的server版本。
> > 搭好梯子，线路要足够稳定，要全局。客户端开启局域网共享。
> > vmware装好ubuntu设置一下主机代理，go语言的代理。
> > 进lienol的openwrt网页，按照他的提示操作。
> > 就有4.14版本的固件了
> 
> 谢谢,知道Lienol可以编译4.14版本,想知道leda怎样编译4.14版本,修改过文件但编译失败

lean取消了好像

pangfeng161, 2020-09-26:
> > > vmware上装ubuntu的server版本。
> > > 搭好梯子，线路要足够稳定，要全局。客户端开启局域网共享。
> > > vmware装好ubuntu设置一下主机代理，go语言的代理。
> > > 进lienol的openwrt网页，按照他的提示操作。
> > > 就有4.14版本的固件了
> > 
> > 
> > 谢谢,知道Lienol可以编译4.14版本,想知道leda怎样编译4.14版本,修改过文件但编译失败
> 
> lean取消了好像

好可惜,希望加回来

Dimitri2020007, 2020-09-27:
你不知道lean的openwrt版本很激进吗？软件内核都是最新的版本。更的是openwrt的snapshot版本。
回来那只是可能了。
不要抱太多期望就是了。

badcrazy, 2020-09-30:
lean不是把老版本放https://github.com/coolsnowwolf/openwrt/tree/lede-17.01了吗，编译这个吧

Dimitri2020007, 2020-10-01:
谢谢上面那位。想编译老版本的找这个

https://github.com/coolsnowwolf/openwrt.git

和openwrt一样的版本都是19.07

这个lede用的是openwrt的snapshot版本

hilobekerfulaisien, 2020-10-15:
虽然我不是电子电路专业的，为啥不直接做成package，大家安装官方openwrt再搜就可以了？

closed 2021-06-20 by coolsnowwolf