coolsnowwolf/lede #5545: 文件系统出错，插入USB硬盘后重启就死机

mkcczh, 2020-09-25:
各位大佬，遇到疑难杂症了，请帮帮忙

刷了恩山便宜的LEAN大的OPENWRT, 挂载硬盘出状况，开机后用指令mount挂载ext4、ntfs-3g挂载ntfs都能正常工作，但是重启只要USB口有硬盘（ext4、ntfs都试过一样）就死机，IP地址都分配不了，更进不了系统，拔掉USB硬盘重启又没有问题了，不知道是什么回事， 。我把系统msata硬盘换了试，一样不行。怀疑是文件系统或者分区表有错误，所以启动时自动挂载出错，试着把自动挂载选项都去掉，还是不行。启动项内加入指令也不行，加了指令直接进不了系统，要重刷才能进。

我每次都是用PE删除所有分区再格式化ext4，physdiskwrite -u 写盘img，每次都有最后几个字节出错写不完，但能开机，我后来用parted -l 修复。但分区表系统还是有错误信息。

Device      Start     End Sectors  Size Type
/dev/sda1     512   66047   65536   32M EFI System
/dev/sda2   66560 1704959 1638400  800M Microsoft basic data
/dev/sda128    34     511     478  239K BIOS boot
Partition table entries are not in disk order.

这个错误我网上查了下，不敢修，怕影响开机分区配置。挂载点里面ＵＵＩＤ也有错误，／ｂｏｏｔ和挂载ｕ盘都显示ＵＵＩＤ不存在，不能用ＵＵＩＤ自动挂载，只能用指令。

试了两个不同人编译的LEAN大的固件，都一样的问题，怀疑是固件的问题，不知道有没有大佬知道是什么回事？？

mkcczh, 2020-10-04:
刷回OpenWrt R20.7.1 GDQ v7.1[2020], gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2668-c0f3364)，也就是7月前的版本，一切顺畅，ntfs 和 EXT4 的USB硬盘即插即用，重启也认盘。刷了两个不同出处编译LEAN的版本只要是7月前的都可以用。之后的版本，到10月1号的为止都不行，应该是固件的问题吧，怀疑可能是新固件的驱动或者分区表的调整问题。mark一下，给以后遇到同样问题的人一个例子。

closed 2020-10-04 by mkcczh

lanarap, 2021-11-24:
> 各位大佬，遇到疑难杂症了，请帮帮忙
> 
> 刷了恩山编译的LEAN大的OPENWRT, 挂载硬盘出状况，开机后用指令mount挂载ext4、ntfs-3g挂载ntfs都能正常工作，但是重启只要USB口有硬盘（ext4、ntfs都试过一样）就死机，IP地址都分配不了，更进不了系统，拔掉USB硬盘重启又没有问题了，不知道是什么回事， 。我把系统msata硬盘换了试，一样不行。怀疑是文件系统或者分区表有错误，所以启动时自动挂载出错，试着把自动挂载选项都去掉，还是不行。启动项内加入指令也不行，加了指令直接进不了系统，要重刷才能进。
> 
> 我每次都是用PE删除所有分区再格式化ext4，physdiskwrite -u 写盘img，每次都有最后几个字节出错写不完，但能开机，我后来用parted -l 修复。但分区表系统还是有错误信息。
> 
> Device Start End Sectors Size Type /dev/sda1 512 66047 65536 32M EFI System /dev/sda2 66560 1704959 1638400 800M Microsoft basic data /dev/sda128 34 511 478 239K BIOS boot Partition table entries are not in disk order.
> 
> 这个错误我网上查了下，不敢修，怕影响开机分区配置。挂载点里面ＵＵＩＤ也有错误，／ｂｏｏｔ和挂载ｕ盘都显示ＵＵＩＤ不存在，不能用ＵＵＩＤ自动挂载，只能用指令。
> 
> 试了两个不同人编译的LEAN大的固件，都一样的问题，怀疑是固件的问题，不知道有没有大佬知道是什么回事？？

> 刷回OpenWrt R20.7.1 GDQ v7.1[2020], gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2668-c0f3364)，也就是7月前的版本，一切顺畅，ntfs 和 EXT4 的USB硬盘即插即用，重启也认盘。刷了两个不同出处编译LEAN的版本只要是7月前的都可以用。之后的版本，到10月1号的为止都不行，应该是固件的问题吧，怀疑可能是新固件的驱动或者分区表的调整问题。mark一下，给以后遇到同样问题的人一个例子

> 刷回OpenWrt R20.7.1 GDQ v7.1[2020], gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2668-c0f3364)，也就是7月前的版本，一切顺畅，ntfs 和 EXT4 的USB硬盘即插即用，重启也认盘。刷了两个不同出处编译LEAN的版本只要是7月前的都可以用。之后的版本，到10月1号的为止都不行，应该是固件的问题吧，怀疑可能是新固件的驱动或者分区表的调整问题。mark一下，给以后遇到同样问题的人一个例子。

老哥解决了吗？我也是前几个月的固件没问题，现在的只要挂了硬盘，重启就是unknown filrlsystem，卡在grub那里。