coolsnowwolf/lede #5546: 最新编译openwrt固件UPnP对活动的重定向不能正常显示

shu0802, 2020-09-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 最新编译openwrt固件UPnP对活动的重定向不能正常显示

A：

### (2) 路由器型号和固件版本:X86_64 ,

内核版本 | 5.4.61
-- | --

固件版本 | OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)
-- | --

A：

### (3) 详细日志

A：
[image]
[image]

Dimitri2020007, 2020-09-25:
你提供的日志无upnp的报错啊，没有启用额外的日志啊，有额外日志就能输出upnp的错误信息了啊

shu0802, 2020-09-25:
> 你提供的日志无upnp的报错啊，没有启用额外的日志啊，有额外日志就能输出upnp的错误信息了啊
> 另外用迅雷随便下载一个文件再看看upnp有没有端口映射啊

这个是不是被关闭了，要如何处理？
[image]
[image]

Dimitri2020007, 2020-09-25:
你的upnp应该是正常的，要是不相信的话你可以打开路由器的upnp管理界面。然后打开迅雷。随便下载一个文件，就下载个centos。在这过程中观察一下是否有活动的upnp重定向端口。

shu0802, 2020-09-26:
> 你的upnp应该是正常的，要是不相信的话你可以先打开路由器的upnp管理界面。再打开迅雷。随便下载一个文件，一定要bt下载。在这过程中观察一下是否有活动的upnp重定向端口。刷新下upnp管理界面就是了。好了就可以关闭日志功能节省空间。

好的，我再试试。只是原来用tr下载，现在同样用tr下载，就有映射出去的端口在这里显示，现在没有了。

closed 2020-09-26 by shu0802