coolsnowwolf/lede #5553: 请问我用windows 的Ubuntu子系统编译N1的OP过程中出现以下错误该如何解决

Cxp1220, 2020-09-26:
touch /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61/.image
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/lg/lede/target/linux/armvirt/image'
bash: *1024*1024: syntax error: operand expected (error token is "*1024*1024")
make[5]: Nothing to be done for 'compile'.

Ljzkirito, 2020-09-26:
不要用wsl，兼容性不好。
装个虚拟机，或者用multipass[https://github.com/canonical/multipass/releases](url)
我用的multipass，和Ubuntu虚拟机一样的

Dimitri2020007, 2020-09-26:
wsl是不支持文件系统大小写识别的。劝你还是装个debian或ubuntu虚拟机编译。
即使你能成功解决wsl编译的问题，时间上也是比虚拟机慢很多的。

yinzhefuyun, 2020-09-26:
如果是固态的话试一试把wsl切换成wsl2看看行不行

Dimitri2020007, 2020-09-26:
首先搞定了wsl2的文件系统大小写识别

Cxp1220, 2020-09-27:
> wsl是不支持文件系统大小写识别的。劝你还是装个debian或ubuntu虚拟机编译。
> 即使你能成功解决wsl编译的问题，时间上也是比虚拟机慢很多的。

我换了虚拟机也是一样的问题

yinzhefuyun, 2020-09-27:
> > wsl是不支持文件系统大小写识别的。劝你还是装个debian或ubuntu虚拟机编译。
> > 即使你能成功解决wsl编译的问题，时间上也是比虚拟机慢很多的。
> 
> 能编译出来，但是我换了虚拟机也是一样出现这个问题
我记得是可以的，至少wsl2是可以的
<img src="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAA+EAAAJvCAYAAAAUdND6AAAAAXNSR0IArs4c6QAAAARnQU1BAACxjwv8YQUAAAAJcEhZcwAADsMAAA7DAcdvqGQAAOfXSURBVHhe7P0JfF1Xfeh9S2eSZNPblsLltqXMQ5haOtACSax5locApYW2lKEtM4SUeWiBAIEyBSghZI4T2/EkW7ZsybNsWZLtxPEcx3EmO3FmAvdzn/vc+77vc5//e9aWt728stZee5+zNZ6f9fl+tPd/WHvrSCXn333OPlUjw8MyvGuXDO/cKUM7dsj2bdtk65YtMjgwIH/03q8AAAAAAIAEatvWSa65T3It66WmY5PUdW0t2lbc3ipVu4sD+K7iAL5raEh2bN8eDOCbBgelv78/aC50fRoAAAAAAMSg5uhswwrJNa2RTNNaybb0F22UbOuA1HVuk6rhnbtkZ3EAV1fBd2zfIduKQ/jAxo2yft06hnAAAAAAABJQc3T1pUuKQ3hvcDW80F4cwIuDeKZ5g9R1bVdDuLoKvjO4Cr5961bZvHmzbCwO4X19fQzhAAAAAAAkoObomqaVkm1YKbVt66W2Y0DyrRulpnOzzOnZIVVqAN+5Y0i2q5eiF4fwTZs2yYYNG2TNmjUM4QAAAAAAJKDm6Oy8JZJrXClzOvqlpq1fci39wSCuXpZetVPdkK04gAfvBy8O4YPq/eDr10tvby9DOAAAAAAACQRXwhuWSLZ+mdS19UltuxrA1xf1S1Vjn1QNqRuyqbuin70p28DAgKwvDuGrV69mCAcAAAAAIAE1R89tvUNqW1ZKrnGV5FvWypzOAZnbvVkK7QNnh/DiAL5t2zbZErwffEDW9fXJqlWlDeF/3PMBuWpRkwy/44/loXe+KqC2r1rYJG/q/qC1BwAAAACA2UDN0dVvv0HyDUuk0FwcwpvXFAfxdcGV8JqO4hCuroKPvxT9wiF85cpViYbw53V9Ur6/sFEef9fL5cl3vczqsWLuP4oD+txirW0NAAAAAABmMjVHZy+5UTKX3CqFppVSaFkjmYbVUt3QK4X2DVKlPpps/KZs22RzcFO2jdK3tk9WrFgZewhXA/jad/yFdfBWnvqHN8r/WPwdeeYjbw/211z2FgZxAAAAAMCso+bomvpbgiE8c+kSyTetklxzr+Rb+oLPDa86eOiIHDx0WO4+cEjuvGu/jI7tlaGhYdm0ZUvsIfwHixqeM3g/8/F58t9/+hn571d/Wv7HLVeK+vc/+288l//ewkbrWkjRRwbkmDwuV37EkrvAzbJKDsh7rDkAAAAA+HQw08X5Z+udSncdf1A++P3F1pyicqrGliuFmqMzb7tOahpul0LjHcUhfKXUtK2R3+rZGLws/dwQvv/ug7Lvzv2ye3SvLFm+Wn5xw3WxhnD1HvALXoL+7lcGw/f/vmvb2V+ByP8e2yj/96bb5JlPNp6rUy9Nf2Ox17bmrBIMwhf+O7bkqpi1cQboCBcM4VfJlafOLnv236rvhLXpD+FvXvL42aNc+O/CY4b/Lvw5n9M7fPO5XFQfAAAAgImj/tniujg1k00N2f/f//2/rIN4VK5U40P4tZK79BbJNyyVbMNyyTWtlrr2fim0rR8fwg8c1IfwPfKpz3xaLvvrhbGGcHUTtnMDeNH/PbhY5P/8P/LsV98lv7qiXX716Rb5n2uvFfl//88FV8KVby9qtq4584wPuOcHzHHjw+Rzh9v3DKs/TSP+nQPFmDFUnh3KzXXd7OdxPqetf8GAPglXwoPjhccYP89z/8+I4Gc/f3z1+Nj/HxXRfQAAAAAmjvpni+vi1EwF27A9EQO4oubo/CU3SObiGyU77/bgfeHqLunV9askq16OrgbwYAjff0D27rtLRkZG5cYbr5FPfvJ9sYbw3e9407mh+ukPv1Xk//n/yf+18qcXDNvK/+z7ZXEQ/3/lmY/Xn4vteuebrGvOPJbhVw2IpwbkzeF28O+ArCoOmKouGNDPXeEdv7prHZ4TDZoJhvCi88PuxA/h6ljnzuuCgVy58LwvqNV5+gAAAABMHPXPFtfFqZkq+tA9UQO4EgzhF18vhfrFkqtfIrmG5VI9b7nUtPZJjboS/pwhfHRM+jeskN7l18Qawh985yvl6Q/+mfyf3zwt/2vXGvl/njwdvB9cH8AVFfv/HNotv/76e8/FHnjXq6xrjtNfdqz+2YavC2suHMYuzJ2/sqrixWF0STgYn89deAV2vP/cmueGauPYxcH2gpeQnx28zw2SFwyOZ49tXn3WB/bn0Huifu4LY8F6z7n6HHcIv3Ct8DExr+yr/gtfKu5g/nxq3+g7fy7jP+P5f9p5RfUFP+uFPx8AAACA9Kh/trguTs1UCofviRrAFTVHZ956rWTefpPUNi+XfOOKYAif27lBci19tivhY7J1y3JZe8f3Yw3hapAOhvBfPyX/a2c4hJ+/2h1SMXMIv/+driF8fAg8PxAXqQHsgoFS/XqNAe3cvurXB7JweA1zxX/hMKcNbxdcnS6ud0wN2GfP4blDonbsC9ZXzg+1as3zP4eq03+Gs9vq3J1DuL625diWx+XceUQN4RcMrefPN/qx0x6HyHPWGedUdOGrAMadf3zPx5Sg9uxxIvt8Q3iQP/tP+x2b6wEAAACTIe4/W+9UiXM+cWqm0qQN4W+7VrKXFGeNhqXFIXy5FFrU3dHXSKax97lDuHpP+NoNd8iy238YawjXX46urnarl5z/X6t+di4W+p991wXvC9evkjtfjm4d8PRh7rmD3QWxYDB97r/xIc82ZJ7d14bW9wyrWLE2OA+txnfsYP/CIdwWv2A7cqBNcmwj/5wh/OwDEfzTHwPjXCz/zg/Iqlb90/sjXHAOZ1mGX9cQPn68s8dK1HchVRc+Lmp7/F/MnwEAAABA8AzaFtfFqZkq4QCuvuvbttpyqDm6tv4mqWm4TbLziuqXSa5plWQbV0u2aY3tSvge+flPrpQlN8a7En7VQuPGbJuXBMP2uRuzXd569v3g/0f+5/obLqj91sIW65r2oTTBMBo51GpDXbBvrqu2izX68P2d4iB5bj3PsY019au3b16irq5rw+65gXJ8sL1wzbNUnTFIu49t5J8zhOs/t04dXzsv52OnjJ9r3AHWOiQ/5xi2nyuk/b4S9QEAAABIk/pni+vi1EwF29A9UYO4mqPnNN5cHMKL66q7oxeHcPVy9Dkd/VLXscH+cvTvX3m5XPHBBbGG8Dd1fzD4uLFzw3XUR5R96vzArnre0PNB65rhoHfB8HbBADY+fOlXRYNh99zA+dyh9j3Dek4fIC8cTtU66mXo5wfn8f3z52Ib/J4bOz98nz3X4j+1xvh5Fv8Zg+543Bhsz76E+vy6vp/bOI9ShvDIx+78UB0cN3JYV8zHOmScp/G7fbNWH1y1Nn7v1r7gZ3X9fAAAAADKpf7Z4ro4NZMtatieiEFczdE1826QQv2tF1wJV0N4bXu//cZsV175WVnQ/ZZYQ7jyH8bHlCnPfOzS4jB+ufz3qz8t/+OWK4Nfhv4RZeqjzWxrnTc+DJ7/d34QPDeMFYfD8//0fNHZATb8d36ojB7CnzP4Pme4MwbBs2zD9YUDZAzGOT/3arP/577gPIL1wrzxc15AH8KLHI+dbSDW/x8Cz3HB8W258J9ec3bd8J/5+Ln6grjr5wMAAABQrrj/bL1T6a7jD0YO2Sqnamy5Uqg5Wr0nPPP2GyVXrz6ibIXM7VgnueZe18vRx+TrX7tCGue9IfYQPrfrk9J72VsuGMJ1T/3DG4JB/JmPvD3YX/2Ot8jc7k9Z14rHPghPS+pqrfkvyWB+gRn0cwMAAABABVJzdP7i62RO8xLJXLpYMvOWSl1bn9S1r3ffHf3zV3xEXv/qF8YewhU1iH9vYeOFL003qJy6Al7eAK5U6jDKEA4AAAAA05mao+vqb5K6ptvPvRxdvSf8eV0b5LfmD9iH8M9c/jH5y794baIhPPTGng/Itxc1B3c+V58hrqhtdRM293vAk2IIt+engvm2gfF//D8KAAAAAFQiNUerG7NlL7lJMpcsDj6iLNe4Mrg7eqF1nVQdDIbwQ8Uh/G7Zs/dO2T26V7YP7ZTBTQMlDeEAAAAAAFSqYI6+9AapDt4TvkRqmldJbWtf8DnhwcvRTz38oDz80El54P4Tcvz4cRkeHZPv//hq+fAnPsUQDgAAAABAAmqOzl18neTn3So1jcsk37hCCsUBvLZtvdR2bpCqZ55+TJ554hF58swpOXXqIRkZHZVPfPoKecvb5zGEAwAAAACQQHglPD/vFsmo94Srl6M3r5Z8a5/UqM8J/9WTZ+SZx08Vh/CH5fSpB4tD+EhxCP/X4hDewBAOAAAAAEACao7Ovv3a4D3h2YYlFwzhBfU54c88VRzCn3hEnjhzqjiEj18J/9inPiN/+fZ6hnAAAAAAABJQc3T1234hhYbFMqdtZXEQv0OyTavOD+G/Cl6O/mgwhAcvRx9TQ/jl8hcM4QAAAAAAJBJcCb/4l1JoXCw1zcsk37xC5nb1S65lrVQ3rSkO4cGV8EeD94Srl6PvHh2Rj19+RXEI5+XoAAAAAAAkoebozNuuldylN0uuYWlwJTzX0iu1Hf2SaV6r3hP+6LkbswVDuLoSfvln5M8ZwgEAAAAASETN0fmLr5PMxTdJbau6M/oqyajPCW/uHf+IsmefvvBKePCe8E9fLn929uXoVVVVAAAAAADMeurfb37zm7KoObqu4QapabxVcg1LpLZttdR19Em+da17CP/Epy6XP38bQzgwXQ0NDVnjAID08L+1AFB5UhvC62+SOc1LgiE823iH1Lb3SV1nvxTa149/TvhTTz4iTxWH8EdOPSS7xkblU5++Qv7qbeMvR7edGICpMx2eFPLEFECl4H/vAKCypDWEZy6+VvL1t0qm/jbJNCyTOZ3rZG73Bsk0r5GqXz/1qDz9xGl54szD8ujZG7N9rDiEv+Xsy9HVf3zC/wDp30v5j1IpPbNd+Fjqj40tNtPpP5Pr54rKxRG3P8kxwjVdPXre5KvT83G5+lxrm/u22pAtr/fpeX07qjYUp2a6KOc8Xb3hz1/q2nq/b41SjxElznGBUoV/X1F/Z1G5OOL2Rx3fFgcAzD5pDeFVb/2F1DTdLnWtK8ZvzNa8SnIta8Zfjn764YfkgYcfkJP33yf3Hr9Xhkf3yvd/9DP5yMc+ee5KePgfH/N7XKo+ZMtXOtvjEsZmw+MW9fPZ9kv5eeP2q1zS9X3rJdmOisURZy3bflSNmXPFQlFruaiaOHVTrdTzDPtsvWYs6fpx1tTjSdePY6LWnWyz5eeIMtE/Y9rr29YyY/p+KceO269yrnw [... cut ...] veDBcq575OWLgt1PxEj4/a0WuLS14gDw2oXk2qwXW6i9/MLzm2HRW5Gw8l3jARtRnAxrwXlIn13AtBpXneTGKbCyDOVupzGcUKnymwlzem7WS1FzMvWaJsedenCsxySvO49fEZ5O11AgS99ikyrwo5545wmab+88eJY/Np8yVLga+wCa4fsFONvd8drmjzXkV7i265MtVcsOabja/s3kvORvH6uVnyjGf+8jV4/oi15c5V+KCaHItcZ+gh03zTLnyGdngiqEv8esqsB5wb581uRZVTicbZ57XjffLtQ2E/yGgPHsr1/DrYk2uJZl3DLGhHqKgk8/AfUUXky+UuZfEvDa5M82vWYz9eK5oARvxOvk5sea3ck4HG/sQ7i/f783nKHEfeYt5vnkjm3o+A59TvnZM7np7bMyL3ZzbLf/RgPdtZ9iIl7r574JNuNfBP8vObfxa+W+kh035zAfp/wMsEik1SWp/PgAAAABJRU5ErkJggg==" >

Cxp1220, 2020-09-27:
> > > wsl是不支持文件系统大小写识别的。劝你还是装个debian或ubuntu虚拟机编译。
> > > 即使你能成功解决wsl编译的问题，时间上也是比虚拟机慢很多的。
> > 
> > 
> > 能编译出来，但是我换了虚拟机也是一样出现这个问题
> 
> 我记得是可以的，至少wsl2是可以的
> [image]
唉，听了上面大佬的话我把子系统都删了，现在虚拟机也出现同样的问题

Dimitri2020007, 2020-09-27:
用vmware虚拟个ubuntu-server系统。

Cxp1220, 2020-09-27:
> 用vmware虚拟个ubuntu-server系统。不要用hyper-v的虚拟机编译。

感谢回复，现在已换虚拟机，错误同上，股沟了一圈还是不知道咋解决

Cxp1220, 2020-09-27:
> 不要用wsl，兼容性不好。
> 装个虚拟机，或者用multipass[https://github.com/canonical/multipass/releases](url)
> 我用的multipass，和Ubuntu虚拟机一样的

multipass昨天试了下，不太会用，装了个虚拟机，还是出现一样的问题，不知道了。。。

Cxp1220, 2020-09-27:
WARNING: "nf_ct_tcp_be_liberal" [net/netfilter/nf_conntrack] is a static EXPORT_SYMBOL_GPL
WARNING: "nf_ct_tcp_no_window_check" [net/netfilter/nf_conntrack] is a static EXPORT_SYMBOL_GPL
make[5]: Leaving directory '/home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61'
cmp -s /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61/vmlinux /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/vmlinux.debug || { aarch64-openwrt-linux-musl-objcopy -O binary -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61/vmlinux /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/vmlinux; aarch64-openwrt-linux-musl-objcopy -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61/vmlinux /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/vmlinux.elf; cp -fpR /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61/vmlinux /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/vmlinux.debug;  cp -fpR /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61/arch/arm64/boot//Image /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/Image;  }
touch /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.4.61/.image
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/lg/lede/target/linux/armvirt/image'
bash: *1024*1024: syntax error: operand expected (error token is "*1024*1024")
make[5]: Nothing to be done for 'compile'.
mkdir -p /home/lg/lede/bin/targets/armvirt/64 /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/tmp
rm -rf /home/lg/lede/build_dir/target-aarch64_generic_musl/json_info_files
tar -cp --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name --mtime="@1600912884" -C /home/lg/lede/build_dir/target-aarch64_generic_musl/root-armvirt/ . | gzip -9n > /home/lg/lede/bin/targets/armvirt/64/openwrt-armvirt-64-default-rootfs.tar.gz
cp /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/Image /home/lg/lede/bin/targets/armvirt/64/openwrt-armvirt-64-Image
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/lg/lede/build_dir/target-aarch64_generic_musl/root.orig-armvirt TMPDIR=/home/lg/lede/build_dir/target-aarch64_generic_musl/root.orig-armvirt/tmp /home/lg/lede/staging_dir/host/bin/opkg --offline-root /home/lg/lede/build_dir/target-aarch64_generic_musl/root.orig-armvirt --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch aarch64_generic:200 list-installed > /home/lg/lede/bin/targets/armvirt/64/openwrt-armvirt-64-default.manifest
make[5]: Leaving directory '/home/lg/lede/target/linux/armvirt/image'
make[4]: Leaving directory '/home/lg/lede/target/linux/armvirt'
make[3]: Leaving directory '/home/lg/lede/target/linux'
time: target/linux/install#518.16#36.58#196.11
make[2]: Leaving directory '/home/lg/lede'
export MAKEFLAGS= ;make -w -r package/index
make[2]: Entering directory '/home/lg/lede'
Generating package index...
Generating index for package ./base-files_224-r2899-9a482a15a_aarch64_generic.ipk
Generating index for package ./block-mount_2020-05-12-84269037-1_aarch64_generic.ipk
Generating index for package ./fstools_2020-05-12-84269037-1_aarch64_generic.ipk
Generating index for package ./fwtool_2019-11-12-8f7fe925-1_aarch64_generic.ipk
Generating index for package ./ip6tables_1.8.4-1_aarch64_generic.ipk
Generating index for package ./iptables-mod-tproxy_1.8.4-1_aarch64_generic.ipk
Generating index for package ./iptables_1.8.4-1_aarch64_generic.ipk
Generating index for package ./kmod-asn1-decoder_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-aead_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-arc4_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-authenc_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-ecb_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-hash_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-manager_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-null_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-pcompress_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-sha1_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-crypto-user_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-cryptodev_5.4.61+1.10-armvirt-2_aarch64_generic.ipk
Generating index for package ./kmod-fast-classifier_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-ip6tables_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-ipt-conntrack_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-ipt-core_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-ipt-fullconenat_5.4.61+2019-10-21-0cf3b48f-3_aarch64_generic.ipk
Generating index for package ./kmod-ipt-ipset_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-ipt-nat6_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-ipt-nat_5.4.61-1_aarch64_generic.ipk
Generating index for package ./kmod-ipt-raw_5.4.61-1_aarch64_generic.ipk
[... cut ...]
Generating index for package ./zerotier_1.4.6-3_aarch64_generic.ipk
Signing package index...
make[2]: Leaving directory '/home/lg/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/lg/lede'
make[2]: Nothing to be done for 'json_overview_image_info'.
make[2]: Leaving directory '/home/lg/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/lg/lede'
make[2]: Leaving directory '/home/lg/lede'
make[1]: Leaving directory '/home/lg/lede'

Dimitri2020007, 2020-09-27:
你这已经成功了啊，没有make error啊。

Cxp1220, 2020-09-28:
> 你这已经成功了啊，没有make error啊。

  是成功了，但是中间有这个错误就不明白咋回事，不知道会有什么影响

Dimitri2020007, 2020-09-28:
WARNING: "nf_ct_tcp_be_liberal" [net/netfilter/nf_conntrack] is a static EXPORT_SYMBOL_GPL
WARNING: "nf_ct_tcp_no_window_check" [net/netfilter/nf_conntrack] is a static EXPORT_SYMBOL_GPL

这个错误我编译x64固件的时候也碰到过，但它不会影响你的编译。放心继续吧。

Cxp1220, 2020-09-28:
不是这个警告，是下面的1024那里的error

Dimitri2020007, 2020-09-28:
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/lg/lede/target/linux/armvirt/image'
bash: 10241024: syntax error: operand expected (error token is "10241024")
make[5]: Nothing to be done for 'compile'.
mkdir -p /home/lg/lede/bin/targets/armvirt/64 /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/tmp
rm -rf /home/lg/lede/build_dir/target-aarch64_generic_musl/json_info_files

这个是语法错误，和源码有关，但我不知道是哪里的问题。不影响最后生成固件就没有问题。

Cxp1220, 2020-09-28:
> make -C image compile install TARGET_BUILD=
> make[5]: Entering directory '/home/lg/lede/target/linux/armvirt/image'
> bash: 10241024: syntax error: operand expected (error token is "10241024")
> make[5]: Nothing to be done for 'compile'.
> mkdir -p /home/lg/lede/bin/targets/armvirt/64 /home/lg/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/tmp
> rm -rf /home/lg/lede/build_dir/target-aarch64_generic_musl/json_info_files
> 
> 这个是语法错误，和源码有关，但我不知道是哪里的问题。不影响最后生成固件就没有问题。
> 
> 我自己编译x64固件的时候没碰到这错误。

ok,感谢

closed 2021-06-20 by coolsnowwolf