coolsnowwolf/lede #5561: 缩减了固件体积后，问题依旧

Sunnice, 2020-09-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：1.opboot刷完ubi格式固件后，进入op还是存在莫名其妙的flowoffloading配置未保存提示，lede源码和稳定版源码都存在同样问题；2.刷完固件后还是没有无线（固件体积已缩减）使用的是最新源码

### (2) 路由器型号和固件版本

A：竞斗云

### (3) 详细日志

A：
[image]
[image]

mjyhj, 2020-09-27:
系统-软件包
看看空闲空间是不是爆了

Sunnice, 2020-09-27:
> 系统-软件包
> 看看空闲空间是不是爆了

没，还剩下30m左右

Sunnice, 2020-09-27:
用稳定版源码的话，哪怕比lede源码编译的固件还要大也有无线，

AmadeusGhost, 2020-09-27:
你这到底啥情况啊，我昨天编译的都正常（https://ws28.cn/f/3lpkobzvj0s）

Sunnice, 2020-09-27:
> 你这到底啥情况啊，我昨天编译的都正常（ https://ws28.cn/f/3lpkobzvj0s ）

大佬。。。我真的也不知道，关键是这个情况有时候又是正常的，我用lede源码编译，会间歇这样没有无线，然后那个flowoffloading，我是真的摸不着头脑

Dimitri2020007, 2020-09-27:
提供下系统日志

qq1316237299, 2020-09-27:
我今天也碰到这个问题了

Sunnice, 2020-09-27:
> 我今天也碰到这个问题了

果然不止我一个人，你这边有日志吗，可以提供楼上大佬看看

qq1316237299, 2020-09-27:
一开始的日志没了

mjyhj, 2020-09-27:
> > 你这到底啥情况啊，我昨天编译的都正常（ https://ws28.cn/f/3lpkobzvj0s ）
> 
> 大佬。。。我真的也不知道，关键是这个情况有时候又是正常的，我用lede源码编译，会间歇这样没有无线，然后那个flowoffloading，我是真的摸不着头脑

你是不是sfe和flow 一起编译了，又一起打开啊

qq1316237299, 2020-09-27:
就编译了一个sfe

Sunnice, 2020-09-28:
> > > 你这到底啥情况啊，我昨天编译的都正常（ https://ws28.cn/f/3lpkobzvj0s ）
> > 
> > 
> > 大佬。。。我真的也不知道，关键是这个情况有时候又是正常的，我用lede源码编译，会间歇这样没有无线，然后那个flowoffloading，我是真的摸不着头脑
> 
> 你是不是sfe和flow 一起编译了，又一起打开啊

就编译一个sfe，而且这个情况在稳定版源码编译的固件里也会出现，稳定版源码没有flow

Dimitri2020007, 2020-09-28:
x86表示sfe和flowoffload一起用没有任何问题

qq1316237299, 2020-09-28:
> 提供下系统日志

Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Populating IPv6 mangle table
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:    * Zone 'lan'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:    * Zone 'wan'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Set tcp_ecn to off
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Set tcp_syncookies to on
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Set tcp_window_scaling to on
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/usr/share/miniupnpd/firewall.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/var/etc/adbyby.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:    ! Skipping due to path error: No such file or directory
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/etc/ipsec.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/var/etc/passwall.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:    ! Skipping due to path error: No such file or directory
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/var/etc/passwall_server.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:    ! Skipping due to path error: No such file or directory
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/usr/share/softethervpn/firewall.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/var/etc/shadowsocksr.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:    ! Skipping due to path error: No such file or directory
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/etc/wrtbwmon.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/etc/mia.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:  * Running script '/var/etc/unblockmusic.include'
Mon Sep 28 14:04:39 2020 daemon.notice procd: /etc/rc.d/S96appfilter:    ! Skipping due to path error: No such file or directory
Mon Sep 28 14:04:39 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon Sep 28 14:04:40 2020 cron.err crond[6431]: crond (busybox 1.31.1) started, log level 5
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER-USER':No such file or directory
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER-ISOLATION-STAGE-1':No such file or directory
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Sep 28 14:04:40 2020 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
[... cut ...]
Mon Sep 28 14:07:47 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Mon Sep 28 14:08:07 2020 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Mon Sep 28 14:08:07 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Mon Sep 28 14:08:07 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'

Dimitri2020007, 2020-09-28:
注意下emerg信息还有netdata的错误。

提示有地址已经被占用了[nginx]
Mon Sep 28 14:05:10 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Mon Sep 28 14:05:10 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
Mon Sep 28 14:05:10 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
Mon Sep 28 14:05:13 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: still could not bind()

https-dns-proxy应用出现了bug
Mon Sep 28 14:06:41 2020 daemon.info https-dns-proxy[12360]: [E] 1601273201.732658 https_client.c:259 libcurl bug detected: socket closed without ever being read.
Mon Sep 28 14:06:41 2020 daemon.info https-dns-proxy[12360]: [E] 1601273201.732689 https_client.c:260 Activating workaround. PERFORMANCE WILL BE GREATLY DEGRADED!
Mon Sep 28 14:06:41 2020 kern.info kernel: [ 164.495495] https-dns-proxy[12360]: segfault at 0 ip 00007f88147b9e8b sp 00007ffcd8009800 error 4 in libcurl.so.4.6.0[7f88147b8000+2e000]
Mon Sep 28 14:06:41 2020 kern.info kernel: [ 164.509100] Code: 0b 44 89 ee 48 89 df e8 f0 f5 ff ff 5a 89 e8 5b 5d 41 5c 41 5d c3 41 57 41 56 41 55 41 54 49 89 d4 55 53 48 81 ec 58 01 00 00 <4c> 8b 2f 89 74 24 10 64 48 8b 04 25 28 00 00 00 48 89 84 24 48 01

netdata的apps_group无法读取和载入。
Mon Sep 28 14:06:47 2020 user.info : Does not have a configuration file inside /etc/netdata/ebpf.conf. It will try to load stock file. Mon Sep 28 14:06:47 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf' Mon Sep 28 14:06:47 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf' Mon Sep 28 14:06:51 2020 daemon.info https-dns-proxy[12489]: [E] 1601273211.529740 https_client.c:259 libcurl bug detected: socket closed without ever being read. Mon Sep 28 14:06:51 2020 daemon.info https-dns-proxy[12489]: [E] 1601273211.529773 https_client.c:260 Activating workaround. PERFORMANCE WILL BE GREATLY DEGRADED! Mon Sep 28 14:06:51 2020 kern.info kernel: [ 174.236907] https-dns-proxy[12489]: segfault at 0 ip 00007f9dad115e8b sp 00007ffe0e2b4880 error 4 in libcurl.so.4.6.0[7f9dad114000+2e000] Mon Sep 28 14:06:51 2020 kern.info kernel: [ 174.250554] Code: 0b 44 89 ee 48 89 df e8 f0 f5 ff ff 5a 89 e8 5b 5d 41 5c 41 5d c3 41 57 41 56 41 55 41 54 49 89 d4 55 53 48 81 ec 58 01 00 00 <4c> 8b 2f 89 74 24 10 64 48 8b 04 25 28 00 00 00 48 89 84 24 48 01 Mon Sep 28 14:06:51 2020 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 5 seconds since last crash Mon Sep 28 14:07:07 2020 user.info : Does not have a configuration file inside /etc/netdata/ebpf.conf. It will try to load stock file.
Mon Sep 28 14:07:07 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Mon Sep 28 14:07:07 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Mon Sep 28 14:07:27 2020 user.info : Does not have a configuration file inside /etc/netdata/ebpf.conf. It will try to load stock file. Mon Sep 28 14:07:27 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf' Mon Sep 28 14:07:27 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf' Mon Sep 28 14:07:32 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses! Mon Sep 28 14:07:32 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses Mon Sep 28 14:07:35 2020 daemon.info dnsmasq[14200]: Connected to system UBus Mon Sep 28 14:07:35 2020 cron.err crond[14234]: crond (busybox 1. [... cut ...]
Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: remote LL address fe80::0ea4:02ff:feb1:2801
Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: local IP address 122.232.249.155
Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: remote IP address 10.36.80.1
Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: primary DNS address 202.96.113.34
Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: secondary DNS address 202.96.113.35
Mon Sep 28 14:05:54 2020 user.err wsdd2[7483]: error: wsdd-mcast-v4: wsd_send_soap_msg: send

Sunnice, 2020-09-28:
> 注意下emerg信息、error信息还有netdata的错误。
> 
> 提示有地址已经被占用了[nginx]
> Mon Sep 28 14:05:10 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:10 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:10 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
> Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
> Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:11 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
> Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
> Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 127.0.0.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to 192.168.8.1:80 failed (98: Address in use)
> Mon Sep 28 14:05:12 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: bind() to [fd65:6c8:d6c::1]:80 failed (98: Address in use)
> Mon Sep 28 14:05:13 2020 daemon.err nginx[9802]: 2020/09/28 14:05:10 [emerg] 9802#0: still could not bind()
> 
> https-dns-proxy应用出现了bug
> Mon Sep 28 14:06:41 2020 daemon.info https-dns-proxy[12360]: [E] 1601273201.732658 https_client.c:259 libcurl bug detected: socket closed without ever being read.
> Mon Sep 28 14:06:41 2020 daemon.info https-dns-proxy[12360]: [E] 1601273201.732689 https_client.c:260 Activating workaround. PERFORMANCE WILL BE GREATLY DEGRADED!
> Mon Sep 28 14:06:41 2020 kern.info kernel: [ 164.495495] https-dns-proxy[12360]: segfault at 0 ip 00007f88147b9e8b sp 00007ffcd8009800 error 4 in libcurl.so.4.6.0[7f88147b8000+2e000]
> Mon Sep 28 14:06:41 2020 kern.info kernel: [ 164.509100] Code: 0b 44 89 ee 48 89 df e8 f0 f5 ff ff 5a 89 e8 5b 5d 41 5c 41 5d c3 41 57 41 56 41 55 41 54 49 89 d4 55 53 48 81 ec 58 01 00 00 <4c> 8b 2f 89 74 24 10 64 48 8b 04 25 28 00 00 00 48 89 84 24 48 01
> 
> netdata的apps_group无法读取和载入。
> Mon Sep 28 14:06:47 2020 user.info : Does not have a configuration file inside /etc/netdata/ebpf.conf. It will try to load stock file. Mon Sep 28 14:06:47 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf' Mon Sep 28 14:06:47 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf' Mon Sep 28 14:06:51 2020 daemon.info https-dns-proxy[12489]: [E] 1601273211.529740 https_client.c:259 libcurl bug detected: socket closed without ever being read. Mon Sep 28 14:06:51 2020 daemon.info https-dns-proxy[12489]: [E] 1601273211.529773 https_client.c:260 Activating workaround. PERFORMANCE WILL BE GREATLY DEGRADED! Mon Sep 28 14:06:51 2020 kern.info kernel: [ 174.236907] https-dns-proxy[12489]: segfault at 0 ip 00007f9dad115e8b sp 00007ffe0e2b4880 error 4 in libcurl.so.4.6.0[7f9dad114000+2e000] Mon Sep 28 14:06:51 2020 kern.info kernel: [ 174.250554] Code: 0b 44 89 ee 48 89 df e8 f0 f5 ff ff 5a 89 e8 5b 5d 41 5c 41 5d c3 41 57 41 56 41 55 41 54 49 89 d4 55 53 48 81 ec 58 01 00 00 <4c> 8b 2f 89 74 24 10 64 48 8b 04 25 28 00 00 00 48 89 84 24 48 01 Mon Sep 28 14:06:51 2020 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 5 seconds since last crash Mon Sep 28 14:07:07 2020 user.info : Does not have a configuration file inside /etc/netdata/ebpf.conf. It will try to load stock file.
> Mon Sep 28 14:07:07 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
> Mon Sep 28 14:07:07 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
> Mon Sep 28 14:07:27 2020 user.info : Does not have a configuration file inside /etc/netdata/ebpf.conf. It will try to load stock file. Mon Sep 28 14:07:27 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf' Mon Sep 28 14:07:27 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf' Mon Sep 28 14:07:32 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses! Mon Sep 28 14:07:32 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses Mon Sep 28 14:07:35 2020 daemon.info dnsmasq[14200]: Connected to system  [... cut ...]
> Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: local IP address 122.232.249.155
> Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: remote IP address 10.36.80.1
> Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: primary DNS address 202.96.113.34
> Mon Sep 28 14:05:54 2020 daemon.notice pppd[11548]: secondary DNS address 202.96.113.35
> Mon Sep 28 14:05:54 2020 user.err wsdd2[7483]: error: wsdd-mcast-v4: wsd_send_soap_msg: send

我们压根没编译flowoffloading，而且这个配置未保存在稳定版源码编译的固件内也有，稳定版源码是没有flowoff的

Dimitri2020007, 2020-09-28:
源码的bug？

Sunnice, 2020-09-28:
> 源码的bug？

在排错中，可能是添加插件的时候加入了什么奇怪的东西，我这次只用大屌的源码编译了个固件，先试试看

Sunnice, 2020-09-28:
> 你这到底啥情况啊，我昨天编译的都正常（ https://ws28.cn/f/3lpkobzvj0s ）

大佬，flow的配置问题是我添加的其他源码引起的，但是没有WiFi这个问题好像是lede源码的问题，是不是超过64m的固件就会导致没有无线？[lede编译的固件](https://cowtransfer.com/s/26cb1757500e4e)

Sunnice, 2020-09-28:
> 源码的bug？

第一个flow的配置问题可以确定是我添加的其他源码引起的了，但具体是什么还未知，不过没有WiFi这个问题好像是lede源码的问题，不知道是不是超过64m的固件就会导致没有无线？lede编译的固件

Sunnice, 2020-09-28:
> 就编译了一个sfe
> […](#)

除了大屌lede源码，你还添加了哪些？flow那个问题是其他源码引起的

qq1316237299, 2020-09-29:
> > 就编译了一个sfe
> > […](#)
> 
> 除了大屌lede源码，你还添加了哪些？flow那个问题是其他源码引起的

现在好了  ，全部重新编译了以后恢复正常

Sunnice, 2020-09-29:
> > > 就编译了一个sfe
> > > […](#)
> > 
> > 
> > 除了大屌lede源码，你还添加了哪些？flow那个问题是其他源码引起的
> 
> 现在好了 ，全部重新编译了以后恢复正常

你也是竞斗云吗？你编译的固件多大呀？编译后有无线没？

closed 2021-06-20 by coolsnowwolf