coolsnowwolf/lede #5581: ssr无法启动

BecodReyes, 2020-10-01:
## 1.关于你要提交的问题

Q：是否搜索了issue
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

无法正常启动ssr，界面提示“ShadowsocksR Plus+ 未运行”

### (2) 路由器型号和固件版本

X86 https://github.com/coolsnowwolf/lede/commit/9a482a15a3d14849b050ab7556ab9542b47b2c21 https://github.com/fw876/helloworld/commit/9d2478ea587b987133d2a94c90d76e237d3503fd
编译[.config](https://pastebin.ubuntu.com/p/Wfpgrx8Jtw/)

### (3) 详细日志
```
Thu Oct  1 09:28:08 2020 user.notice shadowsocksr: ssr redir tcp error.restart!
Thu Oct  1 09:28:08 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Oct  1 09:28:08 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Oct  1 09:28:08 2020 daemon.info dnsmasq[9207]: Connected to system UBus
Thu Oct  1 09:28:08 2020 cron.err crond[9241]: crond (busybox 1.31.1) started, log level 5
Thu Oct  1 09:28:09 2020 daemon.info uwsgi-luci: uWSGI worker 2 cheaped.
Thu Oct  1 09:28:09 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Oct  1 09:28:09 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Oct  1 09:28:09 2020 daemon.info dnsmasq[9584]: Connected to system UBus
```

Dimitri2020007, 2020-10-01:
你必须要有一个ssr://开头的链接导入到路由器里就可以运行了啊

BecodReyes, 2020-10-01:
> 你必须要有一个ssr://开头的链接导入到路由器里就可以运行了啊

新手小白？
难道我连这种事都不会做？
要是早就解决了这个issues就不会存在了

closed 2021-06-20 by coolsnowwolf