coolsnowwolf/lede #5588: 编译lede target images文件系统选ext4或squashfs有啥区别吗？

Dimitri2020007, 2020-10-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如题

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：

hcym, 2020-10-03:
ex的可以挂载修改，我都是用ex的

secretandanon, 2020-10-03:
ext4是可修改的，另一个是只读的，比如，路由器的恢复出厂设置功能就依赖这个只读功能

Dimitri2020007, 2020-10-03:
我编译的有block模块、filesystem模块、有线\无线网卡驱动模块，luci-app模块、freeradius、ssr+。模块比较多。选ext4对生成固件有影响吗？编译到最后安装ipk和模块的时候总是会出现argument list too long是什么原因，有人能解释下吗？

是选ext4的问题吗？还是编译模块选太多了？有其他原因的也来说下。谢谢了。

9l, 2020-10-03:
你用什么系统编译？裸机装的 linux？还是 vps，还是虚拟机，什么类型的虚拟机，virtualbox? vmware? hyper-v? WSL/WSL2?

Dimitri2020007, 2020-10-03:
我用的虚拟机vmware，本地编译。已设置代理、dl反复执行make -j8 download，是下载完整的没有错误。

一开始以为是wsl/wsl2的问题。换了虚拟机和脚本编译都是错误的。脚本提示是超时。运行6小时左右。（脚本）
虚拟机编译7小时+。编译过程顺利，就是最后安装ipk和模块文件的时候提示argument list too long。脚本超时可能和这个错误有关。
相关问题是#5538（已关闭）

之后在target image中尝试过以下选项

只用ext4
只使用squashfs
启用或者禁用ramdisk
调整kernel partition size和root part size
减少模块编译数量

编译模块789到800个左右（包含硬盘所有kmod、有线和无线网卡kmod、input kmod、文件系统kmod、netfliter kmod、luci web后台（只选了1个）、luci主题、luci-app（去除功能上重复的像dns这类的）、iputils所有、freeradius默认、iperf3、ssr+、ssr python、ssr-libev、v2ray、trojan、utlites下的disc、editor、filesystem）

全是单线程编译。没使用多线程。
系统是ubuntu-server 64位的最新版本。20.04.1

Dimitri2020007, 2020-10-04:
目前测试argument list too long错误与文件系统无关，今天测试不修改luci的3个内容。

luci-proto（保持默认选项）
luci-library（保持默认选项）
freifunk（保持默认选项）

通过。发现了软件包冲突。

9l, 2020-10-04:
其实。。。没必要选这么多的包进去，很多根本都用不上

Dimitri2020007, 2020-10-04:
我一直装虚拟机用的。esxi、hyper-v、kvm都有。ssr可以取消一些、odhcpd所有可以取消、filesystem（utilties下的）保持默认不修改。数量已经控制到737个左右了。

closed 2020-10-05 by Dimitri2020007