coolsnowwolf/lede #5589: v2ray,能运行，但不能通过GOOGLE检测

panther706, 2020-10-03:
反馈bug/问题模板，提建议请删除

设备：WNDR4300
固件版本：OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)，10.3最新拉取源码 。
v2ray,显示运行中，但不能通过GOOGLE检测，不能上油管，换了几个机场确定不是机场问题。换成55R就正常使用。

[ 0.000000] Linux version 4.19.138 (pwh@pwh-machine) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2899-9a482a15a)) #0 Thu Sep 24 02:01:24 2020
[ 0.000000] bootconsole [early0] enabled
[ 0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[ 0.000000] MIPS: machine is Netgear WNDR4300
[ 0.000000] SoC: Atheros AR9344 rev 2
[ 0.000000] Determined physical RAM map:
[ 0.000000] memory: 08000000 @ 00000000 (usable)
[ 0.000000] Initrd not found or empty - disabling initrd
[ 0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[ 0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[ 0.000000] Zone ranges:
[ 0.000000] Normal [mem 0x0000000000000000-0x0000000007ffffff]
[ 0.000000] Movable zone start for each node
[ 0.000000] Early memory node ranges
[ 0.000000] node 0: [mem 0x0000000000000000-0x0000000007ffffff]
[ 0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[ 0.000000] On node 0 totalpages: 32768
[ 0.000000] Normal zone: 288 pages used for memmap
[ 0.000000] Normal zone: 0 pages reserved
[ 0.000000] Normal zone: 32768 pages, LIFO batch:7
[ 0.000000] random: get_random_bytes called from start_kernel+0x98/0x4a4 with crng_init=0
[ 0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=132768
[ 0.000000] pcpu-alloc: [0] 0
[ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 32480
[ 0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[ 0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[ 0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[ 0.000000] Writing ErrCtl register=00000000
[ 0.000000] Readback ErrCtl register=00000000
[ 0.000000] Memory: 121972K/131072K available (4703K kernel code, 169K rwdata, 1076K rodata, 1212K init, 206K bss, 9100K reserved, 0K cma-reserved)
[ 0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[ 0.000000] NR_IRQS: 51
[ 0.000000] CPU clock: 560.000 MHz
[ 0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6825930166 ns
[ 0.000010] sched_clock: 32 bits at 280MHz, resolution 3ns, wraps every 7669584382ns
[ 0.008319] Calibrating delay loop... 278.78 BogoMIPS (lpj=557568)
[ 0.050874] pid_max: default: 32768 minimum: 301
[ 0.056055] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[ 0.063114] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[ 0.076497] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[ 0.086918] futex hash table entries: 256 (order: -1, 3072 bytes)
[ 0.093579] pinctrl core: initialized pinctrl subsystem
[ 0.099726] NET: Registered protocol family 16
[ 0.112187] GPIO line 18 (netgear:ext:lna0) hogged as output/high
[ 0.118736] GPIO line 19 (netgear:ext:lna1) hogged as output/high
[ 0.149060] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[ 0.161812] clocksource: Switched to clocksource MIPS
[ 0.168489] NET: Registered protocol family 2
[ 0.174201] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[ 0.182471] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[ 0.189923] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[ 0.196716] TCP: Hash tables configured (established 1024 bind 1024)
[ 0.203634] UDP hash table entries: 256 (order: 0, 4096 bytes)
[ 0.209911] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[ 0.216943] NET: Registered protocol family 1
[ 0.221678] PCI: CLS 0 bytes, default 32
[ 0.225398] Crashlog allocated RAM at address 0x3f00000
[ 0.232696] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[ 0.246846] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[ 0.253106] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[ 0.280328] io scheduler noop registered
[ 0.284568] io scheduler deadline registered (default)
[ 0.292940] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[ 0.300182] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[... 236 lines cut ...]
[ 79.991002] br-lan: port 3(wlan0) entered blocking state
[ 79.996509] br-lan: port 3(wlan0) entered disabled state
[ 80.002325] device wlan0 entered promiscuous mode
[ 80.007393] br-lan: port 3(wlan0) entered blocking state
[ 80.012836] br-lan: port 3(wlan0) entered forwarding state
[ 80.149970] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 361.572599] pppoe-wan: renamed from ppp0
[ 403.223947] device wlan0 left promiscuous mode
[ 403.228703] br-lan: port 3(wlan0) entered disabled state
[ 404.110792] device wlan1 left promiscuous mode
[ 404.115567] br-lan: port 2(wlan1) entered disabled state
[ 409.429214] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 409.489084] br-lan: port 2(wlan0) entered blocking state
[ 409.494549] br-lan: port 2(wlan0) entered disabled state
[ 409.500366] device wlan0 entered promiscuous mode
[ 409.986029] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 409.992741] br-lan: port 2(wlan0) entered blocking state
[ 409.998186] br-lan: port 2(wlan0) entered forwarding state
[ 410.064285] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[ 410.107095] br-lan: port 3(wlan1) entered blocking state
[ 410.112559] br-lan: port 3(wlan1) entered disabled state
[ 410.118409] device wlan1 entered promiscuous mode
[ 410.123390] br-lan: port 3(wlan1) entered blocking state
[ 410.128845] br-lan: port 3(wlan1) entered forwarding state
[ 410.181047] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready

panther706, 2020-10-03:
[https://github.com/panther706/lede/blob/master/%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE%202020-10-03%20072121.png](url)

[https://github.com/panther706/lede/blob/master/%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE%202020-10-03%20072226.png](url)

panther706, 2020-10-03:
[image]
[image]

smallprogram, 2020-10-03:
[image]
[image]

刚刚编译，并没有发现你说的这种问题

panther706, 2020-10-03:
> [image]
> [image]
> 
> 刚刚编译，并没有发现你说的这种问题

可能软路由能用，MIPS平台支持不太友好。

sevengod007, 2020-10-03:
检查你的v2ray版本，4.28遇到了你所说的问题，换成4.27问题解决

panther706, 2020-10-03:
> 检查你的v2ray版本，4.28遇到了你所说的问题，换成4.27问题解决

具体怎么操作，我试试看

sevengod007, 2020-10-03:
软件包搜索v2ray,确认版本是否为4.28，卸载‘’v2ray‘’,文件传输这里上传这个https://openwrt.download/R20.9.4/packages/x86_64/base/v2ray_4.27.5-1_x86_64.ipk并安装试一试

panther706, 2020-10-03:
> 软件包搜索v2ray,确认版本是否为4.28，卸载‘’v2ray‘’,文件传输这里上传这个https://openwrt.download/R20.9.4/packages/x86_64/base/v2ray_4.27.5-1_x86_64.ipk
> 并安装试一试

好的，谢了。但我不是软路由，这个IPK通用吗？

sevengod007, 2020-10-03:
> > 粗略搜索v2ray，确认版本是否为4.28，卸载“ v2ray”，文件传输此处上传这个https://openwrt.download/R20.9.4/packages/x86_64/base/v2ray_4.27.5-1_x86_64.ipk
> > 并安装试一试
> 
> 好的，谢了。但我不是软路由，这个IPK通用吗？

未知，你可以找到对应你设备的文件来安装或者重新拉取编译一下，最新的源码v2ray好像已经更新到了4.30。看楼上的截图4.30正常。这个链接里你看下能不能找到你需要的
https://openwrt.download/?dir=R20.9.4

panther706, 2020-10-03:
> 软件包搜索v2ray,确认版本是否为4.28，卸载‘’v2ray‘’,文件传输这里上传这个https://openwrt.download/R20.9.4/packages/x86_64/base/v2ray_4.27.5-1_x86_64.ipk
> 并安装试一试

刚刚试了，安装出错，不同平台，不能用

panther706, 2020-10-03:
真的是版本问题，v2ray降级到4.27.5，问题暂时解决
[image]
[image]

closed 2020-10-03 by panther706