coolsnowwolf/lede #5590: NanoPi R2S wan获取不到IP

rongcolor, 2020-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

光猫是正常的，最新编译（10.2）wan口获取不到IP（光猫DHCP），r2s断网状态。刷回Commits on Sep 24, 2020就正常了。各位都正常吗？

### (2) 路由器型号和固件版本

NanoPi R2S  （Commits on Oct 3, 2020）

joesdu, 2020-10-03:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> 光猫是正常的，最新编译（10.2）wan口获取不到IP（光猫DHCP），r2s断网状态。刷回Commits on Sep 24, 2020就正常了。各位都正常吗？
> 
> ### (2) 路由器型号和固件版本
> NanoPi R2S （Commits on Oct 3, 2020）

我也不正常,DHCP直接连不上.我刷回了我8月份编译的系统就好了.

1715173329, 2020-10-03:
有问题有问题，日志不给我们靠算来给你解决问题？

rongcolor, 2020-10-03:
> 有问题有问题，日志不给我们靠算来给你解决问题？

不好意思，刚又刷回去把日志补上。麻烦看看，谢谢。

1715173329, 2020-10-03:
网口没接反么？看起来没什么问题
清理掉配置重新刷一下看看，最好用 ext4 格式的固件

joesdu, 2020-10-03:
> 网口没接反么？看起来没什么问题
> 清理掉配置重新刷一下看看，最好用 ext4 格式的固

试过了,两个格式都不行.接口肯定没反.

joesdu, 2020-10-03:
> 网口没接反么？看起来没什么问题
> 清理掉配置重新刷一下看看，最好用 ext4 格式的固件

使用9a482a15a3d14849b050ab7556ab9542b47b2c21这个commit的代码编译后能够正常使用,肯定是后边几次代码中有问题.

1715173329, 2020-10-03:
target/linux/rockchip/patches-5.4/006-rockchip-rk3328-add-compatible-to-NanoPi-R2S-etherne.patch

把这个文件删了再编译试试

joesdu, 2020-10-03:
> target/linux/rockchip/patches-5.4/006-rockchip-rk3328-add-compatible-to-NanoPi-R2S-etherne.patch
> 
> 把这个文件删了再编译试试

稍等,正在编译

joesdu, 2020-10-03:
> target/linux/rockchip/patches-5.4/006-rockchip-rk3328-add-compatible-to-NanoPi-R2S-etherne.patch
> 
> 把这个文件删了再编译试试

删掉后可行.

1715173329, 2020-10-03:
上游出了点问题，暂时先这样处理一下吧，等两天再看看

9l, 2020-10-03:
上游的patch未能适配新的更新，这种需要依赖外部patch的，不要追新

1715173329, 2020-10-03:
这不是适配问题
至于外部 patch，没有的话你 r2s 想在 5.4 内核上面跑？想桃子呢

rongcolor, 2020-10-03:
> > target/linux/rockchip/patches-5.4/006-rockchip-rk3328-add-compatible-to-NanoPi-R2S-etherne.patch
> > 把这个文件删了再编译试试
> 
> 删掉后可行.

Thu Jan 21 16:56:00 2016 kern.info kernel: [  353.755896] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [Generic PHY]
Thu Jan 21 16:56:00 2016 daemon.notice netifd: Interface 'wan' is enabled
Thu Jan 21 16:56:00 2016 kern.info kernel: [  353.767761] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
Thu Jan 21 16:56:00 2016 kern.warn kernel: [  353.768466] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
Thu Jan 21 16:56:00 2016 kern.info kernel: [  353.769084] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
Thu Jan 21 16:56:04 2016 daemon.notice netifd: Network device 'eth0' link is up
Thu Jan 21 16:56:04 2016 daemon.notice netifd: Interface 'wan' has link connectivity
Thu Jan 21 16:56:04 2016 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jan 21 16:56:04 2016 kern.info kernel: [  357.852746] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Thu Jan 21 16:56:04 2016 daemon.notice netifd: wan (4502): udhcpc: started, v1.31.1
Thu Jan 21 16:56:04 2016 daemon.notice netifd: wan (4502): udhcpc: sending discover
Thu Jan 21 16:56:07 2016 daemon.notice netifd: wan (4502): udhcpc: sending discover
Thu Jan 21 16:56:10 2016 daemon.notice netifd: wan (4502): udhcpc: sending discover

尴尬了，我这删掉都不行，刷回之前的（Commits on Sep 24, 2020）就正常。这个版本，make clean、接口没动过、静态IP、 一直用的ext4 格式、ubuntu环境都重新做了，还是不行。你的已经正常了吗？

1715173329, 2020-10-03:
试试 #5596

rongcolor, 2020-10-04:
> 试试 #5596

 谢谢，正常了。

joesdu, 2020-10-04:
有人能合并下分支吗?不会用git拉pr中的代码.

TendernessRadif, 2020-10-05:
> > 试试 #5596
> 
> 谢谢，正常了。

老哥 请问你咋搞成功的啊？

467815891a, 2020-10-07:
我也碰到这个问题，真的奇怪，还好大家踩过坑了

PR #5596 (2020-10-07): Revert "uboot-rockchip: update NanoPi R2S patches"
This reverts commit bda6f6572be630bcf9a1c8a429e40d8a53033af5.

This commit breaks the onboard ethernet on some units. Revert it for
now.

Fixes: #5590
closed 2020-10-07 by coolsnowwolf