coolsnowwolf/lede #5591: 编译错误，请大佬们帮助

noyjcl, 2020-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
编译固件错误
### (1) 具体问题
make[2]: *** [package/Makefile:71: package/install] Error 1
make[2]: Leaving directory '/home/yujiang/lede'
make[1]: *** [package/Makefile:108: /home/yujiang/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yujiang/lede'
make: *** [/home/yujiang/lede/include/toplevel.mk:224: world] Error 2
A：

### (2) 路由器型号和固件版本

A：x86 64

### (3) 详细日志

A：make[2]: *** [package/Makefile:71: package/install] Error 1
make[2]: Leaving directory '/home/yujiang/lede'
make[1]: *** [package/Makefile:108: /home/yujiang/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yujiang/lede'
make: *** [/home/yujiang/lede/include/toplevel.mk:224: world] Error 2

1715173329, 2020-10-03:
你这日志好详细哦

Dimitri2020007, 2020-10-04:
请提供完整的make error信息，error信息不完整

closed 2021-06-20 by coolsnowwolf