coolsnowwolf/lede #5592: 编译失败，大佬们帮忙看看

YipSzeho, 2020-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败了

### (2) 路由器型号和固件版本

A：Pi4b

### (3) 详细日志

A：Collected errors:
 * check_data_file_clashes: Package luci-i18n-app-zh-cn wants to install file /home/yipszeho/lede/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/usr/lib/lua/luci/i18n/unblockmusic.zh-cn.lmo
	But that file is already provided by package  * luci-i18n-unblockmusic-zh-cn
 * opkg_install_cmd: Cannot install package luci-i18n-app-zh-cn.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/yipszeho/lede'
make[1]: *** [package/Makefile:108: /home/yipszeho/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yipszeho/lede'
make: *** [/home/yipszeho/lede/include/toplevel.mk:224：world] 错误 2

BecodReyes, 2020-10-03:
luci-i18n-unblockmusic-zh-cn和luci-i18n-app-zh-cn冲突
请考虑去一留一

Dimitri2020007, 2020-10-04:
检查软件包冲突：软件包luci-i18n-app-zh-cn想要安装到文件/home/yipszeho/lede/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/usr/lib/lua/luci/i18n/unblockmusic.zh-cn.lmo

但该软件所有功能已经由luci-i18n-unblockmusic-zh-cn所提供

请在编译时取消luci-i18n-app-zh-cn这个软件包

closed 2020-10-16 by YipSzeho