coolsnowwolf/lede #5594: ssrplus+运行后任何网站都无法访问

yjiawqgj, 2020-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ssrplus+运行后，不管是国内的还是国外全部上不去了，而且节点的ping都会连接失败!也就是说只要成功运行ssrplus+就相当于全面断网。已经确定节点没有问题，手机和V2RAYN全部可以正常使用！

### (2) 路由器型号和固件版本

A：主机型号 | Intel(R) Celeron(R) CPU 3865U @ 1.80GHz : 1 Core 1 Thread

固件版本 | OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)

### (3) 详细日志

A：Sat Oct  3 21:33:42 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:44 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:46 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:48 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:50 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:52 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:54 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:56 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:33:58 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:00 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:02 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:04 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:06 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:08 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:10 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:12 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:14 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:16 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:18 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:20 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:22 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:24 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:26 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:28 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:30 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:32 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:34 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:36 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:38 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:40 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:42 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:44 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:46 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:48 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:50 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:52 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:54 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:56 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:34:58 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:00 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:02 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:04 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:06 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:08 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:10 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:12 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:14 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:16 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 21:35:18 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
[... 628 lines cut ...]
Sat Oct  3 22:16:52 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:16:54 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:16:56 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:16:58 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:00 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:02 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:04 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:06 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:08 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:10 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:12 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:14 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:16 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:18 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:20 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:22 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:24 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:26 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:28 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:30 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:32 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:34 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:36 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:38 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Oct  3 22:17:40 2020 daemon.err netdata[5907]: PROCFILE: Cannot open file '/proc/sysvipc/shm'

Dimitri2020007, 2020-10-03:
编译时候选了ipset吗？报错的是守护进程netdata，无法打开文件/proc/sysvipc/shm这个文件。

yjiawqgj, 2020-10-04:
谢谢是这个问题已经解决!

closed 2020-10-04 by yjiawqgj

Reoooh, 2021-04-18:
> 谢谢是这个问题已经解决!

请问怎么解决的？我也遇到相同的问题了

yjiawqgj, 2021-04-18:
> > 谢谢是这个问题已经解决！
> 
> 请问怎么解决的？我也遇到相同的问题了

编译时候选ipset