coolsnowwolf/lede #5599: kernel: 5.4.68 编译的固件PPPoE频繁断连

Ljzkirito, 2020-10-04:
Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：昨天更新源码后，保留配置升级，PPPoE频繁断连，关闭Flow Offloading没有效果，自己Fork到自己仓库，Revert "kernel: bump 5.4 to 5.4.68 (coolsnowwolf#5555)"后正常。

### (2) 路由器型号和固件版本

A：XY-C5，最新源码，Github Actions编译

### (3) 详细日志

A：没有保存，，，

AmadeusGhost, 2020-10-04:
所以没有任何日志要怎么知道问题在哪，，，

Ljzkirito, 2020-10-04:
> 所以没有任何日志要怎么知道问题在哪，，，

那我再刷一次保存个日志，，，

jchen-git, 2020-10-04:
我的XY-C5也遇到这个问题，启动后大概10多分钟就丢失了pppoe-wan连接，看日志是peer发了LCP reset

AmadeusGhost, 2020-10-04:
> A：昨天更新源码后，保留配置升级，PPPoE频繁断连

如果不保留配置升级的话正常嘛

Myoko, 2020-10-04:
旧内核升级到新的内核 还是建议别保存配置升级！

jchen-git, 2020-10-04:
不保存配置也是一样的，下面是系统日志

Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: PAP authentication succeeded
Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: peer from calling number 70:7B:E8:C2:E7:05 authorized
Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: local  LL address fe80::2091:b59c:d21b:4f5d
Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: remote LL address fe80::727b:e8ff:fec2:e705
Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: local  IP address 10.70.76.81
Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: remote IP address 10.70.0.1
Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: primary   DNS address 202.106.195.68
Sun Oct  4 13:45:54 2020 daemon.notice pppd[2855]: secondary DNS address 202.106.46.151
Sun Oct  4 13:45:54 2020 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sun Oct  4 13:45:54 2020 daemon.notice netifd: Interface 'wan' is now up
Sun Oct  4 13:45:54 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sun Oct  4 13:45:54 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Sun Oct  4 13:45:54 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sun Oct  4 13:45:54 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Sun Oct  4 13:45:54 2020 daemon.notice netifd: Interface 'wan_6' is now up
Sun Oct  4 13:45:54 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Oct  4 13:45:55 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Sun Oct  4 13:49:28 2020 daemon.info pppd[2855]: System time change detected.
**Sun Oct  4 13:57:08 2020 daemon.info pppd[2855]: LCP terminated by peer**
Sun Oct  4 13:57:08 2020 daemon.info pppd[2855]: Connect time 7.9 minutes.
Sun Oct  4 13:57:08 2020 daemon.info pppd[2855]: Sent 1917609 bytes, received 76544677 bytes.
Sun Oct  4 13:57:08 2020 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sun Oct  4 13:57:08 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Sun Oct  4 13:57:08 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Sun Oct  4 13:57:08 2020 daemon.notice netifd: Interface 'wan_6' is now down
Sun Oct  4 13:57:08 2020 daemon.notice netifd: Interface 'wan_6' is disabled
Sun Oct  4 13:57:08 2020 daemon.notice netifd: Interface 'wan' has lost the connection
Sun Oct  4 13:57:11 2020 daemon.notice pppd[2855]: Connection terminated.
Sun Oct  4 13:57:11 2020 daemon.info pppd[2855]: Connect time 7.9 minutes.
Sun Oct  4 13:57:11 2020 daemon.info pppd[2855]: Sent 1917609 bytes, received 76544677 bytes.
Sun Oct  4 13:57:11 2020 daemon.info pppd[2855]: Sent PADT
Sun Oct  4 13:57:11 2020 daemon.notice pppd[2855]: Modem hangup
Sun Oct  4 13:57:11 2020 daemon.info pppd[2855]: Exit.
Sun Oct  4 13:57:11 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Oct  4 13:57:11 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sun Oct  4 13:57:11 2020 daemon.err insmod: module is already loaded - slhc
Sun Oct  4 13:57:11 2020 daemon.err insmod: module is already loaded - ppp_generic
Sun Oct  4 13:57:11 2020 daemon.err insmod: module is already loaded - pppox
Sun Oct  4 13:57:11 2020 daemon.err insmod: module is already loaded - pppoe
Sun Oct  4 13:57:11 2020 daemon.notice netifd: wan (3433): uci: Entry not found
Sun Oct  4 13:57:11 2020 daemon.notice netifd: wan (3433): sh: out of range
Sun Oct  4 13:57:12 2020 daemon.info pppd[3448]: Plugin rp-pppoe.so loaded.
Sun Oct  4 13:57:12 2020 daemon.info pppd[3448]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Oct  4 13:57:12 2020 daemon.notice pppd[3448]: pppd 2.4.8 started by root, uid 0
Sun Oct  4 13:57:17 2020 daemon.info pppd[3448]: PPP session is 32914
Sun Oct  4 13:57:17 2020 daemon.warn pppd[3448]: Connected to 70:7b:e8:c2:e7:05 via interface wan
Sun Oct  4 13:57:17 2020 kern.info kernel: [  558.608642] pppoe-wan: renamed from ppp0
Sun Oct  4 13:57:17 2020 daemon.info pppd[3448]: Renamed interface ppp0 to pppoe-wan
Sun Oct  4 13:57:17 2020 daemon.info pppd[3448]: Using interface pppoe-wan
Sun Oct  4 13:57:17 2020 daemon.notice pppd[3448]: Connect: pppoe-wan <--> wan
Sun Oct  4 13:57:17 2020 daemon.info pppd[3448]: syncppp not active
Sun Oct  4 13:57:20 2020 daemon.info pppd[3448]: syncppp not active
Sun Oct  4 13:57:20 2020 daemon.notice pppd[3448]: PAP authentication succeeded
Sun Oct  4 13:57:20 2020 daemon.notice pppd[3448]: peer from calling number 70:7B:E8:C2:E7:05 authorized
Sun Oct  4 13:57:27 2020 daemon.notice pppd[3448]: local  IP address 10.70.87.116
Sun Oct  4 13:57:27 2020 daemon.notice pppd[3448]: remote IP address 10.70.0.1
Sun Oct  4 13:57:27 2020 daemon.notice pppd[3448]: primary   DNS address 202.106.195.68
Sun Oct  4 13:57:27 2020 daemon.notice pppd[3448]: secondary DNS address 202.106.46.151
Sun Oct  4 13:57:27 2020 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sun Oct  4 13:57:27 2020 daemon.notice netifd: Interface 'wan' is now up
Sun Oct  4 13:57:27 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Oct  4 13:57:38 2020 daemon.notice pppd[3448]: local  LL address fe80::318b:573a:350e:c489
Sun Oct  4 13:57:38 2020 daemon.notice pppd[3448]: remote LL address fe80::727b:e8ff:fec2:e705
Sun Oct  4 13:57:38 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sun Oct  4 13:57:38 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Sun Oct  4 13:57:38 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sun Oct  4 13:57:38 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Sun Oct  4 13:57:38 2020 daemon.notice netifd: Interface 'wan_6' is now up
Sun Oct  4 13:57:38 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)

AmadeusGhost, 2020-10-04:
试试这个 commit ( https://github.com/AmadeusGhost/lede/commit/2e74f09b8b2964012a3a55f67c7f3137fe86008a) 能不能解决问题

Myoko, 2020-10-04:
物理机 纯路由 没有遇到这个问题！开机3个多小时都没有断过

blueveryday, 2020-10-04:
> 不保存配置也是一样的，下面是系统日志
>
> Sun Oct 4 13:45:54 2020 daemon.notice pppd[2855]: PAP authentication succeeded
> Sun Oct 4 13:45:54 2020 daemon.notice pppd[2855]: peer from calling number 70:7B:E8:C2:E7:05 [...]

HC5962，跟你问题类似，wan+man方式，man有问题，关掉man就正常了。
我现在准备恢复到5.4内核了。

jchen-git, 2020-10-05:
> 试试这个 commit ( [AmadeusGhost@2e74f09](https://github.com/AmadeusGhost/lede/commit/2e74f09b8b2964012a3a55f67c7f3137fe86008a)) 能不能解决问题

不知道怎么去抓你这个commit去编译，git玩的不熟 ^_^

blueveryday, 2020-10-05:
pppoe报错rename，然后拨号链接显示连上的，但是网络实际是断了，MAN口的iptv还可以使用。

AmadeusGhost, 2020-10-05:
> 不知道怎么去抓你这个commit去编译，git玩的不熟 ^_^

```
cd lede
wget https://github.com/AmadeusGhost/lede/commit/2e74f09b8b2964012a3a55f67c7f3137fe86008a.patch
git apply 2e74f09b8b2964012a3a55f67c7f3137fe86008a.patch
```

Ljzkirito, 2020-10-05:
> 试试这个 commit ( [AmadeusGhost@2e74f09](https://github.com/AmadeusGhost/lede/commit/2e74f09b8b2964012a3a55f67c7f3137fe86008a)) 能不能解决问题

昨天`git clone --depth 1 https://github.com/AmadeusGhost/lede -b modified openwrt`
今天不保留配置升级，仍然出现PPPoE显示连上，但网络实际断连的情况

Ljzkirito, 2020-10-05:
> > 试试这个 commit ( [AmadeusGhost@2e74f09](https://github.com/AmadeusGhost/lede/commit/2e74f09b8b2964012a3a55f67c7f3137fe86008a)) 能不能解决问题
> 
> 昨天`git clone --depth 1 https://github.com/AmadeusGhost/lede -b modified openwrt`
> 今天不保留配置升级，仍然出现PPPoE显示连上，但网络实际断连的情况

日志如下
Sun Oct  4 19:28:28 2020 kern.notice kernel: [    0.000000] Linux version 5.4.68 (Ljzkirito@GitHub Actions) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0-2e74f09)) #0 SMP Sun Oct 4 11:28:20 2020
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] MIPS: machine is XiaoYu XY-C5
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sun Oct  4 19:28:28 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Oct  4 19:28:28 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Oct  4 19:28:28 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Zone ranges:
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Early memory node ranges
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
Sun Oct  4 19:28:28 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Sun Oct  4 19:28:28 2020 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sun Oct  4 19:28:28 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Oct  4 19:28:28 2020 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Sun Oct  4 19:28:28 2020 kern.debug kernel: [    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
Sun Oct  4 19:28:28 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
Sun Oct  4 19:28:28 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
Sun Oct  4 19:28:28 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00005660
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00005660
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] Memory: 509864K/524288K available (5738K kernel code, 253K rwdata, 1244K rodata, 1264K init, 238K bss, 14424K reserved, 0K cma-reserved, 262144K highmem)
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun Oct  4 19:28:28 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] CPU Clock: 1000MHz
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3822520893 ns
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.000008] sched_clock: 32 bits at 500MHz, resolution 2ns, wraps every 4294967295ns
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.007798] Calibrating delay loop... 663.55 BogoMIPS (lpj=1327104)
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.045996] pid_max: default: 32768 minimum: 301
Sun Oct  4 19:28:28 2020 kern.info kernel: [    0.050750] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... cut ...]
Mon Oct  5 09:09:54 2020 daemon.notice pppd[9521]: remote IP address 100.69.0.1
Mon Oct  5 09:09:54 2020 daemon.notice pppd[9521]: primary   DNS address 211.137.191.26
Mon Oct  5 09:09:54 2020 daemon.notice pppd[9521]: secondary DNS address 218.201.96.130
Mon Oct  5 09:10:01 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Mon Oct  5 09:10:13 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Mon Oct  5 09:10:13 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)

AmadeusGhost, 2020-10-05:
试试 #5605 ，另外就是在 https://github.com/AmadeusGhost/lede/commit/34d9c51bbb9828f5eaeb9730e8c19e0b898e048b 把小娱 C5 的有线驱动改成 4.14 上的了（这个 commit 不是解决问题的方法，不过可以开 hwnat 了）。

Ljzkirito, 2020-10-06:
> 试试 https://github.com/AmadeusGhost/lede/tree/ramips ，另外就是在 [AmadeusGhost@34d9c51](https://github.com/AmadeusGhost/lede/commit/34d9c51bbb9828f5eaeb9730e8c19e0b898e048b) 上把小娱 C5 的有线驱动改成 4.14 上的了（可以开 hwnat 了）。

开机一小时，PPPoE网络连接正常，未发现断网。

AmadeusGhost, 2020-10-06:
> 开机一小时，PPPoE网络连接正常，未发现断网。

已发 pr

blueveryday, 2020-10-07:
> > 开机一小时，PPPoE网络连接正常，未发现断网。
> 
> 已发 pr

HC5962的这个问题能不能一并解决一下？同样存在相同的问题。

Ljzkirito, 2020-10-07:
> > > 开机一小时，PPPoE网络连接正常，未发现断网。
> > 
> > 
> > 已发 pr
> 
> HC5962的这个问题能不能一并解决一下？同样存在相同的问题。

HC5962也是mt7621设备吧，是的话应该一并修复了

blueveryday, 2020-10-07:
> > > > 开机一小时，PPPoE网络连接正常，未发现断网。
> > > 
> > > 
> > > 已发 pr
> > 
> > 
> > HC5962的这个问题能不能一并解决一下？同样存在相同的问题。
> 
> HC5962也是mt7621设备吧，是的话应该一并修复了

对，就是mt7621的，现在再用5.4的内核。

Ljzkirito, 2020-10-07:
> 对，就是mt7621的，现在再用5.4的内核。

此[pr](https://github.com/coolsnowwolf/lede/pull/5605)是针对mediatek芯片修复的，HC5962也已经修复了

blueveryday, 2020-10-07:
> 即

合并了我测试一哈。

Ljzkirito, 2020-10-07:
> 合并了我测试一哈。

心急的话可以`git clone https://github.com/Ljzkirito/lede`试试，这是我Fork [AmadeusGhost/lede ramips](https://github.com/AmadeusGhost/lede/tree/ramips)分支的，已经合并相关commit，.config直接复制过去即可

Myoko, 2020-10-07:
有没有试过4.19.138 内核？

Ljzkirito, 2020-10-07:
> 有没有试过4.19.138 内核？

没有试过

Myoko, 2020-10-07:
> > 有没有试过4.19.138 内核？
> 
> 没有试过，也不知道怎么更换lean的内核版本

云端编一个试试看！

Ljzkirito, 2020-10-07:
> > > 有没有试过4.19.138 内核？
> > 
> > 
> > 没有试过，也不知道怎么更换lean的内核版本
> 
> 云端编一个试试看！

所以怎么编译4.19.138 内核的版本？不是编程出身的，不会呀。😂

Myoko, 2020-10-07:
> 看[P3TERX](https://p3terx.com/archives/build-openwrt-with-github-actions.html)Actions-OpenWrt

用P3TERX的Actions-OpenWrt 在云端编译就可以了！

AmadeusGhost, 2020-10-07:
> 有没有试过4.19.138 内核？

ramips 只有 4.14 和 5.4，而且驱动都同步官方了，回去意义不大

Myoko, 2020-10-07:
> > 有没有试过4.19.138 内核？
> 
> ramips 只有 4.14 和 5.4，而且驱动都同步官方了，回去意义不大

ramips 那就没办法了！

closed 2020-10-07 by coolsnowwolf

blueveryday, 2020-10-08:
重新编译，刷了，好像不会断网了，但是有个信息一直在提示。
似乎有断流的情况（我不确定，还需要再观察一下）。
我的网络：HC5962+wan+man(iptv)，单线复用。
IPTV保持流畅使用，希望没有问题，我今天观察一天试试。

AmadeusGhost, 2020-10-08:
@blueveryday 你的问题不一样，请另开一个 issue

PR #5605 (2020-10-07): ramips: XiaoYu-C5: reuse ralink-eth driver
fix #5599